ZoneMinder/zoneminder #1162: Errors on fresh ubuntu install

Simanas, 2015-11-14:
I am struggling with this for last few days... I am python/postgresql/nginx guy, so php/mysql/apache stack is a bit alien for me. Did this a few times, getting the same results. 

So I start with a fresh install of Ubuntu Server 14.04.3 LTS with OpenSSH Server, and LAMP Server on board as described in wiki here:
https://wiki.zoneminder.com/Ubuntu_Server_14.04_64-bit_with_Zoneminder_1.28.1_the_easy_way. 

Later I follow every single line and in that wiki until I get ZoneMiner running on my little home server. 

That's cool, but then I can't get any video stream. I guess that's because of all of these errors:

[image]

Any ideas what am I doing wrong, or is there something missing in wiki?

angelrr7702, 2015-11-15:
I have the same problem , I try old fix relate to the same type of error and still the same error . I assuming this a new problem ,,, introduced at 1.28.108 and ubuntu before of that one it worked fine for me ...  the problem when you installing with ppa:iconnor/zoneminder-master with recently update 10/30/2015 for vivid

pliablepixels, 2015-11-17:
https://forums.zoneminder.com/viewtopic.php?f=32&t=23375&p=89087#p89087

knnniggett, 2015-11-18:
This does not describe a bug. Rather, it is a common configuration issue. 
Please refer to our github posting rules and then create a post in our user forum:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2015-11-18 by knnniggett

angelrr7702, 2015-11-19:
Sorry but at the moment any of the common configuration issue fix the problem and the link for forums fix was before of this error .. my container was working on that time but broke when update to the new 1.28.108 and now still broken with 1.28.109 ..