ZoneMinder/zoneminder #1163: [patch] php54 incompatibility in views/image.php

abishai, 2015-11-15:
I upgraded to latest zmrepo development version and noticed that imagescale function was intruduced. This one is from php 5.5 and is not available in CentOS7 repo.

As CentOS7 is supported by ZM developers, I think workaround is required. I wrote simple wrapper around imagecopyresampled()

// Compatibility for PHP 5.4 
if (!function_exists('imagescale'))
{
	function imagescale($image, $new_width, $new_height = -1, int $mode = 0)
	{
            $mode; // Not supported
      
            $new_height = ($new_height == -1) ? imagesy($image) : $new_height;
            $imageNew = imagecreatetruecolor($new_width, $new_height);
            imagecopyresampled($imageNew, $image, 0, 0, 0, 0, (int)$new_width, (int)$new_height, imagesx($image), imagesy($image));
            assert($new_width);

            return $imageNew;
	}
}

pliablepixels, 2015-11-15:
@abishai  - can you convert this into a Pull Request and reference it from this issue?  I can test it by forcing it on.

abishai, 2015-11-15:
Never did that before, hope, I did it right
https://github.com/ZoneMinder/ZoneMinder/pull/1164

pliablepixels, 2015-11-16:
Works well.

commit c0216f74 (2015-11-16): Merge pull request #1164 from abishai/master

compat for php 5.4.  Fixes #1163
closed 2015-11-16 by kylejohnson