ZoneMinder/zoneminder #1167: Upgrade issue - bug in version comparison

escot, 2015-11-19:
Hi
I was upgrading from 1.27 (installed in April'14, I didn't touch it since then)  to the latest 1.28.109.  
The upgrade went smoothly until the zmupdate.pl step  - which actually did nothing except setting the new version number in mysql:

```
root@host:/home# zmupdate.pl

Initiating database upgrade to version 1.28.109 from version 1.27

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop :

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /var/tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.28.109
Loading config from DB
Saving config to DB

Database upgrade to version 1.28.109 successful.
```
It took me time to realize that zmupdate actually failed and did nothing with the database, and more time to get the root cause of this: the version comparison in pl is buggy, i.e. it does not compare 1.27 (without dot zero at the end!) to 1.28.x.

Workaround that helped me - the following command did the job finally:

```
root@host:/home# zmupdate.pl --version=1.27.0
Initiating database upgrade to version 1.28.109 from version 1.27.0

WARNING - You have specified an upgrade from version 1.27.0 but the database version found is 1.27. Is this correct?
Press enter to continue or ctrl-C to abort :

Please ensure that ZoneMinder is stopped on your system prior to upgrading the database.
Press enter to continue or ctrl-C to stop :

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /var/tmp/zm if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.28.109
Loading config from DB
Saving config to DB
Upgrading DB to 1.27.0 from 1.27.0

Database successfully upgraded to version 1.27.0.
Upgrading DB to 1.27.99.0 from 1.27.0

Database successfully upgraded to version 1.27.99.0.
Upgrading DB to 1.28.0 from 1.27.0

Database successfully upgraded to version 1.28.0.
Upgrading DB to 1.28.1 from 1.27.0

Database successfully upgraded to version 1.28.1.
Upgrading DB to 1.28.99 from 1.27.0

Database successfully upgraded to version 1.28.99.
Upgrading DB to 1.28.100 from 1.27.0

Database successfully upgraded to version 1.28.100.
Upgrading DB to 1.28.101 from 1.27.0

Database successfully upgraded to version 1.28.101.
Upgrading DB to 1.28.102 from 1.27.0

Database successfully upgraded to version 1.28.102.
Upgrading DB to 1.28.103 from 1.27.0

Database successfully upgraded to version 1.28.103.
Upgrading DB to 1.28.104 from 1.27.0

Database successfully upgraded to version 1.28.104.
Upgrading DB to 1.28.105 from 1.27.0

Database successfully upgraded to version 1.28.105.
Upgrading DB to 1.28.106 from 1.27.0

Database successfully upgraded to version 1.28.106.
Upgrading DB to 1.28.107 from 1.27.0

Database successfully upgraded to version 1.28.107.
Upgrading DB to 1.28.108 from 1.27.0

Database successfully upgraded to version 1.28.108.
Upgrading DB to 1.28.109 from 1.27.0

Database successfully upgraded to version 1.28.109.

Database upgrade to version 1.28.109 successful.

```

So basically, we have 2 bugs here:
1/ version comparison
2/ zmupdate modifies the current version is the DB despite it actually failed to upgrade!

I believe this will affect more and more users and is easy to fix.

pliablepixels, 2015-11-21:
I believe this was fixed in the master branch https://github.com/ZoneMinder/ZoneMinder/pull/1094

knnniggett, 2015-11-22:
Looks like he is using a recent build from master. We'll have to take another look at how zmupdate.pl does its version checking.

escot, 2015-11-22:
yes, I was using the most recent master, so  #1094 doesn't solve it.  
thank you.

pliablepixels, 2015-11-22:
Yes, indeed it seems other parts of the code are using "lt" and "gt" operators which would likely fail. I'll check the flow again when I am back

SteveGilvarry, 2015-12-02:
Ok so I did some testing and the issue is how the conversion between decimal version and dotted decimal versions is done.
```
#!/usr/bin/perl -w
use strict;
use version;

my ($vcurrent) = version->parse( '1.28.102' );
print $vcurrent->normal, "\n";
my ($vold) = version->parse( '1.27' );
print $vold->normal, "\n";
if ($vcurrent ge $vold ) {
        print( "Upgrading DB to $vcurrent from $vold\n" );
}
```
returns 
v1.28.102
v1.270.0

Adding .0 to the end of 1.27 makes it work.

SteveGilvarry, 2015-12-02:
Adding v in front of the version numbers will let it know that they are all dotted decimal and 1.27 is actually 1.27.0. Fix inbound later today.

SteveGilvarry, 2015-12-16:
Closing this, especially as it was a blocker for 1.29, but the primary issue is fixed in #1176.
I do agree that the secondary issue with the upgrade script  not failing is still valid, happy to be guided by team if they want to create an new issue to track that not attached to 1.29.

knnniggett, 2016-01-01:
I'm reopening this as this bug does not appear to have been completely fixed in the 1.29.0 release candidate:

```
Upgrading database to version 1.28.110
Loading config from DB
Saving config to DB
Upgrading DB to 1.28.110 from 1.28.109
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.28.110.
Upgrading DB to 1.29.0 from 1.28.109
Warning: Using a password on the command line interface can be insecure.

Database successfully upgraded to version 1.29.0.

Database upgrade to version 1.28.110 successful.
```

So far the error appears to be non-fatal. Merely the order is not right.

SteveGilvarry, 2016-01-01:
@knnniggett I think that is coming from mysql where we are passing password in via -p switch. I suspect the change is in mysql that is warning about using that switch. Never saw it during my testing, so maybe you have later version.

knnniggett, 2016-01-01:
This is beginning to look like user error on my behalf. The sort algorithum in zmupte.pl is working fine, but my Base.pm contains a ZM_VERSION of 1.28.110.  That value comes direct from a make macro.

knnniggett, 2016-01-01:
Found the problem and fixed in commit 99b65f0
Then merge that commit over to the 1.29.0 branch. 
The sort in zmupdate.pl is working fine.

closed 2016-01-01 by knnniggett