ZoneMinder/zoneminder #1179: Libvlc segmentation fault

SteveGilvarry, 2015-12-04:
Different fault to the one described already in another issue. This seems be affecting more users than before and for me a camera that used to work now seg faults as soon as I swap to libvlc.
https://forums.zoneminder.com/viewtopic.php?f=32&t=23556

==26625== Invalid read of size 8
==26625==    at 0x9DA66B7: std::string::clear() (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.19)
==26625==    by 0x4707C5: Authenticator::reset() (zm_rtsp_auth.cpp:50)
==26625==    by 0x470677: Authenticator::~Authenticator() (zm_rtsp_auth.cpp:46)
==26625==    by 0x20C45A6B: ??? (in /usr/lib/vlc/plugins/access/liblive555_plugin.so)
==26625==    by 0x20C45F59: ??? (in /usr/lib/vlc/plugins/access/liblive555_plugin.so)
==26625==    by 0xD1B4177: module_load (in /usr/lib/libvlccore.so.7.0.0)
==26625==    by 0xD1B472D: vlc_module_load (in /usr/lib/libvlccore.so.7.0.0)
==26625==    by 0xD174247: demux_New (in /usr/lib/libvlccore.so.7.0.0)
==26625==    by 0xD180B95: InputSourceInit (in /usr/lib/libvlccore.so.7.0.0)
==26625==    by 0xD181B6A: Init (in /usr/lib/libvlccore.so.7.0.0)
==26625==    by 0xD1850E5: Run (in /usr/lib/libvlccore.so.7.0.0)
==26625==    by 0x7605181: start_thread (pthread_create.c:312)
==26625==  Address 0xffffffffffffffe8 is not stack'd, malloc'd or (recently) free'd

knnniggett, 2015-12-06:
I've spent some time collecting data on this, and here is what I've found so far:
- LibvlcCamera::PrimeCapture, part of zm_libvlc_camera.cpp, is working fine on my system. libvlc_media_player_play is being called, and it is returning 0, indicating success.
- The segfaults are occuruing when LibvlcCamera::Capture is called. The code never gets past the while loop, which implies there is something wrong going on with mLibvlcData.newImage.getUpdatedValue. The segfaults go away when I comment out everything and just return 0, but obviously I get no video.
- I find it interesting that zm_rtsp_auth is mentioned in the crash trace above. I didn't expect to see that in a libvlc crash. vlc has its own authenticator in the live555 library, so I would expect to see zoneminder's being called as well. Since my segfault does not metnion this, I may be looking at a difference segfault, but I don't know yet,
-

knnniggett, 2015-12-06:
Something I could use some help on is what in the world this is called in C++ land:
`template <class T> const T ThreadData<T>::getUpdatedValue( int secs ) const`

This is some kind of way to implicitly create objects/methods based off this template, but I haven't got a clue how this works exactly or what kind of keywords I can use to learn about it.

SteveGilvarry, 2015-12-07:
Not making any presumptions here. http://www.codeproject.com/Articles/257589/An-Idiots-Guide-to-Cplusplus-Templates-Part

SteveGilvarry, 2015-12-07:
@knnniggett I agree it is odd, but all the backtraces posted by others in the thread seem to have the same thing after liblive555.
```
Oct 17 10:24:50 zm zmc_m3[15090]: ERR [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x10340) [0x7f1b89241340]]
Oct 17 10:24:50 zm zmc_m3[15090]: ERR [Backtrace 2: /usr/bin/zmc(_ZN13Authenticator5resetEv+0x10) [0x7f1b8a3ab3d0]]
Oct 17 10:24:50 zm zmc_m3[15090]: ERR [Backtrace 3: /usr/bin/zmc(_ZN13AuthenticatorD1Ev+0x1e) [0x7f1b8a3ab44e]]
Oct 17 10:24:50 zm zmc_m3[15090]: ERR [Backtrace 4: /usr/lib/vlc/plugins/access/liblive555_plugin.so(+0x6ebb) 
```

knnniggett, 2015-12-07:
@SteveGilvarry Can you comment out the code in LibvlcCamera::Capture and just return 0 to see if the fault disappears on your end? That will at least tell us if we are experiencing the same issue.

sogood007, 2015-12-07:
The fact that the destructor Authenticator::~Authenticator() is called in the middle of VLC. I wonder it is because there was some exception get throw so C++ is trying to clean thing up and unwinding some stack (if Authenticator was allocated from stack). The zm_thread.cpp has a lot of throw include indirectly from getUpdatedValue() through wait().   Just $0.02

knnniggett, 2015-12-09:
The last time I troubleshoot a fault like this, it took me about 2 months so I appreciate everyone's patience in this matter. All comments are appreciated. This will probably take someone pointing out something I hadn't considered.

I made a little progress last night. Turning up debug to level 8 shows zoneminder calling Condition::wait just before crashing.

So at the moment, I'm looking at the pthread_cond_timedwait command as that looks to be what is causing the crash. https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_thread.cpp#L129

If an exception is being thrown, that information does not appear to be written anywhere. I grep'ed my entire /var/log folder for the error text, which did not reveal anything.  

I need to see if I can get valgrind to reveal additional info.

sogood007, 2015-12-09:
This aligns with my theory.  
For some unknown reason, the timedwait is returning with error code which cause the exception
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_thread.cpp#L129

While C++ is handling the exception, something wrong happened and crashed zm.  That is the reason why there is nothing in the log. 

Authenticator class appears allocated from heap so I looked for "delete" and found this
https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_rtsp.cpp#L227

I am guessing there is some destructor being call and the Authenticator get delete but the Authenticator object is not valid so as the crash.

SteveGilvarry, 2015-12-16:
@knnniggett I was hitting something as I came out of PrimeCapture and couldn't see anything other than vector destructors. But then it seemed to change and now it seems consistently at the grab an image stage. https://github.com/ZoneMinder/ZoneMinder/blob/master/src/zm_libvlc_camera.cpp#L204
I also found that live555 has this http://www.live555.com/liveMedia/doxygen/html/classAuthenticator.html, no idea if it could be causing issues. @connortechnology we need your C++ fu to have a peek at libvlc.

SteveGilvarry, 2015-12-16:
@knnniggett still testing but changing Authenticator class to zmAuthenticator has got me past my segfault code in debug.

knnniggett, 2015-12-16:
Nice. 
Mental note: class names must be unique, including all the classes from external libraries. 

You know, as picky as the gcc compiler is, I'm surprised this did not generate a compiler warning.

sogood007, 2015-12-16:
Great.  That explains the crash.  Because the live555 is a dynamic linked library.  Unless we include header files of both Authenticators into a .cpp, gcc won't notice the conflict.   When the dynamic loader loaded the symbol and code from live555's so, it won't load live555 Authenticator as it has from from ZM.  loader doesn't know the difference of the content.   I guess.... When the exception happens, live555 trying to destruct its own Authenticator but it ends up used the ZM's Authenticator destructor so as the segfault.  Rename ZM's authenticator will help or all the ZM classes can be used its own c++ namespace.

SteveGilvarry, 2015-12-16:
@sogood007 I was looking at doing namespaces but maybe quicker option is just what I have done so far. If you had some input on how to achieve namespaces I am happy to implement. is it as simple as putting all zm classes into a zm namespace?

sogood007, 2015-12-16:
@SteveGilvarry to use namespace, it is pretty much need to wrap around all the .h and .cpp with               

```
namespace zm {    
```

and end with

``` 
} 
```` 
e.g. 

```
namespace zm { 
class ... {

}

class ... {
}

}
```

So it takes some editing work in the classes which you want to place into namespace.  That could be a longer term solution.

commit 88ae78ed (2015-12-19): Merge pull request #1190 from SteveGilvarry/libvlc_segfault

Fixes #1179 Libvlc Live555 Segmentation Fault
closed 2015-12-19 by knnniggett