ZoneMinder/zoneminder #1183: RTSP stream in camera working in VLC, but not in libvlc

heximal, 2015-12-08:
Hello,

Here are my system specs:

ZM 1.28.109, installed from ppa:iconner
Ubuntu server 15.10

I have an Escam QD300 (also called "Brick"), which works fine with VLC at full frame rate (~25 fps @ 1280x720), but has problems in Zoneminder.

If using Remote > RTP/RTSP, I see messages about frame decoding error (http://pastebin.com/Lg1uCe4U) every 30 frames. I've discovered that if I change the GOV length from the default 2 to 12, using Onvif Device Manager, I see this error every 400 frames. Either way, the playback and recording in ZM is choppy and there are a considerable number of skipped frames, which makes this camera useless for security purposes (appearing and disappearing objects in the middle of the recording).

If I use FFMPEG, I get a lot more CPU load and playback and recording is pretty much as useless as with RTSP.

If I use libvlc, I can't connect to the camera (no image). I see the following errors:
http://pastebin.com/w61tHg0D

I'd be grateful for any suggestions - this is a cheap, but good quality HD camera, that is gaining popularity.

knnniggett, 2015-12-08:
First, please read our posting guidelines:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

This issue, the way you've written it, is a request for camera configuration, which means it belongs in the user forum. There are users with your exact camera and it is known to work with zoneminder. A quick google reveals someone has even written a blog about the camera in question.
https://www.google.com/search?client=ubuntu&channel=fs&q=zoneminder+qd300&ie=utf-8&oe=utf-8

It is true that part of your request does describe a known bug with the libvlc source type. However, there already two open issues for that. #1179 #343
If you would like to participate and have new information to share then please post into one of the existing issues.

closed 2015-12-08 by knnniggett

heximal, 2015-12-08:
I've read the guidelines - since there are numerous posts about how to get this camera working, just as you said, and none of them work with the current ZM release, it was my conclusion that I was informing the developers that they may have changed something, which makes this, and other cameras, unusable (i.e. a bug report).

I'm most certainly not up to date with bug reports from others - I see I'm not the only one having this issue. Thanks for posting these links.

Is it possible to use vlc server to transcode the camera's RTSP stream to something usable by ZoneMinder? Can you point me in the right direction, if you've tried this in the past?

Thank you!

knnniggett, 2015-12-09:
Point of clarification: The current ZM release is 1.28.1
ZoneMinder 1.28.109 is not a release. It is a development snapshot

This camera is known to work with both.
I'm sorry that camera isn't working for you at the moment, but that doesn't mean it is a bug. If you want to inform us of a camera problem then post in the user forums. We respond to those too.

There are at least two guys in our IRC channel who use that camera. Join us tomorrow if you want.

heximal, 2015-12-09:
I understand, thank you for your help!