ZoneMinder/zoneminder #1186: Failed to set video standard 45056: Device or resource busy just on DVR card.

orsty3001, 2015-12-09:
I'm running 1.28.1 on slackware 14.1. No video will display on the cameras connected to the DVR card. /dev/video0 to /dev/video3 all give the same error.

    
Failed to set video standard 45056: Device or resource busy

It's a Techwell TW6816 8 channel DVR card. The IP cameras work find but this card doesn't with Zoneminder.  I'm able to output the video from the DVR card to a file using ffmepg.

I tried lsof and all video outputs return nothing.

Apache is in the video usergroup.

For some reason Zoneminder thinks the DVR card is busy but everything else can use it without a problem.

knnniggett, 2015-12-09:
Please do not double post. 
You will get a response to your thread in the user forum, when someone comes along who can help with this kind of problem.

closed 2015-12-09 by knnniggett

recklessrewared, 2016-01-18:
Does anyone have update on this? I have the same card that I would like to use with my zoneminder set up.

Danteba, 2016-06-25:
I resolved the problem using the Source Type: Libvlc 
using Source Path: v4l2:///dev/video0:input=1