ZoneMinder/zoneminder #1187: "Add New Monitor" button greyed out when OPT_USE_AUTH is checked

geppo01, 2015-12-09:
It is impossible to add a new monitor from the Web interface when OPT_USE_AUTH is checked. 
The button is greyed out. If I turn off OPT_USE_AUTH then the button reappears and is functional. I am running on Ubuntu 14.04 with Zoneminder 1.28.1

SteveGilvarry, 2015-12-19:
Controlled by Monitor settings under User options. Using admin user I could not replicate your issue, as they have full control or edit for all options. I created a test user with Monitor = View permissions and my Add New Monitor button was greyed out.
http://zoneminder.readthedocs.org/en/latest/userguide/options/options_users.html

closed 2015-12-19 by SteveGilvarry

geppo01, 2015-12-19:
Hi Steve,

I am sure I saw this behavior with both the admin user and a different
user. I will try again, and let you know.

Gerald

SteveGilvarry, 2015-12-19:
Gerald,
Post it over on the forum https://forums.zoneminder.com and if we can replicate we will reopen as a bug here.