ZoneMinder/zoneminder #1188: problem with dir_events being a relative path to webroot.

connortechnology, 2015-12-15:
Currently we allow the use of a relative path from the webroot.  

I think this is dangerous, confusing, and as we get different ways of accessing ZM (different UI's, apis or running with a UI) not really possible.  

So I'd like to deprecate it.

Thoughts?

knnniggett, 2015-12-15:
Agreed, and it has been in the back of my mind to follow up on this.

Over the summer, a user popped into our irc channel and demonstrated that ZoneMinder actually detects when an absolute folder name is used and in the php responds appropriately. Could have sworn this didn't work, but there it was working as intended with no change to the source code required. At least as far as I could tell.

I've been meaning to roll this out into a testing rpm on zmrepo. Just haven't had the time.

So all that really needs to be done here is for us to update our documentation and then repackage zoneminder to use absolute folders, outside the webroot. 

In the case of rpm packaging for zmrepo, that would merely mean removing the symlinks to the events and images folders from the webroot.... I won't have to actually move the real locations of these folders, which makes it easier to migrate existing systems.

knnniggett, 2016-04-20:
As an update, here is some of the existing code that properly handles paths outside the webroot:
https://github.com/ZoneMinder/ZoneMinder/blob/master/web/includes/functions.php#L1246

I would not have believed it was in there had I not tried it myself.

connortechnology, 2016-04-20:
I think it does work at the moment.  I just think it shouldn't be allowed.

josh4trunks, 2016-04-28:
See here for a related suggestion
https://github.com/ZoneMinder/ZoneMinder/issues/601#issuecomment-215324300

SteveGilvarry, 2017-03-18:
Confirm that it still works without events and images folders, if not fix and release

PR #1923 (2017-07-21): Don't create symlinks in ZoneMinder webroot
Fixes #1188 and #601 

needs testing

I'm still investigating the use (or lack) of the cmake var ZM_CONTENTDIR. It may only be needed during a from-source build, in which case I can remove it from both deb & rpm packaging
closed 2017-07-21 by connortechnology