ZoneMinder/zoneminder #1197: The main README - build from source

pliablepixels, 2015-12-19:
@knnniggett - when I was updating the ubuntu.rst file for the official docs, I noticed the build-pkg process of building from source was not working. @connortechnology noted this uses auto tools and may not work. (it failed in build package). If that is the case, should we use cmake instructions instead? I've updated ubuntu.rst with cmake instructions here https://github.com/ZoneMinder/ZoneMinder/blob/master/docs/installationguide/ubuntu.rst

Also, @knnniggett , please see https://github.com/ZoneMinder/ZoneMinder/pull/1171

knnniggett, 2015-12-20:
I'm not familiar with debian build procedure at all so you'll have to talk to someone else to get that part of the documentation updated.

pliablepixels, 2015-12-20:
So #1171 is just about moving out source build instructions out of readme. It confuses new folks imho.

knnniggett, 2015-12-20:
Understood. I'll review & merge any open pr's not directly related to 1.29 after we release.

closed 2017-03-18 by SteveGilvarry