ZoneMinder/zoneminder #1204: All of the Perl code uses "#!/usr/bin/perl"

lembark, 2015-12-25:
This prevents using a modern perl on most systems (e.g., many RH systems are stuck at 5.8 compiled for binary compatability with 5.00503). Simple fix: use "#!/usr/bin/env perl", allowing the user to configure their environment for the preferred perl.

This arrangement also makes testing ZoneMinder with different versions of Perl a whole lot simpler since perlbrew (or a simple path update) can pick the version to use. It also simplifies installation into Docker since the full Perl on most distro's is a bit bloated.

knnniggett, 2015-12-26:
ZoneMinder works just fine on RedHat based systems and their binary clones. 

I do not agree with letting the system auto-pick a Perl interpreter using env for a system package like ZoneMinder.

One of the "features" of using env is all the process names get replaced with generic ones like "perl'. So instead of seeing zmpkg and zmdc in your process list, you see a bunch of instances of "perl" in your process list.

Indeed using env is documented as not allowed in the packaging documentations of distros like Debian in Redhat. 

For example:
https://fedoraproject.org/wiki/Script_Interpreters_%28draft%29

What you'll find if you look at other projects that use env to find their script interpretor, such as Ansible, is by the time the applications makes it into package form, the env command has been stripped out and the shebang has been put back into its more traditional form.

The env tool is handy for users who install and use Perl under their home directory, but ZoneMinder is a system tool, which is not intended to work in that kind of environment. 

ZoneMinder is intended to be built against a specific version of Perl found on the build host. If you want to change the version of Perl then rebuild ZoneMinder. This is by design, and is the same procedure one would have to make when changing the version of ffmpeg, liblv, etc.

closed 2015-12-26 by knnniggett

connortechnology, 2015-12-26:
I could see us merging a PR that made the shebang line configurable so that a developer in your situation could build everything to use env, but our packaging scripts would use the standard /usr/bin/perl.

lembark, 2015-12-30:
On Sat, 26 Dec 2015 07:45:36 -0800
Isaac Connor <notifications@github.com> wrote:

> I could see us merging a PR that made the shebang line configurable 
> so that a developer in your situation could build everything to use 
> env, but our packaging scripts would use the standard 
> /usr/bin/perl.

Problem there is that /usr/bin/perl is useless on many distros,
through use of ancient or badly compiled perl executables. For
exmple, anyone useing perlbrew to get a current version for 
either use or testing cannot make use of ZoneMinder: they cannot
install any modules into /usr/lib, etc, and the installed modules
will not be available to the local copy of perl in any case. 

Also assumes that anyone attempting to package ZoneMinder will all
be using the same version of Perl, which is not guaranteed.

In my own case, I don't have sudo installed on the system -- separate 
issue -- but depending on the distro's perl installation when the 
user may have their own copy introduces make installing the package
nearly impossible.

enjoi

-- 
Steven Lembark                                             3646 Flora Pl
Workhorse Computing                                   St Louis, MO 63110
lembark@wrkhors.com                                      +1 888 359 3508