ZoneMinder/zoneminder #1208: API Change status of monitor - Enabled = true with error

morphey1, 2015-12-28:
Hi,

I have ZM Version: 1.28.1

I write a script in js for ZM's Control with webpp.
I have a problem with the API.

In the docs is written:

http://zoneminder.readthedocs.org/en/latest/api.html#change-state-of-monitor-1

```
curl -XPOST http://server/zm/api/monitors/1.json -d "Monitor[Function]=Modect&Monitor[Enabled]:true"
```
I tried the url (with cookie authentication)  and the result is:

```
$ curl -c cookie.txt -XPOST http://server/zm/api/monitors/3.json -d "Monitor[Function]=Modect&Monitor[Enabled]:\"1\""
{
    "message": "Saved"
}
```

The "Function" of camera  changes correctly to "Modect" (or "Monitor"), but the flag "Enabled" remain in "0" (false) state.

Has anyone had the same problem?
Who can help?

knnniggett, 2015-12-28:
The API is only available in the latest development snapshot. It is not available in 1.28.1.
You either need to upgrade to a development snapshot, currently 1.28.109, or wait until the next release, 1.29.0, which is just around the corner.

closed 2015-12-28 by knnniggett

morphey1, 2015-12-28:
Hi @knnniggett ,

really bad this thing :)

how can I do to enable a single remote camera?

For the dev version, I currently use the repo for Fedora. Can I install the beta via yum?

thanks

knnniggett, 2015-12-28:
If you are using zmrepo to install ZoneMinder on Fedora, you need to enable the testing repo to get 1.28.109. 

And apparently the testing repo file got left out of Fedora 23. I will push up a new zmrepo rpm shortly.

morphey1, 2015-12-28:
Hi @knnniggett ,

still exist a method for change the state from mysql db?

I have fedora 21 and the beta is not present.

knnniggett, 2015-12-28:
Fedora has limited life cycles on their distros, and you can expect zmrepo to follow the same pattern.
Consequently, you need to upgrade to at least Fedora 22.

Please move this conversation to either our IRC channel or the user forum for additional help. 

Our github forum is reserved for bug reports and feature requests only:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules