ZoneMinder/zoneminder #1214: ZM_SERVER_ID uninitialized

knnniggett, 2016-01-01:
@connortechnology 
I'm not sure if this is by design or something that needs fixing.

On a standalone, non-multiserver, machine just upgraded to 1.29.0, when I start zoneminder, I get the following non-fatal error:
```
[abauer@localhost db]$ sudo service zoneminder start
Starting ZoneMinder: Use of uninitialized value $Config{"ZM_SERVER_ID"} in concatenation (.) or string at /usr/bin/zmpkg.pl line 229.
```
The logic that sets ZM_SERVER_ID is here:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Config.pm.in#L106

It first checks if ZM_SERVER_NAME is set and if it is not it checks ZM_SERVER_HOST.

However, in my case, neither ZM_SERVER_NAME nor ZM_SERVER_HOST are set.

So the question is, for a standalone machine running 1.29.0 and not using multiserver, should the logic in Config.pm consider this as a third case, or should it be required for here on after to set ZM_SERVER_HOST in zm.conf?

connortechnology, 2016-01-02:
Yeah... sounds like it should set it, just to remove the warning.  Patch coming in a few.

This is the problem with this last-minute merging of multi-server I guess.  No one has been testing the upgrade case much.  Still, this is just a warning, everything still works. 

I have another patch to fix issues in the db logging.

knnniggett, 2016-01-02:
No sweat. This is why I made today's release just a release candidate.
With all the stuff we merged since our last release, I knew there would some things to work out.

knnniggett, 2016-01-02:
I'll do another round of testing tomorrow.

knnniggett, 2016-01-02:
Looks good.

closed 2016-01-02 by knnniggett