ZoneMinder/zoneminder #1218: Compatibility with old webcams

tommycw1, 2016-01-02:
Had a camera die on me and was trying to use two old webcams I had used with ZM around 2003 time frame.  Neither of these cameras worked in ZM, but did work in cheese and xawtv:

Camera 1 used the gspca_pac207 kernel module
Camera 2 used the gspca_sonixb kernel module 

Camera 1 error with ZM: http://pastebin.com/uRY7gtuL
Camera 2 error with ZM: http://pastebin.com/hACcsTHB

I googled around and found a lot of people complaining these camera work with Cheese and Xawtv, but not applications like Skype.  I finally found the problem here:

https://forums.zoneminder.com/viewtopic.php?t=16354

All I had to do was add 
$ENV{LD_PRELOAD} = '/usr/lib/libv4l/v4l1compat.so';  to my zmdc.pl and these camera worked!  

Seems like a simple 1 line to add to add compatibility to many old cameras laying around out there.  These are the typical cameras first time ZM users likely try.  I do not know if this has negative impact on anything else.

I've attached my zmdc.pl for your viewing pleasure

.

tommycw1, 2016-01-02:
I should have added that I am running ZM 1.28.1 on Ubuntu 15.10.

tommycw1, 2016-04-27:
When upgrading to 1.29.1 I realized I had dug into this a bit more and never updated this issue.  There are actually 2 lines needed:

Insert the following lines just before the "my @daemons = (" line

$ENV{LD_PRELOAD} = '/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so';
$ENV{LD_PRELOAD} = '/usr/lib/x86_64-linux-gnu/libv4l/v4l2convert.so';

I can attach a new copy of my zmdc.pl if needed.  Please let me know.

connortechnology, 2016-05-09:
Just a point to note here... the second line would overwrite the first one in the ENV.  So really you only need the second line.

connortechnology, 2016-05-09:
Also, on ubuntu 1604 it appears to have moved to /lib/x86_64-linux-gnu/libv4l/v4l1compat.so

connortechnology, 2016-05-09:
sorry, my mistake... it's in usr of course.

connortechnology, 2016-05-09:
This used to work with my old stv06xx lego cam.  Now it doesn't seem to anymore on ubuntu1604

closed 2017-03-18 by SteveGilvarry