ZoneMinder/zoneminder #1219: monitorUrl incorrect in non-MultiServer Envinroment

knnniggett, 2016-01-03:
@connortechnology 
Should be easy to fix.

monitorUrl, as defined in watch.js.php looks like this when not in a multiserver environment: `var monitorUrl = 'https://';`

https://github.com/ZoneMinder/ZoneMinder/blob/master/web/skins/classic/views/js/watch.js.php#L51

That breaks a few of the widgets, like the state and the fps, in the Watch view.

So do we redefined $monitor->Server()->Url() to return empyset instead or just ignore it when it contains that particular value?

knnniggett, 2016-01-03:
Thinking out loud here. 
Should we redesign things such that, not selecting a server under the monitor config, results in a value of localhost, rather than emptyset or none? Or would that cause other issues?

connortechnology, 2016-01-03:
I think it needs to return a relative url or something similar... we can't assume that we are accessing localhost.  Easy fix though.

connortechnology, 2016-01-03:
If no Server is set, it uses ZM_BASE_URL ... which used to be defined... but now is not.  Ok.  Need to use ZM_BASE_PATH instead...

PR #1222 (2016-01-04): Multi server
Fixes #1219 

I decided that you were right knnniggett: The Server->Url should just return protocol/hostname, so in the multi-server case, just returning empty is preferred.
closed 2016-01-04 by knnniggett