ZoneMinder/zoneminder #1227: CakePHP version not compatible with PHP7

w41ru5, 2016-01-05:
My system upgraded to PHP7 I have ZoneMinder 1.29 running. I was wanting to try out the API layer, but I get an error with CakePHP:
`Error: Cannot use 'String' as class name as it is reserved`
`File: /srv/http/zoneminder/api/lib/Cake/Utility/String.php`
`Line: 25`

CakePHP (https://github.com/cakephp/cakephp/issues/7573) maked this as an issue they have resolved in versions 3+

onlyjob, 2016-02-21:
IMHO this is quite a problem as upcoming Debian release will be shipped without PHP-5 and only with PHP-7: https://lists.debian.org/debian-devel-announce/2016/01/msg00002.html

Related to #1291.

onlyjob, 2016-02-21:
The simple solution would be to upgrade CakePHP to 2.8 because "[CakePHP 2.8 is compatible with, and tested against PHP7][1]"

[1]: https://github.com/cakephp/cakephp/releases/tag/2.8.0

knnniggett, 2016-02-25:
Resolved by #1306

closed 2016-02-25 by knnniggett

wansco, 2016-06-10:
This probably bears a new thread, but this is where I ended up, so I figured I'd at least post it here.

I had some API issues with a clean install on ubuntu 16.04 related to cakephp. The solution was to add the following lines to /etc/apache2/conf-enabled/zoneminder.conf in the directory www section

  AllowOverride All
  Require all granted

(Note: this isnt related to the original post, but it might be useful to someone)