ZoneMinder/zoneminder #1229: Gather and collect telemetry data from ZoneMinder

kylejohnson, 2016-01-06:
This one might rustle some jimmies, which is why I am putting an issue in for it so as to open up the discussion.

I'd like to gather anonymous statistics from local ZoneMinder installs, and collect them back here at zoneminder.com.  Such stats would include:

* Number of monitors
* Version of ZoneMinder
* Distro and Distro Release
* Server stats (CPU, RAM)
* Amount of disk space (total and free)

This will be implemented in a new daemon (e.g. zmtelemetry.pl), and the user will be able to turn it off easily via the ZoneMinder options page.  By default, the option will be enabled.

A privacy message will be displayed linking to our privacy policy, as well as the ability to quickly turn it off, upon first use / install.

This mirrors how other projects, such as Firefox and Plex, handle collecting anonymous data.
[image]

We will be using this data in order to better understand our user base so that we can better understand which features and issues are most relevant.

knnniggett, 2016-01-06:
I'm good with this as long as we are transparent about it.

Simpler1, 2016-01-07:
I recommend something similar to what the Tomato Anon project has done.  

http://tomato.groov.pl/?p=306

Tomato is router firmware, but the Anon project spells out very clearly what is collected and the results are available for everyone to view.  After installation, the UI prominently displays the users options until they choose to opt in or opt out (which they can change later at any time).

[image]

connortechnology, 2016-01-08:
I would add to the data collected, type of monitor stream (rtsp/http ffmpeg/native/vlc, etc).

But otherwise, as long as it is transparent and opt-outable, I'm fine with ti.

kylejohnson, 2016-01-08:
@connortechnology Agreed, we'll need to come up with a full list of exactly what we're collecting (monitor stream type is definitely important).

alager, 2016-01-20:
@kylejohnson How often would the data be transmitted to the telemetry daemon?

kylejohnson, 2016-01-20:
Undetermined.  Possibly like zmaudit, where it runs at user-configurable interval, with a default of something sane (once a day, twice a day, etc).

kylejohnson, 2016-02-09:
Fixed port, proto and host in commit 609784f9023198aec771840be5bd39f144f3f245

kylejohnson, 2016-02-09:
OK, what data do we want to collect?    Something like this, for round one?

```json
"monitor_count": 500,
"event_count": 132083423424,
"distro": 'Metallinux',
"release": "666",
"zm_version": 0.01,
"system_memory": 1024 # megabytes
"processor_count": 4 # grep processor /proc/cpuinfo | wc -l
"monitors" : [
  {"id":1, "type": "Remote", "function": "Modect", "width": 1920, "height": 1080}
]
```

?

kylejohnson, 2016-02-09:
For date, we'll want the output of `date -In`

knnniggett, 2016-02-09:
Reminder to self to use an easy to find perl module to generate a uuid:
http://stackoverflow.com/questions/18628244/how-we-can-create-a-unique-id-in-perl

alager, 2016-02-09:
@knnniggett  If the uuid is going to be tracked in a/the DB, mysql and postgress can both make this for you, no extra module needed.  I presume since the uuid is going to be used to track installations, and the data will be pushed ( one way), then it will be generated once and stored at each instance of ZM.  In that instance the DB will create it and return it to you without issue, even on insert.

connortechnology, 2016-02-09:
I think processor type would be even more useful to know. 32bit vs 64 vs arm etc.

alager, 2016-02-09:
Would allocated storage size be useful?

kylejohnson, 2016-02-09:
>  mysql and postgress can both make this for you, 

@alager Sorry, not sure what you mean by this.  Could you elaborate?

The backend storage isn't going to be SQL, if that matters.

kylejohnson, 2016-02-09:
> I think processor type would be even more useful to know. 32bit vs 64 vs arm etc.**

You're right - forgot about that.  Maybe the output of `uname -a`.

I'd also love to log the user's IP address.  The output of `curl ifconfig.me/ip` would suffice, but I need to check their acceptable use policy.

alager, 2016-02-10:
@kylejohnson My presumption was/is that each users ZM node will be making the data structure that you outlined above, and then pushing that to your telemetry server for data collection.  In that model you don't want to re-create the uuid each time, since that identifies the node of interest.  Since each node will be gathering some info from the local ZM database instance (mysql mostly) to push to your telemetry DB, it seems like using the local ZM DB to create the uuid would be a natural fit, and no need to install an additional perl module.
[mysql uuid](https://dev.mysql.com/doc/refman/5.5/en/miscellaneous-functions.html#function_uuid)

alager, 2016-02-10:
@kylejohnson 

> The output of curl ifconfig.me/ip

This site doesn't seem to be responding.  Is it reliable?  They are the number one google hit though...
Another method, that I've used in the past, that doesn't rely on a third party, is to just get the IP from the connection when the data is pushed.  Whatever daemon accepts the connection can snag the IP, and then insert it into the data for storage in the telemetry DB.

knnniggett, 2016-02-10:
@alager Not sure why you thought we would be generating a uuid each time we upload. We certainly won't be doing that. The as-yet-to-be-written function call will only be called one time, per database. All subsequent calls will simply read the value out of the Config table.

kylejohnson, 2016-02-10:
@alager 

> This site doesn't seem to be responding. Is it reliable? They are the number one google hit though... Another method, that I've used in the past, that doesn't rely on a third party, is to just get [...]

I've had similar experiences with the site being unreliable.

While I originally planned on grabbing the IP from the connection, as you mentioned, with the current setup, that isn't possible (there's currently no proxy between the client and the end server which can intercept).

I could setup a proxy to do this job, or I could setup a simple php page at zoneminder.com which returns the user's IP address.  Hmm.  The former is probably better as it'll allow me to add other security features to the telemetry server.

kylejohnson, 2016-02-10:
OK, I've got a little script setup at https://wiki.zoneminder.com/ip.php which returns your IP address, so please use that @knnniggett.

I've also got a reverse proxy setup at https://telemetry.zoneminder.com, so that'll be the new host, and the port will be 443.  

Both of these URLs have rate limiting enabled, to help curb abuse.

alager, 2016-02-11:
@knnniggett 

>  function call will only create a uuid one time, per database. All subsequent calls will simply read the value out of the Config table.

Sorry my communications were not clear.  I think we are on the same page.  I was just noting that the function doesn't need to call an additional perl module to get the uuid, when a the ZM system already has something easily capable of making a uuid, and that is the DB (mysql or postgress,  oracle, etc.).

knnniggett, 2016-02-11:
@kylejohnson FYI, I am using a key of "ip" for the ip address. Let me know if this should change.

Due a lack of standards between distros and a lack of a common system tool to return exactly what we need, I am currently using the following to populate the "distro" and "release" tags:
```
$telemetry{distro} = runSysCmd("uname -a");
$telemetry{release} = runSysCmd("cat /etc/*{release,version}");
```

kylejohnson, 2016-02-11:
@knnniggett 

The key of 'ip' should be fine.

> Due a lack of standards between distros and a lack of a common system tool to return exactly what we need, I am currently using the following to populate the "distro" and "release" keys:

Hmm, couple things here. Finding a portable solution is going to be difficult.  On Ubuntu, can get the distro and release here:
```
ubuntu@ip-10-1-5-48:~$ cat /etc/os-release
NAME="Ubuntu"
VERSION="14.04.3 LTS, Trusty Tahr"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 14.04.3 LTS"
VERSION_ID="14.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
```

On Arch, there's the same method:
```
[kjohnson@kyletop ~]$ cat /etc/os-release 
NAME="Arch Linux"
ID=arch
PRETTY_NAME="Arch Linux"
ANSI_COLOR="0;36"
HOME_URL="https://www.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://bugs.archlinux.org/"
```

Does Fedora, Debian, CentOS, etc have /etc/os-release or /etc/lsb-release?

> $telemetry{distro} = runSysCmd("uname -a");

That's not super accurate either.  I think this would be better:
```perl
$telemetry{kernel} = runSysCmd("uname -r");
$telemetry{architecture} = runSysCmd("uname -m");
$telemetry{distro} = runSysCmd("grep '^NAME' /etc/os-release");
# Arch, for example, doesn't have a version
# Do we want VERSION, or VERSION_ID (see above output)
$telemetry{version} = runSysCmd("grep '^VERSION' /etc/os-release || echo 'N/A'");
```

> Any reason why you want nanoseconds returned in the datetime format? 

Not necessarily, no.  My /quick/ reading of the ElasticSearch docs mentions that they either use or expect (I can't recall the exact line) dates to be in an [ISO 8601](https://en.wikipedia.org/wiki/ISO_8601) format.

That said...  Looks like we can have Elasticsearch [automatically generate a _timestamp field](https://www.elastic.co/guide/en/elasticsearch/reference/current/mapping-timestamp-field.html).  I'll enable it and test, so for now, ignore the `date` command / field.

kylejohnson, 2016-02-11:
@knnniggett Also, about what @alager said - would it be more better to use the [mysql uuid() function](https://dev.mysql.com/doc/refman/5.5/en/miscellaneous-functions.html#function_uuid)?  One less perl module, you know.

knnniggett, 2016-02-11:
FreeBSD doesn't have os-release nor version under /etc. There does not appear to be an ascii file under /etc with the release name. It does appear with "uname -s", but that doesn't work for Linux distros.

The name os-release isn't a standard name either, which is why I was using *release. It'll pick up any distro that uses "lsb-release".

In FreeBSD, `uname -r` gives us the release, but in Linux that gives us the kernel.

If we collect the entire output from the following:
uname -a
cat /etc/*release
cat /etc/version

then I am reasonably confident we will get the information we need, in one or more of those three outputs, for most anything capable of running ZoneMinder. The trouble is parsing it, since we can't make any assumptions on where any particular bit of information will be. Guess we will have to do this the hard way. I was hoping to keep this simple, but it looks like I'll have to write something to intelligently parse the information before uploading.

I have not found a Perl library to retrieve this information either.

ISO 8601 does not mandate nanosecond precision, which is why I was asking. As one can see from the Wiki link you provided, there are many formats all considered ISO 8601 compliant. I've already created the datetime field in the script so we might as well leave it in there. 
I'm going to go with this for now:
```perl
DateTime->now( time_zone => "local" )->strftime("%Y-%m-%dT%H:%M:%S%z");
```
which looks like this: `2016-02-11T12:17:58-0600`

kylejohnson, 2016-02-12:
As a non-helpful reply, I'm just going to [leave this here](https://github.com/joeholden/distroscript/blob/master/distro).

knnniggett, 2016-02-12:
Initial distro determination logic has been pushed. 
I still need to work on the regex's. They work but are leaving in an extra quote character.

knnniggett, 2016-02-13:
@kylejohnson I'm getting HTTP 401 unauthorized when attempting to Post to 
https://38092a65757da3cb5f949b6b94aa8fc9.us-east-1.aws.found.io:9243/zmtelemetry/testing/

knnniggett, 2016-02-13:
Note that the telemetry branch has the following new package dependencies:
perl-DateTime
perl-Sys-CPU
perl-Sys-MemInfo
perl-IO-Socket-SSL

Note these are Redhat package names. I did verify the Debian equivalents exist and are available from the one of the main repos.

knnniggett, 2016-02-15:
@kylejohnson Making progress, but now the sever is responding that the request is malformed. http 400.
I'm using this endpoint: `https://zmanon:2b2d0b4skps@telemetry.zoneminder.com/zmtelemetry/testing`

And I'm sending the following in the message body:
`{"zm_version":"1.29.0","monitors":[{"MaxFPS":"1.00","Id":1,"Height":455,"Width":640,"Function":"None","Colours":1,"Type":"File","AlarmMaxFPS":"1.00","Name":"Monitor-1"},{"MaxFPS":"31.00","Id":2,"Height":720,"Width":1280,"Function":"None","Colours":4,"Type":"Ffmpeg","AlarmMaxFPS":"31.00","Name":"Foscam9821"},{"MaxFPS":"0.00","Id":3,"Height":240,"Width":320,"Function":"None","Colours":4,"Type":"Ffmpeg","AlarmMaxFPS":"0.00","Name":"winux"},{"MaxFPS":"1.00","Id":4,"Height":455,"Width":640,"Function":"None","Colours":1,"Type":"File","AlarmMaxFPS":"1.00","Name":"Monitor-4"},{"MaxFPS":"0.00","Id":5,"Height":240,"Width":320,"Function":"None","Colours":3,"Type":"Remote","AlarmMaxFPS":"0.00","Name":"TestMonitor"},{"MaxFPS":"0.00","Id":6,"Height":720,"Width":1280,"Function":"None","Colours":4,"Type":"Remote","AlarmMaxFPS":"0.00","Name":"Foscam9821_RTSP"},{"MaxFPS":"0.00","Id":7,"Height":480,"Width":640,"Function":"None","Colours":4,"Type":"Remote","AlarmMaxFPS":"0.00","Name":"Foscam8910"},{"MaxFPS":"31.00","Id":8,"Height":962,"Width":1280,"Function":"None","Colours":4,"Type":"Ffmpeg","AlarmMaxFPS":"31.00","Name":"Foscam9831"},{"MaxFPS":"31.00","Id":9,"Height":962,"Width":1280,"Function":"None","Colours":4,"Type":"Libvlc","AlarmMaxFPS":"31.00","Name":"Foscam9831_libvlc"}],"distro":"CentOS release","version":"6.7","ip":"12.250.218.126","event_count":3,"uuid":"1049d342-d3fd-11e5-95a1-0800275caefb","processor_count":2,"architecture":"x86_64","timestamp":"2016-02-15T10:05:49-0600","monitor_count":9,"kernel":"2.6.32-573.12.1.el6.x86_64","system_memory":1438748672}`

I know. It's hard to read, but I didn't want to modify the content so you can see exactly what I'm sending.

kylejohnson, 2016-02-15:
@knnniggett That does come out as valid json, hmm.  Here's what I'm doing, and as you can see from the response, it works:
```
[kjohnson@kyle-arch ~]$ curl -u zmanon:2b2d0b4skps -XPOST 'https://telemetry.zoneminder.com/zmtelemetry/testing/?pretty' -d '{ "monitors": 16, "events": 1000, "timestamp": "2016-02-09T13:26:46,915793160-0700" }'
{
  "_index" : "zmtelemetry",
  "_type" : "testing",
  "_id" : "AVLl3V18vIjaThyVTB55",
  "_version" : 1,
  "_shards" : {
    "total" : 2,
    "successful" : 1,
    "failed" : 0
  },
  "created" : true
}
[kjohnson@kyle-arch ~]$ 
```

Note that you don't need the `?pretty`

knnniggett, 2016-02-15:
@kylejohnson If I take your curl example and swap out the JSON message with the one I pasted previously, I get the same http 400 error:
```json
{
  "error" : {
    "root_cause" : [ {
      "type" : "mapper_parsing_exception",
      "reason" : "failed to parse [monitors]"
    } ],
    "type" : "mapper_parsing_exception",
    "reason" : "failed to parse [monitors]",
    "caused_by" : {
      "type" : "illegal_argument_exception",
      "reason" : "unknown property [MaxFPS]"
    }
  },
  "status" : 400
}
```
So it doesn't like that I added the MaxFSP field to the data set. I did this because, as I wrote zmtelemetry.pl, I realised that this gives us the ability to find out how many end users are placing values into this field for their IP cameras (and thus causing problems).

Here are the fields I am currently sending you from the Monitors table:
Id,Name,Type,Function,Width,Height,Colours,MaxFPS,AlarmMaxFPS

Any chance you could make this match on your end?

kylejohnson, 2016-02-15:
@knnniggett We can get around that by changing the `type` (`testing3`) to a new, unused one.  I'm no ES pro yet, but it looks like that when a type ($url/$document/$type) is created for a given document, it attempts to map the fields.  When you add new fields after a type's creation, it doesn't map them automatically.

```
curl -u zmanon:2b2d0b4skps -XPOST 'https://telemetry.zoneminder.com/zmtelemetry/testing5/?pretty' -d '{"zm_version":"1.29.0","monitors":[{"MaxFPS":"1.00","Id":1,"Height":455,"Width":640,"Function":"None","Colours":1,"Type":"File","AlarmMaxFPS":"1.00","Name":"Monitor-1"},{"MaxFPS":"31.00","Id":2,"Height":720,"Width":1280,"Function":"None","Colours":4,"Type":"Ffmpeg","AlarmMaxFPS":"31.00","Name":"Foscam9821"},{"MaxFPS":"0.00","Id":3,"Height":240,"Width":320,"Function":"None","Colours":4,"Type":"Ffmpeg","AlarmMaxFPS":"0.00","Name":"winux"},{"MaxFPS":"1.00","Id":4,"Height":455,"Width":640,"Function":"None","Colours":1,"Type":"File","AlarmMaxFPS":"1.00","Name":"Monitor-4"},{"MaxFPS":"0.00","Id":5,"Height":240,"Width":320,"Function":"None","Colours":3,"Type":"Remote","AlarmMaxFPS":"0.00","Name":"TestMonitor"},{"MaxFPS":"0.00","Id":6,"Height":720,"Width":1280,"Function":"None","Colours":4,"Type":"Remote","AlarmMaxFPS":"0.00","Name":"Foscam9821_RTSP"},{"MaxFPS":"0.00","Id":7,"Height":480,"Width":640,"Function":"None","Colours":4,"Type":"Remote","AlarmMaxFPS":"0.00","Name":"Foscam8910"},{"MaxFPS":"31.00","Id":8,"Height":962,"Width":1280,"Function":"None","Colours":4,"Type":"Ffmpeg","AlarmMaxFPS":"31.00","Name":"Foscam9831"},{"MaxFPS":"31.00","Id":9,"Height":962,"Width":1280,"Function":"None","Colours":4,"Type":"Libvlc","AlarmMaxFPS":"31.00","Name":"Foscam9831_libvlc"}],"distro":"CentOS release","version":"6.7","ip":"12.250.218.126","event_count":3,"uuid":"1049d342-d3fd-11e5-95a1-0800275caefb","processor_count":2,"architecture":"x86_64","timestamp":"2016-02-15T10:05:49-0600","monitor_count":9,"kernel":"2.6.32-573.12.1.el6.x86_64","system_memory":1438748672}'
{
  "_index" : "zmtelemetry",
  "_type" : "testing5",
  "_id" : "AVLmMXdWvIjaThyVTB6C",
  "_version" : 1,
  "_shards" : {
    "total" : 2,
    "successful" : 1,
    "failed" : 0
  },
  "created" : true
}
```

knnniggett, 2016-02-15:
Boom. Success!

```
2016-02-15 12:36:14.799020	zmtelemetry		5920	INF	Telemetry data uploaded successfully.	zmtelemetry.pl
```

The telemetry branch could use some additional testing from someone not running a Redhat distro.

knnniggett, 2016-06-24:
The Telemetry feature has been completed and is verified to work so I will close this one out.

closed 2016-06-24 by knnniggett