ZoneMinder/zoneminder #1231: Update Ubuntu Installation Package Building Guide

knnniggett, 2016-01-08:
This is a placeholder to update the Ubuntu Package Building Instructions to actually build a package:
https://github.com/pliablepixels/ZoneMinder/blob/8244496df1d3b0c8df30a6946930833665e8282f/docs/installationguide/ubuntu.rst#id3

We don't want to create any new procedures. Instead, we want our end users to perform the same steps the development team uses to build a package. For Ubuntu, that means we want to follow what @connortechnology currently does for the  packages in his ppa.

He uses the script found here to build a snapshot: `utils/do_debian_package.sh`
It is intended to be fully automated.

It uses the files found under `distros/ubuntu1204_cmake` .

@pliablepixels @Simpler1
Since I don't use Ubuntu, I ask for your participation

pliablepixels, 2016-01-16:
Okay, I'll work on this the coming week

knnniggett, 2016-01-18:
I created an Ubuntu 14.04 LTS vm and the following seemed to create a set of zoneminder deb files:
```
sudo apt-get install git pbuilder debhelper sphinx-common apache2-dev dh-linktree
git clone https://github.com/ZoneMinder/ZoneMinder.git
cd ZoneMinder/utils
./do_debian_package.sh trusty 3ebeaec binary master
sudo apt-get install {all the dependencies that debhelper complained were missing}
./do_debian_package.sh trusty 2016011701 binary master
```

I was a bit surprised that debhelper does not appear to have the ability to go fetch all the dependencies for you. Perhaps @connortechnology can confirm where or not such a feature exists in a debian environment.

The last step of the script failed because we are not @connortechnology and naturally cannot sign packages as him. In this case we don't want to sign the deb packages, so hopefully there is a way to get the script to skip this part.

@pliablepixels See if you can duplicate this on your end. Will need you to install the deb files and then followup with me what additional steps (if any) need to be performed to get things workings.

The goal is to not document things twice. Refer the reader to existing documentation, rather then duplicating it. 

Also, we want to do everything the same way so we aren't maintaining different sets of documentation that accomplish the same thing.

knnniggett, 2016-01-19:
@pliablepixels 
We made some changes to the build script to streamline things a bit better.

You should call the command like so now:
```bash
./do_debian_package.sh trusty `date +%Y%m%d`01 local master
```

knnniggett, 2016-01-19:
Rather than let the process fail the first time on missing dependencies, install "gdebi" and then try this command:
`gdebi ./debian/control`

pliablepixels, 2016-01-20:
in addition to the modifier do_debian script referred to in #1244, the instructions to build at https://gist.github.com/pliablepixels/0d11b493f77a7c85cd64 -- would request a look over before I convert it to an official doc

knnniggett, 2016-01-23:
@pliablepixels 
The gist looks good. Where is the best place for these instructions to live? We should put it in one and only one place, and then refer to it in other locations to avoid having to maintain multiple copies. 

This kind of script lives in the README file for the rpm packages I build. That would be my preference, but you tell me where it should go.

We need to find a better way to install the dependencies shown in step 2. From the documentation, it looks like the command `gdebi ./debian/control` should do this for the end user. Since the control file doesn't exist yet at that part in the instructions, adding this step to do_debian_package.sh, is probably the way to go.

We want to avoid having to maintain long lists of manually installed dependencies, which change over time and over different versions of ubuntu. That just gives us more work to do.

pliablepixels, 2016-01-24:
@knnniggett:

1) I'd recommend this build from package stays in ubuntu.rst since it is a ubuntu build instruction set (trusty/vivid) - I did a PR for that https://github.com/ZoneMinder/ZoneMinder/pull/1248 - looks like my previous PR was on my fork not on ZM

2) The instructions at https://github.com/ZoneMinder/ZoneMinder/blob/master/README.md for package are not correct anymore (at least  for ubuntu) - I'd recommend they be deleted - I really don't think the main README should be so big 

3) I'd recommend we follow the same process - for each of the ".rst" files, debian, fedora etc - we add a build from package just like we did for ubuntu and link to there from the main README and wherever else we need

PR #1248 (2016-02-22): updated documentation for ubuntu
Resolves #1231
commit 0ef6d55e (2016-02-22): resolves #1231 - ubuntu package from source instructions added
closed 2016-02-22 by knnniggett