ZoneMinder/zoneminder #1245: Suggestion: Montage fix

digdilem, 2016-01-22:
(I'm not used to collaborating and don't know Git, so not doing this myself, but welcome anyone else to take it forward if they think it's a good idea)

Since day one, Zoneminder has had an issue with displaying multiple camera feeds using its montage system. This is caused by limits in browsers limiting the number of concurrent connections to the same server. Firefox can be hacked to adjust this figure, but chrome cannot - and chrome is now the most popular browser, so montages are broken for more than a few cameras.

I have built my own montages interfaces (non-public, sorry, part of a corporate system) which has overcome this using javascript.

Suggestion:

Add an optional (default after testing) montage display type: mjpeg or javascript.

If javascript, then html montage page would be generated (see examples below) which runs a js script (such as below) that reloads all the image tags once a second (or variable in ms) on that page.

The images are STATIC jpgs (mode=single) rather than mjpeg, so the connections are able to close. 

I have tested this with complete satisfaction with 80 concurrent cameras on screen at once. (Network and CPU of my client computer were both suffering with this, though!)  

Server: Zoneminder 1.25 on Debian Wheezy. 
Client: Multiple windows clients (win7 and 8.1) running both Firefox and Chrome (latest versions)

Example layout (taken from my program)
http://i.imgur.com/SOvyi6f.jpg

Example html for montage (wrapped in table cell or framework)

[image]

(Obviously change monitor, scale etc to fit. I've added a link to open a bootstrap modal fullscreen for any camera clicked)

Actual sample code which reloads ALL img tags on the page, adding a random string at the end to bypass caching:

<script>
  setInterval(function() {
      var images = document.images;
      for (var i=0; i<images.length; i++) {
          images[i].src = images[i].src.replace(/\btime=[^&]*/, 'time=' + new Date().getTime());
      }
  }, 1000); 
  </script>

digdilem, 2016-01-22:
Github's stripped my code, so attaching it here.

[montage_fix.txt](https://github.com/ZoneMinder/ZoneMinder/files/100624/montage_fix.txt)

Linwood-F, 2016-01-22:
I started working on something similar, and got it working well enough for me.  I baked in a review function so you could scrub back and forth.  I also had it so you could change individual window size by camera, making some larger and smaller, on the fly.   I did not pursue getting it integrated more fully as I was unsure where the whole video issue would land, and whether it would make still-based review obsolete, so I never tried to get it merged. 

But I do think it's a big gap in functionality that the mainline code doesn't support most browsers.

I think the final version (which I'm pretty sure I pushed) switched from a static path to load to the image server for more compatibility, but still not sure if that would work with the video implementation.

If there are any of the video branch folks around, I wonder if they have insight into that aspect of this sort of fix? 

Or whether the group has developed a roadmap for how they plan to address the browser issue, as I know this technique was not exactly endorsed as the solution some months ago.

knnniggett, 2016-01-22:
This is a duplicate of #441 

@digdilem Thank you taking the time to document your idea. This has been an issue with ZoneMinder for a long time, and we already have an open issue tracking it. We've had internal conversations on this, and you can expect a a true fix for this sometime in the future. I'd expect the fix to be transparent, and to not force the user to compromise stream quality or switch to snapshot mode. As to when this will happen I cannot say. Don't mistake this for lack of desire. We all want this fixed, but we lack the manpower to fix everything all at once.

While the code you've presented does make a good workaround, it does not address the root cause and consequently cannot be used by us as a true fix. This would make a good candidate for the zoneminder wiki, if you are willing.

closed 2016-01-22 by knnniggett

connortechnology, 2016-01-22:
I have a lot of changes to make to montage when we finally release 1.29, and some of this could definitely be integrated.  I think it would be nice to be able to specify a framerate for watching, and I think if we detect chrome, we could have a javascript function go through and refresh the images. Would certainly help a lot until we get a proper fix. I will certainly look at this for 1.30, because I honestly don't know how long it will take to this properly fixed.

In terms of video branch, if you are requesting jpg's from zms, zms will give them to you, regardless of the storage backend.

Linwood-F, 2016-01-22:
@connortechnology, thanks, that's good to know, but is that actually possible historically?  Not talking about real time display but review.  If you are (I'm making this up) storing video as 5 minute segments, and I asked zms for a frame at the 2.5 minute mark, with key-frames every 25 -- how fast is it going to be to retrieve that?

connortechnology, 2016-01-22:
I never said it would be fast.

Historically, you can already ask it for jpgs or an mpeg stream, or whatever you want.  It might be processor intensive though.

Linwood-F, 2016-01-22:
I guess my point is anyone doing this for historical review (i.e. to have multiple cameras without the stream restriction) may have a killer issue down the line with the video stream.  May not.  

Maybe I'm just a worrier.

Is the video branch working reasonably reliably?   I'd be curious to pull it down and try it with a scrubbing-type review against video.

digdilem, 2016-01-22:
Thanks all for considering it. Apologies for duplicating, however that
stalled back in March so if it's brought it to the fore again, I'm not that
sorry...

Ref conflict with upcoming video changes: Presumably zms will continue
providing incoming streams, so however they are stored won't affect it -
unless you meant using ffmpeg or similar to merge multiple feeds into one
video stream, or levering html5 somehow. However, this is a relatively
quick fix that WILL work.

This montage is a common problem in the irc channel, and has been in the
forums for all the years I've been helping support zm (somewhere around 8
years now), so I'm glad it's still on people's radar. :)