ZoneMinder/zoneminder #1246: installation error on raspberry jessie

unbroken75, 2016-01-22:
I followed the instruction to build and install zoneminder on a raspberry with raspian jessie distribution from here:
http://zoneminder.readthedocs.org/en/stable/installationguide/debian.html
but the installation fails with:
```
root@raspberrypi02:/home/pi# dpkg -i zoneminder_1.28.0-wheezy_armhf.deb
Selezionato il pacchetto zoneminder non precedentemente selezionato.
(Lettura del database... 127721 file e directory attualmente installati.)
Preparativi per estrarre zoneminder_1.28.0-wheezy_armhf.deb...
Estrazione di zoneminder (1.28.0-wheezy)...
Configurazione di zoneminder (1.28.0-wheezy)...

Database already at version 1.29.0, update aborted.

● zoneminder.service - LSB: Control ZoneMinder as a Service
   Loaded: loaded (/etc/init.d/zoneminder)
   Active: inactive (dead)

gen 22 11:02:37 raspberrypi02 systemd[1]: Stopped LSB: Control ZoneMinder as a Service.
gen 22 11:02:46 raspberrypi02 systemd[1]: Starting LSB: Control ZoneMinder as a Service...
gen 22 11:02:53 raspberrypi02 zmpkg[3261]: INF [Sanity checking States table...]
gen 22 11:02:58 raspberrypi02 zmpkg[3261]: INF [Command: start]
gen 22 11:02:58 raspberrypi02 pkexec[3271]: pam_unix(polkit-1:session): session opened for user root by (uid=0)
gen 22 11:04:14 raspberrypi02 zoneminder[3256]: Starting ZoneMinder: Job for zoneminder.service canceled.
gen 22 11:04:14 raspberrypi02 zoneminder[3256]: success
gen 22 11:04:14 raspberrypi02 systemd[1]: Stopping LSB: Control ZoneMinder as a Service...
gen 22 11:04:14 raspberrypi02 systemd[1]: Stopped LSB: Control ZoneMinder as a Service.
gen 22 11:39:12 raspberrypi02 systemd[1]: Stopped LSB: Control ZoneMinder as a Service.
invoke-rc.d: initscript zoneminder, action "status" failed.
Job for zoneminder.service failed. See 'systemctl status zoneminder.service' and 'journalctl -xn' for details.
invoke-rc.d: initscript zoneminder, action "start" failed.
dpkg: errore nell'elaborare il pacchetto zoneminder (--install):
 il sottoprocesso installato script di post-installation ha restituito lo stato di errore 1
Elaborazione dei trigger per man-db (2.7.0.2-5)...
Elaborazione dei trigger per systemd (215-17+deb8u2)...
Si sono verificati degli errori nell'elaborazione:
 zoneminder
```
```
root@raspberrypi02:/home/pi# systemctl -l status zoneminder.service
● zoneminder.service - LSB: Control ZoneMinder as a Service
   Loaded: loaded (/etc/init.d/zoneminder)
   Active: failed (Result: timeout) since ven 2016-01-22 11:44:31 CET; 5min ago

gen 22 11:39:35 raspberrypi02 zmpkg[24192]: INF [Sanity checking States table...]
gen 22 11:39:35 raspberrypi02 zmpkg[24192]: INF [Command: start]
gen 22 11:39:36 raspberrypi02 pkexec[24202]: pam_unix(polkit-1:session): session opened for user root by (uid=0)
gen 22 11:44:31 raspberrypi02 systemd[1]: zoneminder.service start operation timed out. Terminating.
gen 22 11:44:31 raspberrypi02 systemd[1]: Failed to start LSB: Control ZoneMinder as a Service.
gen 22 11:44:31 raspberrypi02 zoneminder[24187]: Starting ZoneMinder: Job for zoneminder.service failed. See 'systemctl status zoneminder.service' and 'journalctl -xn' for details.
gen 22 11:44:31 raspberrypi02 systemd[1]: Unit zoneminder.service entered failed state.

```

connortechnology, 2016-01-22:
Why are you installing the wheezy package on jessie?  There is a jessie package for this. Plus it looks like you are installing an older version(1.28.0), than what comes with jessie (1.29)

knnniggett, 2016-01-22:
This is exactly the kind of "how to install" conversation that should happen in the user forum, rather than our github forum.

https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2016-01-22 by knnniggett