ZoneMinder/zoneminder #1253: Provide an ability to quit zms for a specific connkey

pliablepixels, 2016-01-29:
ZMS is written to implement a long running TCP connection continuously streaming images. The only way to quit the connection is to "kill the browser window". While this is fine for a desktop browser, it makes it almost impossible to release zms resources in single page mobile apps (like zmNinja or any other app in the future). Any web application that uses ``img src`` to display images does not have control on the stream and cannot terminate the TCP object on its own which makes it exceeding hard to remove 'zms' processes even when nothing is being viewed.

There is an option to send a "Pause" and "Stop" control command today - that pauses the image streaming but the zms process still continues in memory and headers are still exchanged.

Proposal: We introduce a new CMD_QUIT to zm_event.cpp and zm_monitor.cpp
that basically does this:

```
case CMD_QUIT :
        {
            Debug( 1, "User initiated stream termination" );
            exit(-1);

        }
```
 
I haven't tested this - but I'm going to do that in the next day or two. @connortechnology @knnniggett  - do you have any issues with this approach?

knnniggett, 2016-01-29:
I'm not sure we should return -1 when we exit, but I don't have a problem with this.

connortechnology, 2016-01-29:
I'm cool with it.

pliablepixels, 2016-01-29:
great. Yes, I'm doing exit(0). I'll test some more and PR

PR #1254 (2016-01-29): 1253 zms quit cmd
Resolves #1253
commit 96db6d5f (2016-01-29): Merge pull request #1254 from pliablepixels/1253-zms-quit-cmd

1253 zms quit cmd.  Fixes #1253
closed 2016-01-29 by kylejohnson