ZoneMinder/zoneminder #1256: getStreamCmdResponse stream error: socket_sendto( /var/run/zm/zms-786779s.sock ) failed: No such file or directory - checkStreamForErrors()

matak, 2016-01-31:
can you help me, what means this bug?

knnniggett, 2016-01-31:
That is not a bug but a misconfiguration. 
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Please take this to the user forum. If you want to get a good answer to your query, you will need to describe your system in detail, what distro, how you installed zoneminder, the contents of your zoneminder apache config file, and the value of PATH_ZMS under options.

closed 2016-01-31 by knnniggett

matak, 2016-01-31:
i tried to go through all forum topics and all googled pages, but no answer anywhere i have this problem only after opening new event (every ten minutes)

knnniggett, 2016-01-31:
Well, you must have missed the two user forum topics I just answered last week on the same topic.
I don't know what you were Googling, but the infamous "socket_sendto" errors are one of the most common types of problems reported when configuring ZoneMinder. We get asked this all the time, and it is definitely not a bug. 

We will be glad to help you, just not in our github forum as it is reserved for bug reports and feature requests only. 

Again, please read our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Then go over to the user forum and create a new post, complete with the information you were asked to provide. 

This thread is being locked as it is not productive towards advancing the source code.