ZoneMinder/zoneminder #1259: Bump master to 1.30.0

knnniggett, 2016-02-04:
Looking for feedback from @connortechnology @kylejohnson @SteveGilvarry 

Are we ready to do this?
I thought I'd ask before I just did it.

connortechnology, 2016-02-04:
Well, this brings up the issue of how we structure our versions.  Prior to 1.29, we incremented from 1.28 with each db change.  

I kinda agree with immediately bumping, leaving all 1.29.* increments for point releases.

So 1.30.* will be development versions, never formally released, and our next stable release would be 1.31.  Is that what we are thinking?

kylejohnson, 2016-02-04:
No good reason, but I don't like it.  Couldn't 1.30 be a development version up until we release, and then bump the version number?

I'm going to take this thread another direction:

This is probably worth a read: https://www.kernel.org/doc/Documentation/development-process/2.Process  I went through it earlier this week, and need to do so again.

So, if we want to get to a monthly release cadence, what does that look like?
Perhaps:
1st: Release, open merge window
14th: close merge window
14th - 30th: testing
1st: Release, open merge window

This means that the things merged in the next release are being developed in the previous release.

SteveGilvarry, 2016-02-04:
I would hold off on that as we may have fixes for 1.29 yet. Did we ever come up with a good option for master builds to increment without needing a release? I don't think the next sql file should be 1.29.1 for instance as it means the db gets a minor version bump for every PR with a sql change.
I would like to see a branch where we can work on stretching the goals, something like my playpen here https://github.com/SteveGilvarry/ZoneMinder/tree/ZoneMinder2_The_Quickening which is all the things I want to see in the next major release, the name is a joke, but I think these features are worth a 2.0 jump. Which features? Storage areas, plugins, feature-h264-videostorage, new gui. It may not work, but we can all work together to deliver a set goal. I am all for merging these to master, but the expectation that master is a working solution is already set so maybe this needs to be develop?
Some sort of roadmap would be nice, firstly to let people know where we are heading, control our own scope creep and hold ourselves accountable to deliver some features in timeframes (Within our limited volunteered time allowances).

connortechnology, 2016-02-04:
I guess it comes down to: Do we want to be able to do point releases, like 1.29.1.  I think we should, as there may be minor bugs/issues that we want to fix without requiring people to upgrade to 1.30 or a development build.

I think major work should go on branches, but few of us have the time to follow a bunch of experimental branches, so I agree with @SteveGilvarry that more experimental branch than master would be useful.  I would use it for example.  It would need some in charge of it, pulling from each of the dev branches.

connortechnology, 2016-02-04:
So I think we should bump to 1.30 NOW, saving space for point releases, with the third set of digits reserved for db/major feature additions.  Then 1.31 is the stable release.

kylejohnson, 2016-02-04:
> but I think these features are worth a 2.0 jump
I brought this up on Monday during our weekly scrum.  [Semantic versioning](http://semver.org/) says that major releases only happen when a backwards-incompatible change happens.  A new GUI which completely relies on the API may be backwards incompatible.  Removing the XML skin might also be backwards incompatible.  Adding new features such as those mentioned should be backwards compatible.  That is, what people were doing yesterday, they can continue to do.

That said, we don't have to stick with semver, but it is what we've tried to loosely follow in the past.

> but the expectation that master is a working solution is already set
Indeed, we've gotta keep master mostly stable.

> Some sort of roadmap would be nice, firstly to let people know where we are heading, control our own scope creep and hold ourselves accountable to deliver some features in timeframes (Within our limited volunteered time allowances). 

Bingo bongo.  I harped on this during our scrum as well.  What I did during the 1.29 release was create a [1.30 milestone](https://github.com/ZoneMinder/ZoneMinder/milestones/v1.30.0) and start tagging issues there, only issues which I though we could realistically have released within 1 month.  That's a /loose/ roadmap, and we could move forward creating 1.31, 1.32, ... milestones.

kylejohnson, 2016-02-04:
> I think major work should go on branches, 

Me too, they have to.  We can't be developing directly off of master.  I thought we had the branching strategy solved a long time ago?

connortechnology, 2016-02-04:
I was more leading up to the point that very few people can follow dev branches, so we need someone to constantly pull into an experimental branch that people CAN follow.

mnoorenberghe, 2017-03-18:
I guess this can be closed since master is 1.30.2?

closed 2017-03-18 by SteveGilvarry