ZoneMinder/zoneminder #1261: mod_ssl vs mod_nss

fwittekind4, 2016-02-05:
mod_ssl requirement was added to 1.29, which doesn't co-exist very well with a already installed and configured mod_nss

knnniggett, 2016-02-05:
We can't respond to this until you state your distro and how you installed zoneminder.

fwittekind4, 2016-02-05:
Sorry.. 
CentOS 7

package installed from yum repo: 
http://zmrepo.metroeasthomevetcare.com/el/7/x86_64/

Zoneminder server doubles as IPA server which requires mod_nss

knnniggett, 2016-02-05:
It has never been possible to write an Apache configuration file for ZoneMinder that will work for everyone's needs. What we can do, however, is establish a default policy of more security rather than less. Similar to how Redhat's policy is to ship with the firewall on, I am now packaging all rpm's in zmrepo with HTTPS enabled and have included an auto-redirect from HTTP -> HTTPS. 

Since this likely won't work for everyone, the README included with the ZoneMinder 1.29.0 rpm for CentOS 7 refers the reader to the file README.https. So my first question is, "Did you read that"? Because the contents of that file describe how you can easily turn off ssl, which leads me to my second question. What is stopping you from turning it off?

This change only needs to be done once, future upgrades of the ZoneMinder package will not overwrite this.

fwittekind4, 2016-02-06:
It's also Red Hat policy not to break existing configurations, right now, I can't run "yum update" without breaking my system, and I can't install the new version of the zoneminder package at all, because, I can't have mod_ssl installed.  I have an existing https config, with existing certificates.  I'm not aware of existing Red Hat packages that try to automaticly configure https as part of a package install, even IPA doesn't do this, even though it requires https to be used, and requires it's own certificates to be used.

knnniggett, 2016-02-06:
Well, what you said in your first post is "doesn't work well together", which leaves us to wonder what the hell you mean. I doubt anyone who read that would connect that to "can't yum update without breaking my system", which is what you changed your story to in your second post.

If you really want us to help you, I suggest you provide us real data, rather than whine things aren't going your way, so we have a chance to find a good solution.

I suggest you describe in detail why having mod_ssl on your system causes problems.
I have not found any documentation stating they can't coexist on the same system.

fwittekind4, 2016-02-06:
# yum install mod_ssl
Restart httpd service, it fails to load.

From error log:
Fri Feb 05 20:07:36.742756 2016] [core:crit] [pid 27352] (22)Invalid argument: AH00069: make_sock: for address [::]:443, apr_socket_opt_set: (IPV6_V6ONLY)
(98)Address already in use: AH00072: make_sock: could not bind to address [::]:443
(98)Address already in use: AH00072: make_sock: could not bind to address 0.0.0.0:443
[Fri Feb 05 20:07:36.742905 2016] [mpm_prefork:alert] [pid 27352] no listening sockets available, shutting down

# rpm --erase mod_ssl
And httpd starts just fine.

That breaks the rules that governs Red Hat package updates, and EPEL package updates (I realize you repo is not governed by these same rules, but you did bring up Red Hat policy regarding updates)

Yes, I know it is technically possible to make them coexist by putting mod_ssl on a different port number than mod_nss, but, I really have no desire to have httpd listening on a port I don't need, just so I can have a package installed that I don't need or want.

I have no issue with secure by default. 
I have no issue with http->https redirect by default.
Pulling in the mod_ssl dependency in the package is a problem for me.

Yes, yum install mod_ssl isn't the same as yum update zoneminder, but, it's easier to undo, and since the yum update zoneminder pulls in the mod_ssl package, the result is the same.

connortechnology, 2016-02-06:
So it sounds like mod_ssl and mod_nss provide the same thing and hence conflict.  

I'm the debian/ubuntu guy, so I have to assume that in the package definitions, there is an option to say either mod_ssl or mod_nss.  There is in the debian package description.... I assume that rpm can do the same thing.

I've actually never heard of this mod_nss thing before.  Google has been enlightening. 

This is a pretty fringe case.  Hopefully we can make it all work well though, now that it has been brought to light.

fwittekind4, 2016-02-06:
There are alternatives with rpm as well, similar to how systemd vs sysvinit packages where done in the past, just break the package into multiple rpm files, generated from one srpm.  It is also possible to implement with strictly Apache config, using IfModule directives, assuming the mod_ssl/mod_nss are not pulled in as rpm dependencies.

knnniggett, 2016-02-06:
I'm still reading about this, but the issue seems to be they both want to bind to port 443. Makes sense.
So for the short term, configure mod_ssl to listen to a different port as discussed here:
http://osdir.com/ml/freeipa-devel/2013-10/msg00297.html

Yes, spawning multiple packages is something I am already considering.

connortechnology, 2016-02-06:
Mmm... see in .deb land, you just say mod_ssl | mod_nss .... and leave the config to the user. :)

Still, I like that he can just remove mod_ssl.. doing so in .deb land would remove zm too. 

packaging is complex, mmkay.

fwittekind4, 2016-02-06:
removing mod_ssl would require removing zoneminder in rpm land too.

knnniggett, 2016-02-06:
In rpm-land one cannot do an "or" condition on the package dependencies, like one can do in Debian-land, unfortunately. Rpm does have the ability to remove a package without removing the dependencies with the --nodeps flag. Comes in handy every now and then.

By default mod_nss is configured to run on port 8443, which means it can coexist with mod_ssl right out of the box. So I'm calling B.S. on any claims of RedHat packaging rule violations. This problem was caused by the op's choice to change the default port that mod_nss runs on. I'm not saying I'm not going to help, but I am going to call it what it is.

Long term, we've talked about splitting zoneminder into multiple packages, which I expect to do eventually. However, I think a quicker solution would be to remove the mod_ssl dependency, but leave the SSLRequires directives in the default apache config file. I would then add an extra step to the README, instructing the end user to either a) install mod_ssl or b) comment out the SSLRequires directives.

connortechnology, 2016-02-06:
I don't envy you your job.  I'm happy that it is left in capable hands though.

fwittekind4, 2016-02-06:
SELinux policy changes required as well for changing mod_ssl port number to something other than 443, as SELinux limits port numbers Apache is allowed to bind.  Personally, I think I'll patch the rpm myself vs have mod_ssl installed.   setenforce 0 is not a option I am willing to consider, nor is --nodeps.    As for as the op's choice to change the default configuration of mod_nss, I didn't actually do this myself, it was done past of running the installer for IPA per RedHat documentation.

knnniggett, 2016-02-06:
The --nodeps comment was directed at @connortechnology 
It was not intended as a suggestion for you to try.

There is no working selinux policy for ZoneMinder on CentOS 7 I am aware of, so unless you've written one yourself, ZoneMinder will not work properly with it on. Contributions are welcome if you want to do the work to make one.

While I will consider modify the package dependencies like I mentioned previously, I would recommend you run ZoneMinder in a different environment than the one you are using. With the mod_nss thing, and now the mention of running with selinux on, I am not at all certain you not run into any other problems. This ranks up there in terms of annoying difficulties as trying to get ZoneMinder and Asterisks to play nice together.

fwittekind4, 2016-02-16:
After a bit of research, I determined it is possible, although more complex than what I hear of the Debian way, to set a dependency on one of two packages, i.e. mod_ssl or mod_nss.  Borrowed some of the idea from clamav spec file from EPEL 7.

Requires:       https(zoneminder)
%package ssl
Summary:        Zoneminder with mod_ssl
Group:          Applications/File
Requires:       mod_ssl
Requires:       zoneminder = %version-%release
Provides:               https(zoneminder) = mod_ssl
Conflicts:              https(zoneminder) < mod_ssl
Conflicts:              https(zoneminder) > mod_ssl
BuildArch:              noarch
%package nss
Summary:        Zoneminder with mod_nss
Group:          Applications/File
Requires:       mod_nss
Requires:       zoneminder = %version-%release
Provides:       https(zoneminder) = mod_nss
Conflicts:      https(zoneminder) < mod_nss
Conflicts:      https(zoneminder) > mod_nss
BuildArch:              noarch

Add %description and %files sections those two sub-packages, and it defaults to install mod_ssl version, but, gives admin the option to go the mod_nss route without breaking dependency resolution.

knnniggett, 2016-03-22:
I'm closing this out as the rpm specfiles have been modified to no longer require mod_ssl. The sslrequires statements are still in the apache config file, and a step in the README has been added to direct the user to either a) disable the sslrequires or b) install mod_ssl.

As mentioned previously, I already know how to build zoneminder into multiple packages, but I am not willing to do so at this time. This will happen sometime in the future (due to other reasons).

closed 2016-03-22 by knnniggett