ZoneMinder/zoneminder #1262: Track zmtelemetry

knnniggett, 2016-02-06:
This issue has been created to track the new zmtelemetry branch.

This branch adds a new Perl daemon, zmtelemetry.pl, which uploads usage information to a specific telemetry server. Updates occur every two weeks, and can be controlled from Options by setting the ZM_TELEMETRY_DATA flag. This runs similar to the zoneminder version checker.

The intent is to do this as transparently as possible to assure our end users this data is will be used solely to make a better product.

I've created provisions to upload the output of arbitrary systems calls, and specified variables from the Config table.

Initial framework is done. 
@kylejohnson to followup with
- A list of what Config variables, system calls, or other data we want to collect
- format required by the destination server (hoping this won't require any weird or otherwise hard to find Perl modules)

pliablepixels, 2016-02-06:
I think it might be useful to submit the data as a stringified JSON feed. One potentially useful goal of that zmtelemetry daemon is also to anonymously submit camera information, removing IPs and login information and associated access protocols so it can be put back into zm's add monitor function

connortechnology, 2016-02-07:
Agreed, however currently there isn't a way for people to tell ZM what their camera is.  We need to add Manufacturer/Model fields/input to the monitor page.  Maybe get that out really soon in a 1.29.1 release.

pliablepixels, 2016-02-07:
I guess this is a good time for me to leverage jekyll's _data directory that @kylejohnson talked about.
I threw something together - data is dummy now

http://pliablepixels.github.io/zmcameradb/

The data file that is parse is pretty simple https://github.com/pliablepixels/zmcameradb/blob/gh-pages/_data/cameradb.yml

I'm thinking the primary goal is to first help users identify which URL and connection type to use.

This exports to JSON too - so in future its easy for ZM to consume it.

kylejohnson, 2016-02-09:
Hm, is this a duplicate of #1229 ?

knnniggett, 2016-02-09:
You're absolutely, right. I'm going to close this out. We should continue the conversation over at #1229.

closed 2016-02-09 by knnniggett

kylejohnson, 2016-02-09:
Looks like commit 8a537029ea7a5bd4c91294fd458694393785812f brought the initial daemon into the src.

kylejohnson, 2016-02-09:
@knnniggett Could you change the port from 80 to 443?

knnniggett, 2016-02-09:
Why no, I can't because you already did. :-)