ZoneMinder/zoneminder #1263: jquery 1.7.2 is too old

onlyjob, 2016-02-08:
[web/skins/classic/js/jquery-1.7.2.js](../blob/master/web/skins/classic/js/jquery-1.7.2.js) is too old. Please update to 1.11.3 or above.

knnniggett, 2016-02-08:
Posting the following as a reminder to self that this might not be as easy as replacing the js file with a newer version: https://github.com/ZoneMinder/ZoneMinder/issues/785#issuecomment-95772526

onlyjob, 2016-02-08:
JQuery upgrade guide https://jquery.com/upgrade-guide/1.9/ may be useful...

connortechnology, 2016-02-08:
It probably is that easy.  We don't do much with it.

I'll pull it into my home system and see if anything breaks.

onlyjob, 2016-02-08:
No, @connortechnology, please no minification. How many times we have to go through this?
Minified files are effectively pre-compiled binaries, un-modifiable, unreadable and non-distributable in Debian (i.e. non-free). It is a bad practice to commit pre-built files to Vcs.
Moreover there are even more concerns, including security: https://wiki.debian.org/onlyjob/no-minification

Please always commit uncompressed files. If you believe in minification than you can compress on build time.

connortechnology, 2016-02-08:
Does debian not already package jquery?  Perhaps debian should use the debian package version as you do mootools?

onlyjob, 2016-02-08:
Yes we can use "libjs-jquery" (v1.11.3 at the moment) but that's not the point. It is the best to have no pre-minified files committed. Otherwise there will be maintenance overhead, repacking tarball to throw away non-free content etc. Really it is not worth it: there are serious cons and no pros...
Besides mootools-1.5.1 may be not safe to replace with mootools-1.4.5...

commit 46efe9c1 (2016-02-18): Merge pull request #1298 from onlyjob/jquery

Upgrade bundled jQuery (Closes: #1263)
commit 632e9c8e (2016-02-18): Upgrade bundled jQuery (Closes: #1263)
closed 2016-02-18 by connortechnology