ZoneMinder/zoneminder #1264: zoneminder only installs polkit rules compatible with version >= 0.112

onlyjob, 2016-02-08:
As reported in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=808836

ZoneMinder installs a polkit rule file that only works for polkit >= 0.112
(Even Debian "unstable" have only 0.105 at the moment).

Please consider installing rules compatible with older PolicyKit-1.

knnniggett, 2016-02-08:
Wow, you mean CentOS 7 has something newer than what is in the latest Debian unstable?
I never expected this to happen. Ever. This is refreshing. Hmmmm, I suppose at this time I should moan that those guys at Debian should get their act together and start using newer libraries. :-)

I can look into it, but I may need someone else to test since I apparently don't have an environment with polkit 0.105.

It would have been nice if the op from that thread would have posted an error message or some other data, which would allow one to zero in on the exactly what keywords in the policy file are causing the issue.

onlyjob, 2016-02-08:
Well sometimes they have to take care of transitions which may involve update/rebuild of many depending libraries which takes time and effort and may be not quick/easy... 
Polkit-0.113 is in "experimental" which means that transition is on its way...
Meanwhile we need to think about users of Jessie and "testing"...

This bug was reported against 1.28.1 but 1.29.0 is the same.

I'm also not sure what kind of issues to expect or how to reproduce the problem...

thinrope, 2017-08-30:
This is a bit old, but...

Why is polkit required at all? Looking through the code, it seems to be only used for restarting the daemon (zmsystemctl.pl), right? If that is not required (e.g. use sudo) then lowering the dependencies may be a good thing.

closed 2018-06-22 by stale[bot]