ZoneMinder/zoneminder #1265: 1.29.0: FTBFS with GCC-6 "flexible array member"

onlyjob, 2016-02-08:
As reported in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=812026
ZoneMinder-1.29.0 FTBFS when built by GCC-6:

~~~~
[ 54%] Building CXX object src/CMakeFiles/zm.dir/zm_rtp_data.cpp.o 
cd /build/zoneminder-1.29.0+dfsg/dbuild/src && /usr/lib/ccache/g++-6   -DHAVE_SENDFILE4_SUPPORT=1 -I/build/zoneminder-1.29.0+dfsg/dbuild/src -I/build/zoneminder-1.29.0+dfsg/src -I/build/zoneminder-1.29.0+dfsg/dbuild -I/usr/include/x86_64-linux-gnu  -g -O2 -fPIE -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2  -Wall -D__STDC_CONSTANT_MACROS -O2   -o CMakeFiles/zm.dir/zm_rtp_data.cpp.o -c /build/zoneminder-1.29.0+dfsg/src/zm_rtp_data.cpp                                         
In file included from /build/zoneminder-1.29.0+dfsg/src/zm_rtp_ctrl.cpp:24:0: 
/build/zoneminder-1.29.0+dfsg/src/zm_rtp_ctrl.h:128:31: error: flexible array member 'RtpCtrlThread::RtcpPacket::<anonymous union>::Bye::srcN' in an otherwise empty 'struct RtpCtrlThread::RtcpPacket::<anonymous union>::Bye'                                                                                                                             
                 uint32_t srcN[]     // list of sources 
                               ^ 
/build/zoneminder-1.29.0+dfsg/src/zm_rtp_ctrl.h:126:20: note: in the definition of 'struct RtpCtrlThread::RtcpPacket::<anonymous union>::Bye' 
             struct Bye 
                    ^~~ 
src/CMakeFiles/zm.dir/build.make:713: recipe for target 'src/CMakeFiles/zm.dir/zm_rtp_ctrl.cpp.o' failed 
~~~~

onlyjob, 2016-07-03:
GCC-6 will be default in next Debian release so recently this issue had its severity bumped...

knnniggett, 2016-07-03:
We've already fixed this. Fedora 24 ships with gcc 6 and it builts... let me find you the commit hash

knnniggett, 2016-07-03:
Here it is: see #1491 
I've already had to backport this as a patch for zoneminder 1.29.0 on Fedora 24

onlyjob, 2016-07-03:
Awesome. :)  Thank you very much.

closed 2016-07-03 by knnniggett