ZoneMinder/zoneminder #1272: non-free onvif/wsdl files?

onlyjob, 2016-02-08:
There is a bunch of oddly licensed files in [onvif/wsdl](../tree/master/onvif/wsdl) where I counted 7 _(seven!)_ different new licenses.
Apparently most of those files appear to be non-free as their licenses prohibit modification:

> The Authors grant permission to copy and distribute [...] No other rights are granted by implication, estoppel or otherwise.

  * wsdl.xsd
  * wsdl11soap12.xsd

> No license is granted to modify this document.

  * devicemgmt.wsdl
  * media.wsdl
  * onvif-local.xsd
  * onvif.xsd
  * ptz.wsdl

> However, this document itself may not be modified in any way

  * b-2.xsd
  * bf-2.xsd
  * t-1.xsd
  * wsdd-discovery-1.1-schema-os.xsd
  * wsdd-discovery-1.1-wsdl-os.wsdl

---

Can somebody please confirm if those files are really not used?
It looks like we will have to drop the whole directory...

knnniggett, 2016-02-08:
Github must be lagging today. Related to PR #1273 

So I guess we need to answer the licensing questions first.

onlyjob, 2016-02-08:
> Github must be lagging today

Maybe it was just me working at 5am. ;)

onlyjob, 2016-02-08:
After examining contents of [onvif](../tree/master/onvif) folder I'm not satisfied with lack of licensing and copyright attribution. There is no documentation whatsoever, not even regarding origin(s) of ONVIF or its authors. Does ONVIF have a home page? Where is it maintained (if it _is_ maintained)? How did it get committed without licensing information/review? What is the license for ONVIF files? Where is copyright information?
Such matters must not be neglected.

@altaroca?

I'd regret dropping entire ONVIF component from official Debian packaging but I see no other option unless ONVIF licensing concerns are addressed. I could not find anything relevant in the internet...

knnniggett, 2016-02-09:
You can read about Onvif support in @altaroca 's blog here:
https://altaroca.wordpress.com/the-onvif-series/

Specifically: 
https://altaroca.wordpress.com/2014/07/14/onvif-probing-for-zoneminder/

I don't think any of us had even thought about licensing, but your right if we are going to treat this project as something more than a hobby then everyone involved needs to come to an agreement.

onlyjob, 2016-02-09:
A had a quick look there but could not find a link to repository (or release page) nor anything related to licensing...

knnniggett, 2016-02-09:
Yes, that's correct. There is nothing there related to licensing, only a blog on how to use it.

onlyjob, 2016-02-09:
The following files contain `Copyright (C) Jan M. Hochstein` but no copyright year(s):

    scripts/ZoneMinder/lib/ZoneMinder/Control/onvif.pm
    web/skins/classic/views/onvifprobe.php

altaroca, 2016-02-10:
Let's see, I commited files with these copyrights:
1 - ONVIF and OASIS wsdl files - these have the copyright notice at the beginning of the file. 
2 - Code written by me - I am going to look through those and make sure they all have name and year. Should I make a pull request for that change?
3 - Generated files - these are generated by my soap2perl fork. (If they could be generated by the original soap2perl it would not be necessary to commit them at all. Since they could then be generated during the build process.) Do these files need a copyright notice?

altaroca, 2016-02-10:
Thinking further it would make sense to drop the wsdl files. Since generation from wsdl currently only works on my local box. we do not need them. I could create a README for the onvif folder listing the URLs where they can be downloaded.

onlyjob, 2016-02-10:
Thanks, @altaroca. There is nowhere in your code I could find your copyright or grant of license. Without this information your contribution is (technically speaking) "All Rights Reserved" i.e. proprietary.

It would be great to drop _wsdl_ folder. You do not have to add copyright/license headers to every file but it is important to add copyright statement, license grant as well as to document files' origins and scope i.e. which files/folders this information applies to.

As for generated files it would be best to generate 'em on build time as well as to have build instructions and script(s). It is preferable that generated files are built by original `soap2perl` software. Please submit your patches to soap2perl project. Also please mention all related build tools in documentation as well as their home pages. I'd create `onvif/README` with all that information and maybe `onvif/LICENSE` with full text of the license (unless you've chosen GPL-2+ license for your project).

> it would make sense to drop the wsdl files. Since generation from wsdl currently only works on my local box. we do not need them. 

The problem is that non-free files may not be distributable. If you want to keep ZoneMinder free please don't pollute its source code with non-free licensed stuff.

> I could create a README for the onvif folder listing the URLs where they can be downloaded.

Great, please do that.

knnniggett, 2016-02-10:
@altaroca Thank you for your response. In an effort to have a documented trail that shows that the original author modified the license, rather than someone else, we would ask that you and only you make the changes through a pull request. Of course, we can answer any further if you have any.

altaroca, 2016-02-10:
Added PR #1280 . Please review

knnniggett, 2016-02-11:
@onlyjob Are there any remaining items left to satisfy closing this issue?

SteveGilvarry, 2016-02-13:
@onlyjob are any of the issues showstoppers for packaging 1.29.0 in Debian, or will you be able to patch around them?

SteveGilvarry, 2016-02-13:
@onlyjob sorry should have checked first see you are already doing it. https://packages.qa.debian.org/z/zoneminder.html

onlyjob, 2016-02-13:
> Are there any remaining items left to satisfy closing this issue?

All good, thank you, @altaroca, @knnniggett.

---

No worries, @SteveGilvarry. At the moment 1.29.0 is uploaded to Debian without ONVIF.
There is a technical issue with [FTBFS](https://github.com/ZoneMinder/ZoneMinder/pull/1273#issuecomment-183691047) and some concerns for large /usr/share in arch-dep package... I need to think whether it would be better to ship onvif as a standalone arch-indep package, test etc...

closed 2016-02-13 by onlyjob