ZoneMinder/zoneminder #1274: pre-built file "

onlyjob, 2016-02-09:


onlyjob, 2016-02-09:
@schrorg: Robin, could you please explain the license, origin of this file and how it is generated? This file should be distributed with sources. Thanks.

schrorg, 2016-02-10:
Indeed, one missing thing is the script I used to create the file. I'll make a pull request for that shortly.

And regarding the licence, the original file `src/zm_font.h` does not contain that information either, as you can see in the head of the file. So I would assume it would be GPLv2 as the rest of the code is.

onlyjob, 2016-02-10:
Fair enough, but likely this font file is derived from one of the web fonts that are most probably  distributed under different license (GPL-2+ is unusual license for fonts). We need to identify the original font to determine the license...

schrorg, 2016-02-10:
Okay, the best way would be to find "rthelen" and ask him or her. Does anybody know the author? A quick search discovered only http://en.pudn.com/downloads651/sourcecode/unix_linux/detail2648941_en.html and @rthelen.

POKKAHOH, 2016-02-15:
Excuse me, it may be better to change the method of generating a timestamps? Do not use ASKII table, but beter UTF-8 for support international characters?

schrorg, 2016-02-18:
@POKKAHOH The problem is, the addition of the timestamps to the images happen during the capture process and need to be extremely fast. One would have to use a library for that and then measure the impact of using it.

knnniggett, 2017-05-18:
fixed by #1278 so closing

closed 2017-05-18 by knnniggett