ZoneMinder/zoneminder #1276: license of `scripts/ZoneMinder/lib/ZoneMinder/Control/TVIP862.pm`?

onlyjob, 2016-02-09:
[scripts/ZoneMinder/lib/ZoneMinder/Control/TVIP862.pm](../blob/master/scripts/ZoneMinder/lib/ZoneMinder/Control/TVIP862.pm) file contains fairly confusing licensing information. In the footer of the file there is copyright and "Perl" license but unlike other files where header also contains GPL-2+ license grant `TVIP862.pm` have only GPL disclaimer. Please clarify licensing of the `TVIP862.pm`. @connortechnology, could you clarify please?
I suspect that fragment of GPL-2+ license is missing...

connortechnology, 2016-02-09:
yeah, will have to contact original author I guess.

connortechnology, 2016-02-09:
Given that he seems to not want to be contacted, we may have to remove this script and re-implement it.

onlyjob, 2018-06-16:
No activity doesn't mean that the problem is gone.

onlyjob, 2018-08-15:
Can somebody disable that "stale bot" please? It is wrong to close valid tickets only because they could not get attention... :(

connortechnology, 2018-08-15:
I like the stale bot bugging people about the stale issue, but I agree it shouldn't close them.  However 3 of the 4 people in charge disagree.

TVIP862 has been rewritten and renamed to Trendnet.pm.  I have removed the extra license bits.

closed 2018-08-15 by connortechnology

onlyjob, 2018-08-17:
But there is no licensing information left in the https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/Control/Trendnet.pm

Also copyright on line 4 doesn't look good...

connortechnology, 2018-08-17:
Ok I don't know what happened there.  Please review the latest version.

I assume it is ok to just put the licensing in the perldoc bits at the end instead of first thing in the file.

onlyjob, 2018-08-17:
But I am talking about the latest version changed two days ago... There is no newer version...
Of course it is OK to put licensing info to the end. :)

connortechnology, 2018-08-17:
Crap, too many branches/repos.   

I believe I have pushed my latest version.

onlyjob, 2018-08-18:
:) Now all good. Thanks.

commit bb70c9f1 (2018-08-15): Cleanup Licenses. Fixes #1276