ZoneMinder/zoneminder #1281: Web Console Free space issue

knnniggett, 2016-02-10:
In 1.29.0 we changed the way we determine free space from using the df system command to the call to php native disk_free_space. It made sense at the time. Typically we do want to use native calls.

However, initial reports indicate disk_free_space does not report free disk space the same way df does. It appears to only work correctly on some, but not all, local disks. The bad thing is that disk_free_space does not fail. It simply reports an inaccurate value.

Here is an example comparison between disk_free_space and df on an NFS volume:
https://gist.github.com/pliablepixels/5e025ab3973854d26435

Here is a similar example on a local volume using dev_mapper (probably a fake raid volume):
https://forums.zoneminder.com/viewtopic.php?p=92474&sid=3c5d5089f4f0820e0bdec13804e7371d

In some cases, it appears that dividing the block size from the value reported by disk_free_space gets us a value nearly the same as what df reports. However, we would have to add extra logic to somehow determine when and when not to divide the value by the block size.

@pliablepixels and I have been discussing this, and I can't think of any other solution at the moment than to revert back to using df.

If I don't hear any feedback over the next couple days, you can expect a pull request which will revert commit 8350ddac7cda0b4f314fab2301fd0cb5094d4dc2

pliablepixels, 2016-02-10:
I'm probably complicating the situation more here, but I just tried in on my linux server (prev. test was on a Mac) and I'm getting fairly consistent results for disk_free_space vs. df - be it for remotely mounted drives or local drives.

https://gist.github.com/pliablepixels/c846c25648df4526531c

knnniggett, 2016-04-10:
#1407 is related to this.

closed 2017-03-18 by SteveGilvarry