ZoneMinder/zoneminder #1286: ONVIF documentation issues

onlyjob, 2016-02-14:
I'm struggling to produce a meaningful package description for ONVIF. What ONVIF is?
If anyone can explain then perhaps it would be great to add this information to `onvif/README` as well as to `onvif/modules/lib/ONVIF/Client.pm` and other modules.
There are very little or no POD documentation in ONVIF.

Earlier we have added some POD docs to all executable scripts but `zmonvif-probe.pl` is undocumented (so we can't generate its man page).

All ONVIF man pages for section 3 are affected by [manpage-has-bad-whatis-entry](https://lintian.debian.org/tags/manpage-has-bad-whatis-entry.html) so there are heaps of warnings including more serious "manpage-has-errors-from-pod2man" as well of minor spelling warnings "spelling-error-in-manpage" (mostly on "occurences" --> "occurrences" and "occurance" --> "occurrence").

knnniggett, 2016-02-15:
This is why I provided the link to the authors blog in #1272 
> You can read about Onvif support in @altaroca 's blog here:
> https://altaroca.wordpress.com/the-onvif-series/
> 
> Specifically:
> https://altaroca.wordpress.com/2014/07/14/onvif-probing-for-zoneminder/

This is all there is currently. The level of Onvif support only includes discovery. As shown in the screenshot, one simply clicks on the ONVIF link in the monitor window. It works much like the Probe link right next to it.

This feature is entirely experimental, and so far we have never seen it work for any device other than the author's own camera. It didn't work for the three cameras I tested it on. #1290 pretty much sums up the results I expect most will get, but it would be nice to get this feature out in the wild and verify with a larger user base.

I agree with you that the issues you mentioned should be fixed, but I'm not willing to make this a priority right now when we don't yet know what the final product will look like. This initial offering of onvif support could be the tip of the iceberg. Perhaps someone else can spend some time with this.

If this does not meet your standards for a ZoneMinder 1.29.0 package, then I'd recommend building with Onvif support disabled. None of the other Linux distros, which host 1.29.0 packages, have Onvif enabled in them that I am aware of.

onlyjob, 2016-02-15:
I see... I did not know how experimental ONVIF component is... You've convinced me that it would be best to leave ONVIF out of official packages for now. Packaging ONVIF properly would require significant effort to prepare new arch-indep packages and nail all the warnings...
Looks like it is not worth the effort yet... Thank you.

murre87, 2016-03-09:
Im trying to get onvif working with ubuntu.
I got ZM 1.29 and when pressing the "ONVIF" link in ad new monitor it doesnt find any cameras. 

`
/usr/bin/zmonvif-probe.pl profiles "http://192.168.1.128:1018/onvif/device_service" 1.2 admin admin
`
Answer
`
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"/>MainStreamTooken, MainStream, H264, 1280, 720, 20, `

My cameras is working with onvif in windows and android.
The are working in vlc on this andress: [rtsp://admin:admin@192.168.1.128:554/12](url)

`
/usr/bin/zmonvif-probe.pl -v profiles "http://192.168.1.128:1018/onvif/device_service" 1.2 admin admin
`
Answer
`
Received message:
<GetProfilesResponse xmlns="http://www.onvif.org/ver10/media/wsdl"><Profiles token="MainStreamTooken" fixed="true"><Name xmlns="http://www.onvif.org/ver10/schema">MainStream</Name><VideoSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>video_source_cfg_0</Name><UseCount>1</UseCount><SourceToken>video_source_0</SourceToken><Bounds x="0" y="0" width="1280" height="720"></Bounds></VideoSourceConfiguration><AudioSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>audio_source_configuration</Name><UseCount>2</UseCount><SourceToken>audio_source</SourceToken></AudioSourceConfiguration><VideoEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>video_encoder_configuration</Name><UseCount>2</UseCount><Encoding>H264</Encoding><Resolution><Width>1280</Width><Height>720</Height></Resolution><Quality>3</Quality><RateControl><FrameRateLimit>20</FrameRateLimit><EncodingInterval>20</EncodingInterval><BitrateLimit>1843</BitrateLimit></RateControl><H264><GovLength>20</GovLength><H264Profile>Baseline</H264Profile></H264><Multicast><Address><Type>IPv4</Type><IPv4Address></IPv4Address><IPv6Address></IPv6Address></Address><Port>0</Port><TTL>0</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT10S</SessionTimeout></VideoEncoderConfiguration><AudioEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>audio_encoder_configuration_0</Name><UseCount>3</UseCount><Encoding>G711</Encoding><Bitrate>64</Bitrate><SampleRate>8</SampleRate><Multicast><Address><Type>IPv4</Type><IPv4Address></IPv4Address><IPv6Address></IPv6Address></Address><Port>0</Port><TTL>0</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT10S</SessionTimeout></AudioEncoderConfiguration><PTZConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>PTZConfigName_000</Name><UseCount>2</UseCount><NodeToken>000</NodeToken><DefaultAbsolutePantTiltPositionSpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/PositionGenericSpace</DefaultAbsolutePantTiltPositionSpace><DefaultAbsoluteZoomPositionSpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/PositionGenericSpace</DefaultAbsoluteZoomPositionSpace><DefaultRelativePanTiltTranslationSpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/TranslationGenericSpace</DefaultRelativePanTiltTranslationSpace><DefaultRelativeZoomTranslationSpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/TranslationGenericSpace</DefaultRelativeZoomTranslationSpace><DefaultContinuousPanTiltVelocitySpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/VelocityGenericSpace</DefaultContinuousPanTiltVelocitySpace><DefaultContinuousZoomVelocitySpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/VelocityGenericSpace</DefaultContinuousZoomVelocitySpace><DefaultPTZSpeed><PanTilt x="0.5" y="0.5" space="http://www.onvif.org/ver10/tptz/PanTiltSpaces/GenericSpeedSpace"></PanTilt><Zoom x="0.5" space="http://www.onvif.org/ver10/tptz/ZoomSpaces/ZoomGenericSpeedSpace"></Zoom></DefaultPTZSpeed><DefaultPTZTimeout>PT10S</DefaultPTZTimeout><PanTiltLimits><Range><URI>http://www.onvif.org/ver10/tptz/PanTiltSpaces/PositionGenericSpa</URI><XRange><Min>0</Min><Max>1</Max></XRange><YRange><Min>0</Min><Max>1</Max></YRange></Range></PanTiltLimits><ZoomLimits><Range><URI>http://www.onvif.org/ver10/tptz/ZoomSpaces/PositionGenericSpace</URI><XRange><Min>0</Min><Max>1</Max></XRange></Range></ZoomLimits></PTZConfiguration></Profiles><Profiles token="SubStreamTooken" fixed="true"><Name xmlns="http://www.onvif.org/ver10/schema">SubStream</Name><VideoSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>video_source_cfg_1</Name><UseCount>1</UseCount><SourceToken>video_source_0</SourceToken><Bounds x="0" y="0" width="1280" height="720"></Bounds></VideoSourceConfiguration><AudioSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>audio_source_configuration</Name><UseCount>2</UseCount><SourceToken>audio_source</SourceToken></AudioSourceConfiguration><VideoEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>video_encoder</Name><UseCount>2</UseCount><Encoding>H264</Encoding><Resolution><Width>640</Width><Height>360</Height></Resolution><Quality>3</Quality><RateControl><FrameRateLimit>20</FrameRateLimit><EncodingInterval>25</EncodingInterval><BitrateLimit>614</BitrateLimit></RateControl><H264><GovLength>25</GovLength><H264Profile>Baseline</H264Profile></H264><Multicast><Address><Type>IPv4</Type><IPv4Address></IPv4Address><IPv6Address></IPv6Address></Address><Port>0</Port><TTL>0</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT10S</SessionTimeout></VideoEncoderConfiguration><AudioEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>audio_encoder_configuration_0</Name><UseCount>3</UseCount><Encoding>G711</Encoding><Bitrate>64</Bitrate><SampleRate>8</SampleRate><Multicast><Address><Type>IPv4</Type><IPv4Addr [... cut ...] ><URI>http://www.onvif.org/ver10/tptz/PanTiltSpaces/PositionGenericSpa</URI><XRange><Min>0</Min><Max>1</Max></XRange><YRange><Min>0</Min><Max>1</Max></YRange></Range></PanTiltLimits><ZoomLimits><Range><URI>http://www.onvif.org/ver10/tptz/ZoomSpaces/PositionGenericSpace</URI><XRange><Min>0</Min><Max>1</Max></XRange></Range></ZoomLimits></PTZConfiguration></Profiles></GetProfilesResponse>
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"/>MainStreamTooken, MainStream, H264, 1280, 720, 20,
`

knnniggett, 2016-03-22:
I'm closing this out since the original question concerning documentation has been answered.
The current implementation is self explanatory. We will add official documentation to a future version of zoneminder, once we have a good idea of what the final onvif implementation will look like in zoneminder.

We are tracking camera discovery issues in #1346 . Please post any data concerning camera discovery there.

closed 2016-03-22 by knnniggett

onlyjob, 2016-03-22:
I'd still like to see brief POD documentation in ONVIF modules, eventually....