ZoneMinder/zoneminder #1287: ZoneMinder consultants (new Wiki page)

onlyjob, 2016-02-14:
Hi team,

I want to make a new Wiki page "consultants" listing people and organisations offering commercial ZoneMinder support/services, similar to https://www.debian.org/consultants/ and https://civicrm.org/partners-contributors

I think list of known consultants with recognised skills around ZoneMinder may be of help to those who seek paid assistance and/or support/services as well as for contributors to help them raise some funds and (hopefully) contribute more.

I shall be happy to add my own company and I suppose @connortechnology would be interested to list his business as well.

It is not against GitHub TOS, right?

Please let me know if you have any objections or ideas. Thanks.

connortechnology, 2016-02-16:
Yes this is needed and is on the todo list.

onlyjob, 2016-02-16:
So shall I/we do it now?

knnniggett, 2016-02-16:
Was @kylejohnson planning on putting something like this on our new website?

onlyjob, 2016-02-16:
I think Wiki page might be better as it will be easier to update (or request changes).
For now we could start with wiki page and see how it goes. It can always be changed later.

onlyjob, 2016-07-01:
https://github.com/ZoneMinder/ZoneMinder/wiki/Consultants
Done, nothing more to do here.

closed 2016-07-01 by onlyjob