ZoneMinder/zoneminder #1288: ZM doesnt compile in FreeBSD

rajil, 2016-02-14:
I tried compiling version 1.29 in a FreeBSD 10.2 jail but it failed.

cat config.log
```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by zm configure 1.29.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --prefix=/usr/local/zoneminder --with-libarch=lib64 --with-mysql=/usr/local --with-mariadb=/usr/local --with-polkit=/usr/local

## --------- ##
## Platform. ##
## --------- ##

hostname = zm
uname -m = amd64
uname -r = 10.2-RELEASE-p12
uname -s = FreeBSD
uname -v = FreeBSD 10.2-RELEASE-p12 #4 r289129M: Sun Feb  7 08:12:08 CST 2016     rs4@nas:/usr/obj/usr/src/sys/MOREROUTINGTABLES 

/usr/bin/uname -p = amd64
/bin/uname -X     = unknown

/bin/arch              = unknown
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /sbin
PATH: /bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /usr/games
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /home/zm/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2603: checking for a BSD-compatible install
configure:2671: result: /usr/bin/install -c
configure:2682: checking whether build environment is sane
configure:2737: result: yes
configure:2888: checking for a thread-safe mkdir -p
configure:2927: result: ./install-sh -c -d
configure:2934: checking for gawk
configure:2964: result: no
configure:2934: checking for mawk
configure:2964: result: no
configure:2934: checking for nawk
configure:2950: found /usr/bin/nawk
configure:2961: result: nawk
configure:2972: checking whether make sets $(MAKE)
configure:2994: result: yes
configure:3023: checking whether make supports nested variables
configure:3040: result: yes
configure:3183: checking build system type
configure:3197: result: amd64-unknown-freebsd10.2
configure:3217: checking host system type
configure:3230: result: amd64-unknown-freebsd10.2

## ---------------- ##
## Cache variables. ##
## ---------------- ##

ac_cv_build=amd64-unknown-freebsd10.2
ac_cv_env_CCC_set=''
ac_cv_env_CCC_value=''
ac_cv_env_CC_set=''
ac_cv_env_CC_value=''
ac_cv_env_CFLAGS_set=''
ac_cv_env_CFLAGS_value=''
ac_cv_env_CPPFLAGS_set=''
ac_cv_env_CPPFLAGS_value=''
ac_cv_env_CPP_set=''
ac_cv_env_CPP_value=''
ac_cv_env_CXXCPP_set=''
ac_cv_env_CXXCPP_value=''
ac_cv_env_CXXFLAGS_set=''
ac_cv_env_CXXFLAGS_value=''
ac_cv_env_CXX_set=''
ac_cv_env_CXX_value=''
ac_cv_env_LDFLAGS_set=''
ac_cv_env_LDFLAGS_value=''
ac_cv_env_LIBS_set=''
ac_cv_env_LIBS_value=''
ac_cv_env_ZM_CONFIG_DIR_set=''
ac_cv_env_ZM_CONFIG_DIR_value=''
ac_cv_env_ZM_DB_HOST_set=''
ac_cv_env_ZM_DB_HOST_value=''
ac_cv_env_ZM_DB_NAME_set=''
ac_cv_env_ZM_DB_NAME_value=''
ac_cv_env_ZM_DB_PASS_set=''
ac_cv_env_ZM_DB_PASS_value=''
ac_cv_env_ZM_DB_TYPE_set=''
ac_cv_env_ZM_DB_TYPE_value=''
ac_cv_env_ZM_DB_USER_set=''
ac_cv_env_ZM_DB_USER_value=''
ac_cv_env_ZM_LOGDIR_set=''
ac_cv_env_ZM_LOGDIR_value=''
ac_cv_env_ZM_MYSQL_ENGINE_set=''
ac_cv_env_ZM_MYSQL_ENGINE_value=''
ac_cv_env_ZM_RUNDIR_set=''
ac_cv_env_ZM_RUNDIR_value=''
ac_cv_env_ZM_SOCKDIR_set=''
ac_cv_env_ZM_SOCKDIR_value=''
ac_cv_env_ZM_SSL_LIB_set=''
ac_cv_env_ZM_SSL_LIB_value=''
ac_cv_env_ZM_TMPDIR_set=''
ac_cv_env_ZM_TMPDIR_value=''
ac_cv_env_build_alias_set=''
ac_cv_env_build_alias_value=''
ac_cv_env_host_alias_set=''
ac_cv_env_host_alias_value=''
ac_cv_env_target_alias_set=''
ac_cv_env_target_alias_value=''
ac_cv_host=amd64-unknown-freebsd10.2
ac_cv_path_install='/usr/bin/install -c'
ac_cv_prog_AWK=nawk
ac_cv_prog_make_make_set=yes
am_cv_make_support_nested_variables=yes

## ----------------- ##
## Output variables. ##
## ----------------- ##

ACLOCAL='${SHELL} /usr/home/zm/zoneminder/missing aclocal-1.15'
ALLOCA=''
AMDEPBACKSLASH=''
AMDEP_FALSE=''
AMDEP_TRUE=''
AMTAR='$${TAR-tar}'
AM_BACKSLASH='\'
AM_CXXFLAGS='-D__STDC_CONSTANT_MACROS'
AM_DEFAULT_V='$(AM_DEFAULT_VERBOSITY)'
AM_DEFAULT_VERBOSITY='1'
AM_V='$(V)'
AUTOCONF='${SHELL} /usr/home/zm/zoneminder/missing autoconf'
AUTOHEADER='${SHELL} /usr/home/zm/zoneminder/missing autoheader'
AUTOMAKE='${SHELL} /usr/home/zm/zoneminder/missing automake-1.15'
AWK='nawk'
BINDIR=''
CC=''
CCDEPMODE=''
CFLAGS=''
CGI_PREFIX=''
COND_ONVIF_FALSE=''
COND_ONVIF_TRUE=''
CPP=''
CPPFLAGS=''
CXX=''
CXXCPP=''
CXXDEPMODE=''
CXXFLAGS=''
CYGPATH_W='echo'
DATADIR=''
DEFS=''
DEPDIR=''
ECHO_C=''
ECHO_N='-n'
ECHO_T=''
EGREP=''
ENABLE_MMAP=''
EXEEXT=''
EXTRA_LIBS=''
EXTRA_PERL_LIB=''
FFMPEG_CFLAGS=''
FFMPEG_LIBS=''
FFMPEG_PREFIX=''
GREP=''
HOST_OS='BSD'
INSTALL_DATA='${INSTALL} -m 644'
INSTALL_PROGRAM='${INSTALL}'
INSTALL_SCRIPT='${INSTALL}'
INSTALL_STRIP_PROGRAM='$(install_sh) -c -s'
LDFLAGS='-L/usr/lib/  -L/usr/local/lib64/mysql -L/usr/lib64/mariadb'
LIBDIR=''
LIBOBJS=''
LIBS=''
LIB_ARCH='lib64'
LN_S=''
LTLIBOBJS=''
MAKEINFO='${SHELL} /usr/home/zm/zoneminder/missing makeinfo'
MARIADB_CFLAGS='-I/usr/include'
MARIADB_LIBS='-L/usr/lib64/mariadb'
MARIADB_PREFIX='/usr'
MKDIR_P='./install-sh -c -d'
MYSQL_CFLAGS='-I/usr/local/include'
MYSQL_LIBS='-L/usr/local/lib64/mysql'
MYSQL_PREFIX='/usr/local'
OBJEXT=''
OPT_FFMPEG=''
PACKAGE='zoneminder'
PACKAGE_BUGREPORT='http://www.zoneminder.com/forums/ - Please check FAQ first'
PACKAGE_NAME='zm'
PACKAGE_STRING='zm 1.29.0'
PACKAGE_TARNAME='zoneminder'
PACKAGE_URL='http://www.zoneminder.com/downloads.html'
PACKAGE_VERSION='1.29.0'
PATH_FFMPEG=''
PATH_SEPARATOR=':'
PERL=''
PERL_MM_PARMS=''
PKGDATADIR=''
POLKIT_PREFIX='/usr/local'
POW_LIB=''
RANLIB=''
SENDFILE_LIBS=''
SET_MAKE=''
SHELL='/bin/sh'
STRIP=''
SYSCONFDIR='/usr/local/zoneminder/etc'
VERSION='1.29.0'
WEB_GROUP=''
WEB_HOST=''
WEB_PREFIX=''
WEB_USER=''
ZM_CONFIG=''
ZM_CONFIG_DIR='/usr/local/zoneminder/etc'
ZM_DB_HOST='localhost'
ZM_DB_NAME='zm'
ZM_DB_PASS='zmpass'
ZM_DB_TYPE='mysql'
ZM_DB_USER='zmuser'
ZM_HAS_GNUTLS=''
ZM_HAS_GNUTLS_OPENSSL=''
ZM_HAS_ONVIF=''
ZM_HAS_V4L1=''
ZM_HAS_V4L2=''
ZM_HAS_V4L=''
ZM_LOGDIR='/var/log/zm'
ZM_MYSQL_ENGINE='InnoDB'
ZM_PCRE=''
ZM_PID=''
ZM_RUNDIR='/var/run/zm'
ZM_SOCKDIR='/var/run/zm'
ZM_SSL_LIB='gnutls'
ZM_TMPDIR='/tmp/zm'
ac_ct_CC=''
ac_ct_CXX=''
am__EXEEXT_FALSE=''
am__EXEEXT_TRUE=''
am__fastdepCC_FALSE=''
am__fastdepCC_TRUE=''
am__fastdepCXX_FALSE=''
am__fastdepCXX_TRUE=''
am__include=''
am__isrc=''
am__leading_dot='.'
am__nodep=''
am__quote=''
am__tar='$${TAR-tar} chof - "$$tardir"'
am__untar='$${TAR-tar} xf -'
bindir='${exec_prefix}/bin'
build='amd64-unknown-freebsd10.2'
build_alias=''
build_cpu='amd64'
build_os='freebsd10.2'
build_vendor='unknown'
datadir='${datarootdir}'
datarootdir='${prefix}/share'
docdir='${datarootdir}/doc/${PACKAGE_TARNAME}'
dvidir='${docdir}'
exec_prefix='NONE'
host='amd64-unknown-freebsd10.2'
host_alias=''
host_cpu='amd64'
host_os='freebsd10.2'
host_vendor='unknown'
htmldir='${docdir}'
includedir='${prefix}/include'
infodir='${datarootdir}/info'
install_sh='${SHELL} /usr/home/zm/zoneminder/install-sh'
libdir='${exec_prefix}/lib'
libexecdir='${exec_prefix}/libexec'
localedir='${datarootdir}/locale'
localstatedir='${prefix}/var'
mandir='${datarootdir}/man'
mkdir_p='$(MKDIR_P)'
oldincludedir='/usr/include'
pdfdir='${docdir}'
prefix='/usr/local/zoneminder'
program_transform_name='s,x,x,'
psdir='${docdir}'
sbindir='${exec_prefix}/sbin'
sharedstatedir='${prefix}/com'
sysconfdir='${prefix}/etc'
target_alias=''

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "zm"
#define PACKAGE_TARNAME "zoneminder"
#define PACKAGE_VERSION "1.29.0"
#define PACKAGE_STRING "zm 1.29.0"
#define PACKAGE_BUGREPORT "http://www.zoneminder.com/forums/ - Please check FAQ first"
#define PACKAGE_URL "http://www.zoneminder.com/downloads.html"
#define PACKAGE "zoneminder"
#define VERSION "1.29.0"
#define BSD 1
#define SYSCONFDIR "/usr/local/zoneminder/etc"

configure: exit 2
```

knnniggett, 2016-02-14:
Please see the 1.29.0 release notes. Building from source with autotools it no longer supported and will be removed in an upcoming version. Please use cmake instead.

rajil, 2016-02-14:
Thanks, i tried compiling with cmake. But it gave an error while compiling/linking. Any way to fix this.

```
$mkdir build; cd build 
$cmake ..
-- The C compiler identification is Clang 3.4.1
-- The CXX compiler identification is Clang 3.4.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/CC
-- Check for working CXX compiler: /usr/bin/CC -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAVE_SENDFILE4_SUPPORT
-- Performing Test HAVE_SENDFILE4_SUPPORT - Failed
-- Performing Test HAVE_SENDFILE7_SUPPORT
-- Performing Test HAVE_SENDFILE7_SUPPORT - Success
-- Sendfile support: FreeBSD sendfile()
-- Looking for libv4l1-videodev.h
-- Looking for libv4l1-videodev.h - not found
-- Looking for linux/videodev.h
-- Looking for linux/videodev.h - not found
-- Looking for linux/videodev2.h
-- Looking for linux/videodev2.h - not found
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for backtrace
-- Looking for backtrace - not found
-- Looking for backtrace_symbols
-- Looking for backtrace_symbols - not found
-- Looking for ucontext.h
-- Looking for ucontext.h - found
-- Looking for sys/sendfile.h
-- Looking for sys/sendfile.h - not found
-- Looking for sys/syscall.h
-- Looking for sys/syscall.h - found
-- Looking for syscall
-- Looking for syscall - found
-- Looking for sendfile
-- Looking for sendfile - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of siginfo_t
-- Check size of siginfo_t - done
-- Check size of ucontext_t
-- Check size of ucontext_t - failed
-- Looking for clock_gettime in rt
-- Looking for clock_gettime in rt - found
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.8") 
-- Looking for zlib.h
-- Looking for zlib.h - found
-- Found CURL: /usr/local/lib/libcurl.so (found version "7.47.0") 
-- Looking for curl/curl.h
-- Looking for curl/curl.h - found
-- Found JPEG: /usr/local/lib/libjpeg.so  
-- Looking for include files stdio.h, jpeglib.h
-- Looking for include files stdio.h, jpeglib.h - found
-- Found OpenSSL: /usr/lib/libssl.so;/usr/lib/libcrypto.so (found version "1.0.1p") 
[... 77 lines cut ...]

And when compiling
```
$gmake
-----------------------
[ 74%] Building CXX object src/CMakeFiles/zm.dir/zm_zone.cpp.o
[ 76%] Linking CXX static library libzm.a
[ 76%] Built target zm
Scanning dependencies of target zma
[ 78%] Building CXX object src/CMakeFiles/zma.dir/zma.cpp.o
[ 80%] Linking CXX executable zma
/usr/bin/ld: cannot find -l/usr/lib/libexecinfo.so
CC: error: linker command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/zma.dir/build.make:111: recipe for target 'src/zma' failed
gmake[2]: *** [src/zma] Error 1
CMakeFiles/Makefile2:367: recipe for target 'src/CMakeFiles/zma.dir/all' failed
gmake[1]: *** [src/CMakeFiles/zma.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
gmake: *** [all] Error 2
```
libexecinfo does exist
```
$ ls -la /usr/lib/libexecinfo.so
lrwxr-xr-x  1 root  wheel  16 Oct 11 00:48 /usr/lib/libexecinfo.so -> libexecinfo.so.1
```

knnniggett, 2016-02-14:
Try adding `-lexecinfo` to the linker, similar to this report:
https://bugs.gentoo.org/show_bug.cgi?id=276389

rajil, 2016-02-14:
I modified CMakeLists.txt line 63 to
set(CMAKE_CXX_FLAGS_RELEASE "-Wall -D__STDC_CONSTANT_MACROS -O2 -lexecinfo")

but still got the same error
```
Scanning dependencies of target zma
[ 78%] Building CXX object src/CMakeFiles/zma.dir/zma.cpp.o
CC: warning: -lexecinfo: 'linker' input unused
[ 80%] Linking CXX executable zma
/usr/bin/ld: cannot find -l/usr/lib/libexecinfo.so
CC: error: linker command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/zma.dir/build.make:111: recipe for target 'src/zma' failed
gmake[2]: *** [src/zma] Error 1
CMakeFiles/Makefile2:367: recipe for target 'src/CMakeFiles/zma.dir/all' failed
gmake[1]: *** [src/CMakeFiles/zma.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
gmake: *** [all] Error 2
```

rajil, 2016-02-18:
Here add a '-v' to the CXX flags i get this
```
[ 80%] Linking CXX executable zma
FreeBSD clang version 3.4.1 (tags/RELEASE_34/dot1-final 208032) 20140512
Target: x86_64-unknown-freebsd10.2
Thread model: posix
Selected GCC installation: 
 "/usr/bin/ld" --eh-frame-hdr -dynamic-linker /libexec/ld-elf.so.1 --hash-style=both --enable-new-dtags -o zma /usr/lib/crt1.o /usr/lib/crti.o /usr/lib/crtbegin.o -L/usr/lib -lexecinfo CMakeFiles/zma.dir/zma.cpp.o libzm.a -l/usr/lib/libexecinfo.so -lrt -lz /usr/local/lib/libcurl.so /usr/local/lib/libjpeg.so -lssl -lcrypto -lpthread /usr/local/lib/libpcre.so /usr/local/lib/libgcrypt.so /usr/local/lib/libgnutls-openssl.so /usr/local/lib/mysql/libmysqlclient.so /usr/local/lib/libavformat.so /usr/local/lib/libavcodec.so /usr/local/lib/libavdevice.so /usr/local/lib/libavutil.so /usr/local/lib/libswscale.so /usr/local/lib/libvlc.so -rpath /usr/local/lib:/usr/local/lib/mysql: -lc++ -lm -lgcc --as-needed -lgcc_s --no-as-needed -lc -lgcc --as-needed -lgcc_s --no-as-needed /usr/lib/crtend.o /usr/lib/crtn.o
/usr/bin/ld: cannot find -l/usr/lib/libexecinfo.so
CC: error: linker command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/zma.dir/build.make:111: recipe for target 'src/zma' failed
gmake[2]: *** [src/zma] Error 1
CMakeFiles/Makefile2:367: recipe for target 'src/CMakeFiles/zma.dir/all' failed
gmake[1]: *** [src/CMakeFiles/zma.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
gmake: *** [all] Error 2
```

connortechnology, 2016-02-18:
You know, the message is kindof clear.  It is looking for libexecinfo at /usr/lib/libexecinfo.so, and it obviously isn't there.  You might try something like sudo find /usr -name "libexecinfo*" to find it.

rajil, 2016-02-18:
libexecinfo is present,
$ ls -la  /usr/lib/libexecinfo.so
lrwxr-xr-x  1 root  wheel  16 Oct 11 00:48 /usr/lib/libexecinfo.so -> libexecinfo.so.1

connortechnology, 2016-02-18:
it's a link... does libexecinfo.so.1 exist?

rajil, 2016-02-18:
Yes
```
$ ls -la  /usr/lib/libexecinfo.so.1 
-r--r--r--  1 root  wheel  11152 Oct 11 00:48 /usr/lib/libexecinfo.so.1

$ ldd /usr/lib/libexecinfo.so
/usr/lib/libexecinfo.so:
        libelf.so.1 => /usr/lib/libelf.so.1 (0x801603000)
        libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x801818000)
        libc.so.7 => /lib/libc.so.7 (0x800821000)
```

connortechnology, 2016-02-18:
dang.  My next thought is that it might be a 32bit/64bit thing.

rajil, 2016-02-18:
This is a 64 bit system. Is zm looking for a 32bit library?

connortechnology, 2016-02-18:
So.. I'm not much of a freebsd user, but have managed to build it there before.

At the moment, I can't get anything to build using ports.  Are they really that unstable?

Can you educate me on what steps you took in terms of installing your freebsd system, to get it to this point?

connortechnology, 2016-02-18:
One other thought occurred to me... sometimes positioning on the command line matters.  Maybe the -llibexecinfo is somehow placed wrong.

connortechnology, 2016-02-18:
The other thing I am noticing is that it shouldn't be -lexecinfo.so, it should just be -lexecinfo the .so is extraneous.

rajil, 2016-02-19:
Here are the steps  I did to build this:
1. Setup a FreeBSD jail (I am using 10.2 release)
2. Install the following packages using upstream binaries. There is no need to build using ports since they are already available upstream:
devel/autoconf
devel/automake
devel/cmake
multimedia/ffmpeg
devel/git
devel/gmake
multimedia/linux-f10-libv4l
multimedia/motion
databases/p5-DBD-mysql
databases/p5-DBI
devel/p5-Date-Manip
devel/p5-Devel-Backtrace
devel/p5-ExtUtils-PkgConfig
devel/p5-PkgConfig
devel/p5-Sys-Mmap
www/p5-Test-LWP-UserAgent
ports-mgmt/pkg
sysutils/polkit
devel/py-pkgconfig
multimedia/v4l-utils
multimedia/v4l_compat
multimedia/vlc
3.  Make the following changes in ZM code 
zm/scripts/ZoneMinder/CMakeLists.txt, replace 'make' with 'gmake' on line 27

connortechnology, 2016-02-19:
Data point for historical purposes: commenting out lines 207 to 209 of CMakeLists.txt fixes this build.  the check for the libexecinfo library is returning libexecinfo.so instead of libexecinfo.  Need to look into why.

rajil, 2016-02-20:
ZM is installing all the perl modules in the root (/) folder when doing gmake install. FreeBSD out of the box has the modules at
```
$ls -la /usr/local/lib/perl5/
total 42
drwxr-xr-x   4 root  wheel    4 Feb 20 02:51 .
drwxr-xr-x  26 root  wheel  809 Feb 20 03:13 ..
drwxr-xr-x  57 root  wheel  128 Feb 20 02:51 5.20
drwxr-xr-x  20 root  wheel   26 Feb 20 02:51 site_perl
```

The modules should thus be installed at /usr/local/lib/perl5/site_perl/Zoneminder. How can i fix the perl module path?

After manually moving the perl modules to /usr/local/lib/perl5/site_perl/Zoneminder, i tried to start zoneminder, but it gave an error:
```
$service start zoneminder
Starting zoneminder.
DBD::mysql::st execute failed: Out of range value for column 'Pid' at row 1 at /usr/local/lib/perl5/site_perl/ZoneMinder/Logger.pm line 652
```
How can this be fixed?

knnniggett, 2016-02-20:
ZoneMinder will by default try to install to the vendor_perl folders, not site_perl. You should refer to Perl's Make::Maker documentation to learn where those folder should be on your system.

You can of course override that. See the main CMakeLists.txt file for the proper cmake flag to set.

Couple of things:
- You are calling cmake with two dots ".." rather than one ".". Is this on purpose? Incorrectly setting this can and does cause cmake to place files into unexpected folders.
- You stated that the perl modules are being placed into the root folder, but that last line of data you have shown us shows one of the perl modules, Logger.pm, under the local site_perl folder. This appears to be a contradiction.

rajil, 2016-02-20:
I guess ZM_PERL_MM_PARMS is the variable to be set. Is this how it should be set?
```
ZM_PERL_MM_PARMS:STRING=INSTALLDIRS=/usr/local/lib/perl5/site_perl/
```

- I was building it a separate directory, hence was using '..'. I tried building in the same folder and got the same as result as before.
-Yes on install they are indeed being installed in root. As mentioned before i **manually** moved them to /usr/local/lib/perl5/site_perl/ to verify if ZM would start.

The install logs are attached. 
[gmake.install.txt](https://github.com/ZoneMinder/ZoneMinder/files/139210/gmake.install.txt)
[CMakeCache.txt](https://github.com/ZoneMinder/ZoneMinder/files/139211/CMakeCache.txt)

knnniggett, 2016-02-20:
Not sure what you mean by building it a separate directory.
If you mean building into staging folder, then you should be setting the $DESTDIR env variable before calling cmake. Try setting that and calling cmake with just one "."
Getting this wrong has been known to put Perl modules in weird places.

rajil, 2016-02-20:
I had created a separate build directory. However, based on the log files above you can possibly see i am building in the base directory now with the same end result.

knnniggett, 2016-02-20:
From your build output:
```
-- Installing: //ZoneMinder::Control::PelcoD.3
-- Installing: //ZoneMinder::Trigger::Connection::Example.3
-- Installing: //ZoneMinder.pm
```

See the double slash? The output should be
```
Installing: /path/to/your/systems/vendor/folder/ZoneMinder::Control::PelcoD.3
```
That means that Perl's Make::Maker module is getting an empty string for the value of the vendor folders. I am disappointed it did not stop with an error, but kept on trucking.

Consult the Make::Maker documentation to learn more:
http://perldoc.perl.org/ExtUtils/MakeMaker.html

Scroll down to see all the INSTALLDIRS macros.
To see what values all those macros on your system, do a ` perl '-V:install.*'`

You could probably pick out the vendor folder by just doing a "Perl -V" but that will produce a lot of output.

Knowing why this is happening will help us determine if we can do something on our end to prevent this from happening to other users.

If you don't care about the "why", then try working around the problem by manually setting ZM_PERL_MM_PARMS to something other than the default. Typically you would set INSTALLDIRS to one of perl, site, or vendor, rather than a hard coded folder.  To get more control over the destination you can reading about INSTALL_BASE or PREFIX. Make sure and read through all the documentation because these tend to be tricky and can have unintended side effects. 

It would also be interesting to know what @connortechnology did in his attempt to isntall zoneminder on freebsd.

rajil, 2016-02-20:
Here you go
```
$ perl '-V:install.*'
installarchlib='/usr/local/lib/perl5/5.20/mach';
installbin='/usr/local/bin';
installhtml1dir='';
installhtml3dir='';
installman1dir='/usr/local/lib/perl5/5.20/perl/man/man1';
installman3dir='/usr/local/lib/perl5/5.20/perl/man/man3';
installprefix='/usr/local';
installprefixexp='/usr/local';
installprivlib='/usr/local/lib/perl5/5.20';
installscript='/usr/local/bin';
installsitearch='/usr/local/lib/perl5/site_perl/mach/5.20';
installsitebin='/usr/local/bin';
installsitehtml1dir='';
installsitehtml3dir='';
installsitelib='/usr/local/lib/perl5/site_perl';
installsiteman1dir='/usr/local/lib/perl5/site_perl/man/man1';
installsiteman3dir='/usr/local/lib/perl5/site_perl/man/man3';
installsitescript='/usr/local/bin';
installstyle='lib/perl5';
installusrbinperl='undef';
installvendorarch='';
installvendorbin='';
installvendorhtml1dir='';
installvendorhtml3dir='';
installvendorlib='';
installvendorman1dir='';
installvendorman3dir='';
installvendorscript='';
```
I changed the 'vendor' to 'site' in CMakeLists.txt line 139 
```
set(ZM_PERL_MM_PARMS INSTALLDIRS=site 
```

However, it still installed all the perl packages in root!

knnniggett, 2016-02-20:
Are you deleting the CmakeCache.txt  each time before re-running cmake?
Please post the full build output again so we can see if anything has changed.

rajil, 2016-02-20:
I tried again after deleting the CMakeFiles folder. And now it does install the perl packages in /usr/local/lib/perl5/site_perl

Here is what i start zmpkg
```
root@zm:~ # zmpkg.pl version
1.29.0
root@zm:~ # zmpkg.pl start
root@zm:~ # zmpkg.pl status
root@zm:~ # 
root@zm:~ # tail -1 /var/log/messages
Feb 20 14:33:58 zm zmpkg[13379]: ERR [Unable to find valid 'su' syntax]
```

knnniggett, 2016-02-20:
Here is the subroutine that is complaining about no valid su syntax:
https://github.com/ZoneMinder/ZoneMinder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/General.pm#L93

Here is an incomplete list of things that can make this sub return failure:
- the sudo command is missing from your system
- the Apache web server is not running under the same account name you built zoneminder with (ZM_WEB_USER).
- /bin/sh is missing

To get a better idea of where it is failing, turn on debug and look for the debug output coming from your perl scripts.

rajil, 2016-02-20:
In my rig

- sudo is present
```
root@zm:~ # which sh sudo
/bin/sh
/usr/local/bin/sudo
```
But i dont have apache configured since i was hoping to use nginx.

I built ZM again using '-DCMAKE_BUILD_TYPE=Debug', but i dont see any other message except the previous error (Unable to find valid 'su' syntax ).

knnniggett, 2016-02-21:
You should be able to use nginx as long it has equivalents to Apache's Alias and Script Alias directives. It probably does. The important thing is the account name in which you run the web server under has to be the same account name you told ZoneMinder when you built it. The ZM_WEB_USER.

-DCMAKE_BUILD_TYPE=Debug build C debugging symbols which we don't need in this case.

Get the web portal working after installing zoneminder. From the web portal Options, turn on debug and enable debug output to the web log and component files. You don't have to have ZoneMinder "running" to do this.

Then try and start zoneminder. This time it should tell you exactly what line of Perl script is causing the error message.

rajil, 2016-02-21:
I decided to go with Apache for now. Following is the error i got in the apache log on accessing the webpage
```
[Sat Feb 20 20:36:27.720500 2016] [:error] [pid 45776] [client 192.16.1.1:35582] FAT [Can't write log entry 'INSERT INTO Logs ( TimeKey, Component, Pid, Level, Code, Message, File, Line ) values ( ?, ?, ?, ?, ?, ?, ?, ? )': SQLSTATE[22003]: Numeric value out of ra
nge: 1264 Out of range value for column 'Pid' at row 1]
```

connortechnology, 2016-03-04:
I would be tempted to turn off database logging and use syslog or component files instead.  

At least for now.

connortechnology, 2016-03-04:
also, try 
describe Logs
to see why an integer is an invalid value for Pid.

kylejohnson, 2016-03-07:
I'm about to run through a ZM install on my local box.  Will report any such woes here as well.

kylejohnson, 2016-03-11:
Dumping some notes in here re: the `su` error:
```
zm1 /usr/local/lib/perl5/site_perl/ZoneMinder >zmpkg.pl status
Testing valid shell syntax
Testing "sudo -u www true"
Test failed, ''
Testing "su www --shell=/bin/sh --command='true'"
Test failed, 'This account is currently not available.'
Testing "su www -c 'true'"
Test failed, 'This account is currently not available.'
```

The first test (with `sudo`) doesn't seem right.  It should have succeeded:
```
zm1 /usr/local/lib/perl5/site_perl/ZoneMinder >sudo -u www true
zm1 /usr/local/lib/perl5/site_perl/ZoneMinder >echo $?
0
```

Hmmm.

Played around some more, and it looks like the the `sudo` command, which on freebsd lives in `/usr/local/bin/`, vs. `sh`, which lives in `/bin/`, can't be found by perl:

```
zm1 /usr/local/lib/perl5/site_perl/ZoneMinder >zmpkg.pl status
Testing valid shell syntax
FUBAR Testing "sudo -u www true 2>&1 > /dev/null"
COMMAND: sudo -u www true 2>&1 > /dev/null
status: 32512
Test failed, 'sudo: not found'
Testing "su www --shell=/bin/sh --command='true'"
Test failed, 'This account is currently not available.'
Testing "su www -c 'true'"
Test failed, 'This account is currently not available.'
```

I'll submit a PR for this.

kylejohnson, 2016-03-11:
I'm going to close this thread out.  With the recent commits from @knnniggett and myself, ZM now installs just fine on FreeBSD.  There's some shoring up to do, but it is usable:
```
zm1 / >zmpkg.pl status
stopped
zm1 />zmpkg.pl start
zm1 / >zmpkg.pl status
running
```

closed 2016-03-11 by kylejohnson

17garcol17, 2016-04-14:
Hello. Commits in which the installation on FreeBSD is successful is already available for download? I when configuring - ZoneMinder requires MySQL headers - check that MySQL development packages are installed. Had a Similar problem with Gentoo users. https://forums.gentoo.org/viewtopic-t-1029214.html?sid=86227d856875f9d19456738f299ca1fd

knnniggett, 2016-04-14:
@17garcol17 Please do not double post. Since your issue is not a bug, but a configuration issue, the appropriate place to have this conversation is in the thread you already created in the user forum.