ZoneMinder/zoneminder #1290: Hikvision and onvif support in ZoneMinder-1.29

POKKAHOH, 2016-02-15:
Some trouble with onvif support on Hikvision DS-2CD833. Cam not detected by UI.

Run in term
`/usr/bin/zmonvif-probe.pl -v probe`
Answer
```
Probing for SOAP 1.1
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
Received message:
ARRAY(0x1898148)
Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message:
not well-formed (invalid token) at line 1, column 5, byte 5 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 79
 at line 1 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82
.
Message was:
ARRAY(0x1898148)</faultstring><faultactor>urn:localhost</faultactor></Fault>
Received message:
HASH(0x1898238)
Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message:
not well-formed (invalid token) at line 1, column 4, byte 4 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 79
 at line 1 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82
.
Message was:
HASH(0x1898238)</faultstring><faultactor>urn:localhost</faultactor></Fault>
Probing for SOAP 1.2
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
Received message:
HASH(0x1898238)
Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message:
not well-formed (invalid token) at line 1, column 4, byte 4 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 79
 at line 1 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82
.
Message was:
HASH(0x1898238)</faultstring><faultactor>urn:localhost</faultactor></Fault>
```
Run in term (default admin user and password, **witout** -v key)
`/usr/bin/zmonvif-probe.pl profiles "http://192.168.2.65/onvif/device_service" 1.2 admin 12345` 

Received message:
```
Profile_1, mainStream, H264, 640, 480, 2, rtsp://192.168.2.65:554/Streaming/Channels/1?transportmode=mcast&profile=Profile_1
Profile_2, subStream, H264, 352, 288, 25, rtsp://192.168.2.65:554/Streaming/Channels/2?transportmode=mcast&profile=Profile_2
```

with a key -v obtain additional information
`<GetProfilesResponse xmlns="http://www.onvif.org/ver10/media/wsdl"><Profiles token="Profile_1" fixed="true"><Name xmlns="http://www.onvif.org/ver10/schema">mainStream</Name><VideoSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoSourceConfig</Name><UseCount>2</UseCount><SourceToken>VideoSource_1</SourceToken><Bounds x="0" y="0" width="640" height="480"></Bounds></VideoSourceConfiguration><AudioSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>AudioSourceConfig</Name><UseCount>2</UseCount><SourceToken>AudioSourceChannel1</SourceToken></AudioSourceConfiguration><VideoEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoEncoder_1</Name><UseCount>1</UseCount><Encoding>H264</Encoding><Resolution><Width>640</Width><Height>480</Height></Resolution><Quality>3.000000</Quality><RateControl><FrameRateLimit>2</FrameRateLimit><EncodingInterval>1</EncodingInterval><BitrateLimit>2048</BitrateLimit></RateControl><H264><GovLength>4</GovLength><H264Profile>Baseline</H264Profile></H264><Multicast><Address><Type>IPv4</Type><IPv4Address>0.0.0.0</IPv4Address></Address><Port>8600</Port><TTL>1</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT5S</SessionTimeout></VideoEncoderConfiguration><AudioEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>AudioEncoderConfig</Name><UseCount>2</UseCount><Encoding>G711</Encoding><Bitrate>64</Bitrate><SampleRate>8</SampleRate><Multicast><Address><Type>IPv4</Type><IPv4Address>0.0.0.0</IPv4Address></Address><Port>8600</Port><TTL>1</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT5S</SessionTimeout></AudioEncoderConfiguration><VideoAnalyticsConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoAnalyticsName</Name><UseCount>2</UseCount><AnalyticsEngineConfiguration><AnalyticsModule Name="MyCellMotionModule" Type="tt:CellMotionEngine"><Parameters><SimpleItem Name="Sensitivity" Value="0"></SimpleItem><ElementItem Name="Layout"></ElementItem></Parameters></AnalyticsModule><AnalyticsModule Name="MyTamperDetecModule" Type="hikxsd:TamperEngine"><Parameters><SimpleItem Name="Sensitivity" Value="0"></SimpleItem><ElementItem Name="Transformation"></ElementItem><ElementItem Name="Field"></ElementItem></Parameters></AnalyticsModule></AnalyticsEngineConfiguration><RuleEngineConfiguration><Rule Name="MyMotionDetectorRule" Type="tt:CellMotionDetector"><Parameters><SimpleItem Name="MinCount" Value="5"></SimpleItem><SimpleItem Name="AlarmOnDelay" Value="100"></SimpleItem><SimpleItem Name="AlarmOffDelay" Value="100"></SimpleItem><SimpleItem Name="ActiveCells" Value="zwA="></SimpleItem></Parameters></Rule><Rule Name="MyTamperDetectorRule" Type="hikxsd:TamperDetector"><Parameters><ElementItem Name="Field"></ElementItem></Parameters></Rule></RuleEngineConfiguration></VideoAnalyticsConfiguration><PTZConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>PTZ</Name><UseCount>2</UseCount><NodeToken>PTZNODETOKEN</NodeToken><DefaultContinuousPanTiltVelocitySpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/VelocityGenericSpace</DefaultContinuousPanTiltVelocitySpace><DefaultContinuousZoomVelocitySpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/VelocityGenericSpace</DefaultContinuousZoomVelocitySpace><DefaultPTZTimeout>PT300S</DefaultPTZTimeout></PTZConfiguration><MetadataConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>metaData</Name><UseCount>2</UseCount><PTZStatus><Status>false</Status><Position>false</Position></PTZStatus><Analytics>false</Analytics><Multicast><Address><Type>IPv4</Type><IPv4Address>0.0.0.0</IPv4Address></Address><Port>8600</Port><TTL>1</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT5S</SessionTimeout><AnalyticsEngineConfiguration></AnalyticsEngineConfiguration></MetadataConfiguration><Extension xmlns="http://www.onvif.org/ver10/schema"></Extension></Profiles><Profiles token="Profile_2" fixed="true"><Name xmlns="http://www.onvif.org/ver10/schema">subStream</Name><VideoSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoSourceConfig</Name><UseCount>2</UseCount><SourceToken>VideoSource_1</SourceToken><Bounds x="0" y="0" width="640" height="480"></Bounds></VideoSourceConfiguration><AudioSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>AudioSourceConfig</Name><UseCount>2</UseCount><SourceToken>AudioSourceChannel1</SourceToken></AudioSourceConfiguration><VideoEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoEncoder_2</Name><UseCount>1</UseCount><Encoding>H264</Encoding><Resolution><Width>352</Width><Height>288</Height></Resolution><Quality>3.000000</Quality><RateControl><FrameRateLimit>25</FrameRateLimit><EncodingInterval>1</EncodingInterval><BitrateLimit>512</BitrateLimit></RateControl><H264><GovLength>25</GovLength><H264Profile>Baseline</H264Profile></H264><Multicast><Address><Type>IPv4</Type><IPv4Address>0.0.0.0</IPv4Address></Address><Port>8600</Port><TTL>1</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT5S</SessionTimeout></VideoEncoderConfiguration><AudioEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>AudioEncoderConfig</Name><UseCount>2</UseCount><Encoding>G711</Encoding><Bitrate>64</Bitrate><SampleRate>8</SampleRate><Multicast><Address><Type>IPv4</Type><IPv4Address>0.0.0.0</IPv4Address></Address><Port>8600</Port><TTL>1</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT5S</SessionTimeout></AudioEncoderConfiguration><VideoAnalyticsConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoAnalyticsName</Name><UseCount>2</UseCount><AnalyticsEngineConfiguration><AnalyticsModule Name="MyCellMotionModule" Type="tt:CellMotionEngine"><Parameters><SimpleItem Name="Sensitivity" Value="0"></SimpleItem><ElementItem Name="Layout"></ElementItem></Parameters></AnalyticsModule><AnalyticsModule Name="MyTamperDetecModule" Type="hikxsd:TamperEngine"><Parameters><SimpleItem Name="Sensitivity" Value="0"></SimpleItem><ElementItem Name="Transformation"></ElementItem><ElementItem Name="Field"></ElementItem></Parameters></AnalyticsModule></AnalyticsEngineConfiguration><RuleEngineConfiguration><Rule Name="MyMotionDetectorRule" Type="tt:CellMoti [... cut ...]
Profile_1, mainStream, H264, 640, 480, 2, rtsp://192.168.2.65:554/Streaming/Channels/1?transportmode=mcast&profile=Profile_1
Profile_2, subStream, H264, 352, 288, 25, rtsp://192.168.2.65:554/Streaming/Channels/2?transportmode=mcast&profile=Profile_2
`
Change ZoneMinder-1.29.0\web\skins\classic\views\onvifprobe.php 
so skip STEP1 and send to STEP2 correct IP address...

Excellent - we can see picture...

Question - how to change (learn how to change) the first request to the camera, in order to obtain from it the answer?

altaroca, 2016-03-15:
The "Error deserializing message: not well-formed (invalid token)" means that the received message cannot be interpreted according to the expected XML schema (XSD).
Currently the only way to debug this is to record the transferred HTTP packets with wireshark (or a similar tool). 
@POKKAHOH can you provide me with the packets your camera sends when probed with 
`zmonvif-probe.pl -v probe`

mattpepin, 2016-03-23:
I have a similar error with my Foscam FI9821W:

Probing for SOAP 1.1
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/vendor_perl/SOAP/WSDL/Client.pm line 205.
Probing for SOAP 1.2
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsdd="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd" xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" xmlns:wsc="http://schemas.xmlsoap.org/ws/2005/02/sc" xmlns:ds="http://www.w3.org/2000/09/xmldsig#" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tdn="http://www.onvif.org/ver10/network/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl"><SOAP-ENV:Header><wsa:MessageID>uuid:302169d6-9349-914e-98e6-d051d2ff3fee</wsa:MessageID><wsa:To SOAP-ENV:mustUnderstand="true">urn:schemas-xmlsoap-org:ws:2005:04:discovery</wsa:To><wsa:Action SOAP-ENV:mustUnderstand="true">http://schemas.xmlsoap.org/ws/2005/04/discovery/Hello</wsa:Action></SOAP-ENV:Header><SOAP-ENV:Body><wsdd:Hello><wsa:EndpointReference><wsa:Address>urn:uuid:88d87a05-d7f7-7e45-b1e9-00626E500D6E</wsa:Address></wsa:EndpointReference><wsdd:Types>tdn:NetworkVideoTransmitter</wsdd:Types><wsdd:Scopes>onvif://www.onvif.org/type/Network_Video_Transmitter onvif://www.onvif.org/type/video_encoder onvif://www.onvif.org/type/audio_encoder onvif://www.onvif.org/hardware/IPC-model onvif://www.onvif.org/name/IPC-model onvif://www.onvif.org/type/ptz onvif://www.onvif.org/location/country/china</wsdd:Scopes><wsdd:XAddrs>http://192.168.1.236:888/onvif/device_service</wsdd:XAddrs><wsdd:MetadataVersion>1</wsdd:MetadataVersion></wsdd:Hello></SOAP-ENV:Body></SOAP-ENV:Envelope>

Use of uninitialized value $result in concatenation (.) or string at /usr/bin/zmonvif-probe.pl line 109.
Error deserializing message:

cdragon, 2016-04-22:
The "Use of uninitialized value in concatenation (.)" part of this error seems to happen even if you probe an IP with no device at that address.  Ie: `zmonvif-probe.pl -v probe http://192.168.1.12:8899` produces:

```
Probing for SOAP 1.1
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
Probing for SOAP 1.2
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/SOAP/WSDL/Client.pm line 205.
```

There is no device at .1.12.  The Usage blurb zmonvif-probe.pl prints out isn't clear that you're supposed to start the parameter with http:// or include a port number so for awhile I was trying things like `zmonvif-probe.pl -v probe 192.168.1.11` which produces exactly the same output as above.

I did finally get this response when I probed a no-name IMX225 sensor camera board from China with the correct syntax `zmonvif-probe.pl -v probe http://192.168.1.11:8899` (where 8899 is the ONVIF port):

```
Probing for SOAP 1.1
Received message:
<?xml version="1.0" encoding="utf-8"?><soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsd="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof" xmlns:pub="http://schemas.microsoft.com/windows/pub/2005/07"><soap:Header><wsa:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</wsa:To><wsa:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Hello</wsa:Action><wsa:MessageID>urn:uuid:d724454b-cdc1-48e7-925b-25a744e6c6ae</wsa:MessageID><wsd:AppSequence InstanceId="2755" SequenceId="urn:uuid:94506137-106b-43ed-ad34-6ba9c88d5209" MessageNumber="77"></wsd:AppSequence></soap:Header><soap:Body><wsd:Hello><wsa:EndpointReference><wsa:Address>urn:uuid:c69f7c8c-cfe5-44ce-8c9c-47a55d291d90</wsa:Address></wsa:EndpointReference><wsd:Types>wsdp:Device pub:Computer</wsd:Types><wsd:MetadataVersion>35</wsd:MetadataVersion></wsd:Hello></soap:Body></soap:Envelope>
Error deserializing message. No message returned from deserializer.
Received message:
<?xml version="1.0" encoding="utf-8"?><soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsd="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof" xmlns:pub="http://schemas.microsoft.com/windows/pub/2005/07"><soap:Header><wsa:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</wsa:To><wsa:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Hello</wsa:Action><wsa:MessageID>urn:uuid:80acddc6-93a6-4945-a312-d96a864158f9</wsa:MessageID><wsd:AppSequence InstanceId="2755" SequenceId="urn:uuid:94506137-106b-43ed-ad34-6ba9c88d5209" MessageNumber="78"></wsd:AppSequence></soap:Header><soap:Body><wsd:Hello><wsa:EndpointReference><wsa:Address>urn:uuid:c69f7c8c-cfe5-44ce-8c9c-47a55d291d90</wsa:Address></wsa:EndpointReference><wsd:Types>wsdp:Device pub:Computer</wsd:Types><wsd:MetadataVersion>36</wsd:MetadataVersion></wsd:Hello></soap:Body></soap:Envelope>
Error deserializing message. No message returned from deserializer.
Received message:
<?xml version="1.0" encoding="utf-8"?><soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsd="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof" xmlns:pub="http://schemas.microsoft.com/windows/pub/2005/07"><soap:Header><wsa:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</wsa:To><wsa:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Hello</wsa:Action><wsa:MessageID>urn:uuid:80acddc6-93a6-4945-a312-d96a864158f9</wsa:MessageID><wsd:AppSequence InstanceId="2755" SequenceId="urn:uuid:94506137-106b-43ed-ad34-6ba9c88d5209" MessageNumber="78"></wsd:AppSequence></soap:Header><soap:Body><wsd:Hello><wsa:EndpointReference><wsa:Address>urn:uuid:c69f7c8c-cfe5-44ce-8c9c-47a55d291d90</wsa:Address></wsa:EndpointReference><wsd:Types>wsdp:Device pub:Computer</wsd:Types><wsd:MetadataVersion>36</wsd:MetadataVersion></wsd:Hello></soap:Body></soap:Envelope>
Error deserializing message. No message returned from deserializer.
Received message:
<?xml version="1.0" encoding="utf-8"?><soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsd="http://schemas.xmlsoap.org/ws/2005/04/discovery"><soap:Header><wsa:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</wsa:To><wsa:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Resolve</wsa:Action><wsa:MessageID>urn:uuid:ce7e7cdc-2340-4091-984b-078984c91ab2</wsa:MessageID></soap:Header><soap:Body><wsd:Resolve><wsa:EndpointReference><wsa:Address>urn:uuid:c69f7c8c-cfe5-44ce-8c9c-47a55d291d90</wsa:Address></wsa:EndpointReference></wsd:Resolve></soap:Body></soap:Envelope>
Error deserializing message. No message returned from deserializer.
Received message:
<?xml version="1.0" encoding="utf-8"?><soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsd="http://schemas.xmlsoap.org/ws/2005/04/discovery"><soap:Header><wsa:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</wsa:To><wsa:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Resolve</wsa:Action><wsa:MessageID>urn:uuid:ce7e7cdc-2340-4091-984b-078984c91ab2</wsa:MessageID></soap:Header><soap:Body><wsd:Reso [... cut ...]
A different no-name camera board from China with IMX178 sensor doesn't seem to respond to probing its ONVIF port using `zmonvif-probe.pl -v probe http://192.168.1.31:2000`.  It just returns the same response as probing a non-existent device.  On the other hand, it responds correctly to `zmonvif-probe.pl -v profiles http://192.168.1.31:2000 1.1 admin <password>`.  I won't share the response since it contains passwords but there are no errors.  I also tried using the command above with `probe` instead of `profiles` and got the same response as probing a non-existent device.

I get the same non-existent device response from probing an HIKvision DS-2CD2632F on port 8000.

altaroca, 2016-04-23:
We fixed some of these errors in #1346 and #1411. You would have to clone and build yourself however. 
The probing command takes no URL - anything after `probe` is ignored. It is just `zmonvif-probe.pl -v probe`.

closed 2017-03-18 by SteveGilvarry