ZoneMinder/zoneminder #1291: Upgrading cake-php that we bundle

pliablepixels, 2016-02-16:
ref: https://forums.zoneminder.com/viewtopic.php?f=34&t=24306
Looks like PHP7 is bundled with some systems and that's incompatible with CakePHP 2.5-RC1 which is what we bundle inside api/lib/Cake

@kylejohnson - I would need your help here - I tried the git cake upgrade tool but that threw a bunch of errors. So far I tried replacing 2.5 with 2.8 doing a simple
```
cd /usr/share/zoneminder/api/lib
mv Cake Cake_old
git clone -b 2.x git://github.com/cakephp/cakephp.git
ln -s cakephp/lib/Cake/ .
```
That works with PHP 5.4 but I don't have PHP7 to test. Specifically, I don't know if it will work if we don't also upgrade    the app dir via the tool.

onlyjob, 2016-02-17:
Related: #814

onlyjob, 2016-02-22:
https://secure.php.net/manual/en/migration70.php

By the way ZM interface works fine with PHP7 as far as I can tell. I did not test API though and I'm not sure how to do it. Do we have a test application to probe basic API functions?

knnniggett, 2016-02-25:
Resolved by #1306

closed 2016-02-25 by knnniggett