ZoneMinder/zoneminder #1294: zoneminder fails to start with root mounted ro, var is mounted rw

dantob, 2016-02-16:
zoneminder is failing to start when the root filesystem is mounted read only, var is mounted as writable. This is all I'm seeing in the logs

`-- Unit zoneminder.service has begun starting up.
Feb 16 22:33:24 eagle-eye zmpkg[548]: INF [Command: start]
Feb 16 22:33:24 eagle-eye zmdc[561]: INF [Server starting at 16/02/16 22:33:24]
Feb 16 22:33:27 eagle-eye zmdc[561]: FAT [Can't bind: Read-only file system]
Feb 16 22:33:34 eagle-eye zmdc[558]: FAT [Can't connect: No such file or directory]
Feb 16 22:33:35 eagle-eye zmpkg[548]: ERR [Unable to run "/usr/bin/zmdc.pl startup", output is "Starting server"]
Feb 16 22:33:35 eagle-eye systemd[1]: zoneminder.service: Control process exited, code=exited status=255
Feb 16 22:33:35 eagle-eye systemd[1]: Failed to start ZoneMinder CCTV recording and security system.
-- Subject: Unit zoneminder.service has failed`

dantob, 2016-02-16:
`rootfs / rootfs rw 0 0
proc /proc proc rw,nosuid,nodev,noexec,relatime 0 0
sys /sys sysfs rw,nosuid,nodev,noexec,relatime 0 0
dev /dev devtmpfs rw,nosuid,relatime,size=512640k,nr_inodes=128160,mode=755 0 0
run /run tmpfs rw,nosuid,nodev,relatime,mode=755 0 0
/dev/sdc2 / ext4 ro,relatime,data=ordered 0 0
securityfs /sys/kernel/security securityfs rw,nosuid,nodev,noexec,relatime 0 0
tmpfs /dev/shm tmpfs rw,nosuid,nodev 0 0
devpts /dev/pts devpts rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000 0 0
tmpfs /sys/fs/cgroup tmpfs ro,nosuid,nodev,noexec,mode=755 0 0
cgroup /sys/fs/cgroup/systemd cgroup rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/usr/lib/systemd/systemd-cgroups-agent,name=systemd 0 0
pstore /sys/fs/pstore pstore rw,nosuid,nodev,noexec,relatime 0 0
cgroup /sys/fs/cgroup/freezer cgroup rw,nosuid,nodev,noexec,relatime,freezer 0 0
cgroup /sys/fs/cgroup/cpuset cgroup rw,nosuid,nodev,noexec,relatime,cpuset 0 0
cgroup /sys/fs/cgroup/cpu,cpuacct cgroup rw,nosuid,nodev,noexec,relatime,cpuacct,cpu 0 0
cgroup /sys/fs/cgroup/blkio cgroup rw,nosuid,nodev,noexec,relatime,blkio 0 0
cgroup /sys/fs/cgroup/memory cgroup rw,nosuid,nodev,noexec,relatime,memory 0 0
cgroup /sys/fs/cgroup/net_cls cgroup rw,nosuid,nodev,noexec,relatime,net_cls 0 0
cgroup /sys/fs/cgroup/devices cgroup rw,nosuid,nodev,noexec,relatime,devices 0 0
systemd-1 /proc/sys/fs/binfmt_misc autofs rw,relatime,fd=32,pgrp=1,timeout=0,minproto=5,maxproto=5,direct 0 0
configfs /sys/kernel/config configfs rw,relatime 0 0
tmpfs /tmp tmpfs rw 0 0
mqueue /dev/mqueue mqueue rw,relatime 0 0
hugetlbfs /dev/hugepages hugetlbfs rw,relatime 0 0
debugfs /sys/kernel/debug debugfs rw,relatime 0 0
/dev/sdc1 /boot ext2 ro,relatime 0 0
/dev/mapper/varvg-var /var ext4 rw,relatime,data=ordered 0 0
tmpfs /run/user/0 tmpfs rw,nosuid,nodev,relatime,size=103004k,mode=700 0 0
`

knnniggett, 2016-02-16:
This does not describe a bug, but a configuration error.

`FAT [Can't bind: Read-only file system] `
That means you configured ZoneMinder to create its socket files somewhere under your root filesystem.

If you need further assistance, please read our posting rules:
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

Then create a post in our user forum, which describes exactly how you built and installed ZoneMinder.

closed 2016-02-16 by knnniggett

connortechnology, 2016-02-16:
I have made a PR that improves the error logs to tell us where it is trying to create these sock files.

dantob, 2016-02-17:
`    cmake -DCMAKE_INSTALL_PREFIX=/usr \
          -DZM_PERL_SUBPREFIX=/lib/perl5 \
          -DZM_WEBDIR=/srv/http/zoneminder \
          -DZM_CGIDIR=/srv/http/cgi-bin \
          -DZM_WEB_USER=http \
          -DZM_CONTENTDIR=/var/cache/zoneminder \
          -DZM_LOGDIR=/var/log/zoneminder \
          -DZM_RUNDIR=/run/zoneminder \
          -DZM_TMPDIR=/var/lib/zoneminder/temp \
          -DZM_SOCKDIR=/var/lib/zoneminder/sock .`

[daniel@eagle-eye zoneminder]$ ls -la /srv/zoneminder/
total 16
drwxr-xr-x 4 http http 4096 Oct 24 13:22 .
drwxr-xr-x 5 root root 4096 Oct 24 13:23 ..
drwxr-xr-x 2 http http 4096 Feb 17 14:15 socks
drwxr-xr-x 4 http http 4096 Oct 27 14:59 tmp

something odd is happening here, who is creating this directory? (package file list [http://hastebin.com/xipujenebu.pas](http://hastebin.com/xipujenebu.pas)

`zmdc.pl 28229 http    3u  unix 0xf4e4c000      0t0    39762 /srv/zoneminder/socks/zmdc.sock type=STREAM
zmdc.pl 28229 http    4u  unix 0xf56bb680      0t0    41404 /srv/zoneminder/socks/zmdc.sock type=STREAM`

I am using the archlinux AUR package, you can find more info here [https://aur.archlinux.org/cgit/aur.git/tree/?h=zoneminder](https://aur.archlinux.org/cgit/aur.git/tree/?h=zoneminder)

(I realise you asked me to move to the forums, but something definately seems bugged here)

knnniggett, 2016-02-17:
We certainly aren't going to provide free help to those who don't care to follow the rules. These rules are in pace to keep our github forum from turning into a circus of issues which have nothing to do with improving the source code. Thread locked.