ZoneMinder/zoneminder #1299: Ubuntu ( zoneminder error ) Beginner

JasminSAB, 2016-02-19:
Hello Guys, I'm first time on this website, just google it, followed this guide to setup a zoneminder, on my virtubalbox server, installation of linux server i have been using was : ubuntu-14.04.3-server-amd64 
Guide i was following : https://github.com/pliablepixels/zmNinja/wiki/Easy-Way:-Ubuntu-14,-Install-ZoneMinder-API-from-Package

Note, I'm using linux 2 days. 
What I want is to setup a zoneminder with D link 930 l camera, to make them record in pictures ofc, video no sound. 
I'm using right now D-Link router , fixed ip of my camera to static 192.168.0.56 

The error i got in log on my is this one : 

date/time
2016-02-19 01:48:50.887701	
Component
web_js
Server 
( nothing ) 
Level
ERR
Message
Uncaught TypeError: Cannot read property 'isDisplayed' of undefined	
File 
http://192.168.0.102/zm/tools/mootools/mootools-more.js	
Line
1 

If you people can help me I would really appreciate It. 
If You got any tips like to build any system (linux pref ubuntu ), on one pc ( without using Oracle VirtualBox ) and guide from start, how to setup system from start, to end, and how to setup zoneminder I would be thankful to You whole my life. Since if You people help me, this would make me find a job in decent company in Bosnia and Herzegovina, to work for them. 

Thank You very much
with full respect Jasmin

JasminSAB, 2016-02-19:
Once I open http://192.168.0.56/zm/api it says this : 
Error: File Not Found

getfile: Cannot open URL(/etc_ro/web/zm/api,No such file or directory)

Also once I open http://192.168.0.102/zm/api/host/getVersion.json it says this : Warning: mkdir(): File exists in /usr/share/zoneminder/www/api/lib/Cake/Cache/Engine/FileEngine.php on line 379

Warning: _cake_core_ cache was unable to write 'cake_dev_en-us' to File cache in /usr/share/zoneminder/www/api/lib/Cake/Cache/Cache.php on line 323

Warning: /usr/share/zoneminder/www/api/app/tmp/cache/persistent/ is not writable in /usr/share/zoneminder/www/api/lib/Cake/Cache/Engine/FileEngine.php on line 384

Fatal error: Uncaught exception 'CacheException' with message 'Cache engine _cake_core_ is not properly configured.' in /usr/share/zoneminder/www/api/lib/Cake/Cache/Cache.php:181 Stack trace: #0 /usr/share/zoneminder/www/api/lib/Cake/Cache/Cache.php(151): Cache::_buildEngine('_cake_core_') #1 /usr/share/zoneminder/www/api/app/Config/core.php(376): Cache::config('_cake_core_', Array) #2 /usr/share/zoneminder/www/api/lib/Cake/Core/Configure.php(72): include('/usr/share/zone...') #3 /usr/share/zoneminder/www/api/lib/Cake/bootstrap.php(175): Configure::bootstrap(true) #4 /usr/share/zoneminder/www/api/app/webroot/index.php(90): include('/usr/share/zone...') #5 {main} thrown in /usr/share/zoneminder/www/api/lib/Cake/Cache/Cache.php on line 181

Do I need this API at all? How do I set my user and password for zoneminder would be great to know. Thank You very much for reading my topic, thank You.

knnniggett, 2016-02-19:
Hi @JasminSAB 
The ZoneMinder user forum at forums.zoneminder.com is the place for technical support questions.

closed 2016-02-19 by knnniggett

JasminSAB, 2016-02-19:
Hi @knnniggett T
Thank You Sir.

antares2001, 2016-04-27:
Can you please link to the forum entry? Google brings me here but not on the forum post.