ZoneMinder/zoneminder #1307: zoneminder.com: outdated downloads information

onlyjob, 2016-02-26:
On https://www.zoneminder.com/downloads/ the following text could be seen:

> If you are a Debian user, Zoneminder is available in the "unstable" distribution. Thanks to Peter Howard for his considerable efforts in achieving this.

I'd like to suggest some corrections please:

##### Remove mentioning of `"unstable" distribution`.

We do not recommend to install from "unstable" suite -- that's why it is called "unstable".

It is better to include the following link to show packages in all suites: https://packages.debian.org/search?keywords=zoneminder&searchon=sourcenames&suite=all&section=all or give a link to "PTS" page: https://tracker.debian.org/pkg/zoneminder

##### Drop outdated credits to Peter Howard.

Although in the past Peter introduced ZoneMinder to Debian for the first time, his packages then decayed and was eventually removed even from "unstable". Yours truly subsequently re-introduced heavily refurbished ZoneMinder packaging. Please note that IMHO it is unnecessary to give credits for this work in "downloads" section of the zoneminder.com web site. Link to packages.debian.org or PTS page is sufficient to find history of package maintenance...

By the way who is controlling zoneminder.com?

SteveGilvarry, 2016-02-26:
New website is under development so not much on old .com site being changed.

http://zoneminder.github.io/downloads/

Option to submit your own content as Debian is currently lore ipsum
https://github.com/ZoneMinder/zoneminder.github.io/blob/master/downloads.html

onlyjob, 2016-02-26:
Thank you.

zoneminder.github.io have some privacy issues though as it fetches resources directly from 

 * fonts.googleapis.com
 * bootstrapcdn.com
 * gaug.es
 * pingdom.net

and do not render properly when 3rd party domains are not available...

kylejohnson, 2016-02-26:
By the time the site is launched, I plan to serve more of the assets locally (such as the CSS and JavaScript).

Pingdom and gauges are there to stay, however.  I might even add Google analytics.

As a user, I hear you.  I run NoScript and I block most of what Google, Facebook, etc serve.

As the guy running the site and trying to drive more users, gauges / Google analytics provides valuable information.

Pingdom helps to ensure that I know when the site is down, unresponsive, or slow.

Please do submit a PR for the new downloads page.  We could use the help.

Kyle

connortechnology, 2016-02-26:
I roll my own pingdom style service.  We could always use it if pingdom became too offensive.

SteveGilvarry, 2016-07-27:
New website active, privacy discussion outside the scope of zoneminder.

closed 2016-07-27 by SteveGilvarry