ZoneMinder/zoneminder #1318: Ошибка timezone

azerikt, 2016-03-02:
Warning: gettimeofday(): It is not safe to rely on the system's timezone settings. You are *required* to use the date.timezone setting or the date_default_timezone_set() function. In case you used any of those methods and you are still getting this warning, you most likely misspelled the timezone identifier. We selected the timezone 'UTC' for now, but please set date.timezone to select your timezone. in /usr/share/zoneminder/www/includes/logger.php on line 412

Warning: strftime(): It is not safe to rely on the system's timezone settings. You are *required* to use the date.timezone setting or the date_default_timezone_set() function. In case you used any of those methods and you are still getting this warning, you most likely misspelled the timezone identifier. We selected the timezone 'UTC' for now, but please set date.timezone to select your timezone. in /usr/share/zoneminder/www/includes/logger.php on line 413
ZoneMinder is not installed properly: php's date.timezone is not set to a valid timezone

в php.ini указан Europe/Helsinki

Что делать с такой ошибкой?

knnniggett, 2016-03-02:
This is configuration error, not a bug report. To fix it, you need to do what the error message says and set date.timezone.

Please following our posting guidelines and take this to the user forum. Thank you.
https://github.com/ZoneMinder/ZoneMinder/wiki/Github-Posting-Rules

closed 2016-03-02 by knnniggett

azerikt, 2016-03-02:
я прощу прощения, но tomezone настроен корректно!

knnniggett, 2016-03-02:
You might think your php timezone is configured correctly, but it is not. The data does not lie.
If you want assistance then create a post in the user forum. Locking thread.