ZoneMinder/zoneminder #1322: Add input checking / validation when creating a group of monitors.

bodegard, 2016-03-03:
Currently, it allows you to attempt to create a group, then press OK, but if you do not have any monitors selected it gives and SQL error. I think there should be some kind of validation that alerts the user that no monitors are selected, or allow it to create an empty group to add to later.

`SQL-ERR 'SQLSTATE[23000]: Integrity constraint violation: 1048 Column 'MonitorIds' cannot be null', statement was 'INSERT INTO Groups SET Name=?, MonitorIds=?'`

knnniggett, 2016-03-03:
Oddly enough, I just mentioned in #1317 that I seem to recall some place in the web console where the buttons lit up incorrectly. I think this was it.
If memory serves, there should be some existing javascript that controls the state of the buttons on that popup. Guess it isn't working right. Will have to look into it.

knnniggett, 2016-03-09:
@bodegard Please test #1326

bodegard, 2016-03-09:
@knnniggett #1326 That fixed the issue. Thanks

PR #1326 (2016-03-09): disable save button when nothing slected. fixes #1322
Fixes #1322 on my machine. Please test.
commit 3280aff8 (2016-03-09): disable save button when nothing slected. fixes #1322
commit d51d95ce (2016-03-09): Merge pull request #1326 from knnniggett/fixGroupSaveBtn

disable save button when nothing slected. fixes #1322
closed 2016-03-09 by connortechnology