ZoneMinder/zoneminder #1341: Install problem from zmrepo on Scientific Linux 6.7  releasever

Atheror, 2016-03-16:
Following instructions from: https://zmrepo.zoneminder.com/
EL6 (CentOS/RHEL/**SL**)
Supposed to be the same on Centos, Rhel or Scientific Linux

Installation instructions for EL6 distributions (i386 or x86_64 architectures)

```
wget http://zmrepo.zoneminder.com/el/6/i386/zmrepo-6-7.el6.noarch.rpm
sudo yum install --nogpgcheck zmrepo-6-7.el6.noarch.rpm
sudo yum install zoneminder
less /usr/share/doc/zoneminder-*/README.CentOS
```

When I try to install Zoneminder I get this error:
```

[root@sumlnxfs015 ~]# sudo yum install zoneminder
Complementos cargados:security
Configurando el proceso de instalación
http://zmrepo.connortechnology.com/el/6.7/x86_64/repodata/repomd.xml: [Errno 14] PYCURL ERROR 22 - "The requested URL returned error: 404 Not Found"
Intentando con otro espejo.
http://67.53.78.182/el/6.7/x86_64/repodata/repomd.xml: [Errno 14] PYCURL ERROR 22 - "The requested URL returned error: 404 Not Found"
Intentando con otro espejo.
zmrepo                                                                                                                                          |  13 kB     00:00 ... 
http://zmrepo.mehvc.com/el/6.7/x86_64/repodata/repomd.xml: [Errno -1] Error importing repomd.xml for zmrepo: Damaged repomd.xml file
Intentando con otro espejo.
Error: Cannot retrieve repository metadata (repomd.xml) for repository: zmrepo. Please verify its path and try again

```
I checked and those URL doesnt exist, because yum repo file is using $releasever and on my SL6 it is 6.7 right now.

FIXED I have to modify zoneminder repo file manually to change **$releasever** to  **6** what it should be on Centos and RHEL I suppose.
Now URL are OK and install process continues as usual.

Another options shoul be to use **--releasever=6**  parameter on yum command each time

knnniggett, 2016-03-16:
Sounds like SL includes the minor release value in the $releasever variable. Redhat and CentOS did not start doing this until release 7. I will add symlinks to zmrepo so yum can find the files it needs for release 6.x. The mirrors will sync overnight so this should be fixed tomorrow.

knnniggett, 2016-03-17:
symlnks are in now in place which should fix this.

closed 2016-03-17 by knnniggett