ZoneMinder/zoneminder #1345: Zoneminder ONVIF perl dependencies

altaroca, 2016-03-19:
<forwarded  on behalf of another user>
Installed Zm 1.29 from ppa and got the “ONVIF” link in “add new monitor menu”
http://192.168.1.109/zm/index.php?view=onvifprobe&mid=0

but when pressed i get this:
Unable to probe network cameras, status is ‘2’

partially solved it by installing the following packages:

libclass-std-fast-perl
libdata-dump-perl
libsoap-wsdl-perl
libdigest-sha-perl
libdigest-perl
libio-socket-multicast-perl

In addition, ‘/usr/share/perl5/WSSecurity/SecuritySerializer.pm’ must be edited to use the ‘SHA’ package instead of ‘SHA1’ (since Ubuntu doesn’t have a SHA1 perl module as of at least 12.04).

I guess we should check for these packages before build and installation.

knnniggett, 2016-03-19:
Currently, the only way to get ZoneMinder with Onvif support is to build from the master branch. That is why the other user has run into this issue. Today, the only way to get ZoneMinder with Onvif support is to build from the master branch. Alternatively, the user can wait until the next release. I am hopeful to determine when that will be Monday evening.

We have already addressed the issue with the SHA1 module in #1319 but again that won't make it to the end user until after the next release.

closed 2016-03-19 by knnniggett

Toasty27, 2016-03-19:
I'm the user who reported this issue (originally on @altaroca 's blog)

I should clarify, I built a ZoneMinder deb package from source with the ZM_ONVIF=OK flag, which is why I had the OVNIF option available. The Probe function wasn't picking up my cameras, so I figured I'd try the OVNIF probe (even though it's  experimental). Wasn't looking forward to manually adding 50+ cameras, although since they're all the same, I can probably script it with the API and curl.

Anyway, thanks for the info. I'll keep an eye out for the next update.

knnniggett, 2016-03-19:
@toasty27 We only recently added the necessary dependencies to the Debian packaging scripts. If you update your local copy of master and rebuild a new package, the issues mentioned in this issue should be gone.

To answer your question about manually adding identical cameras, you should look at zmcamtool.pl. You use zmcamtool.pl to create a preset from the first camera, and then use that preset for the other 49 cameras.