ZoneMinder/zoneminder #1346: ONVIF probing camera not detected

altaroca, 2016-03-19:
(forwarded on behalf of another user)
The ONVIF probing window opens and displays properly, but none of my cameras are discovered. Running 
`zmonvif-probe.pl -v probe` 
in terminal gives me the following output: 
`Probing for SOAP 1.1
Received message:
ARRAY(0x252ef50)
Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message:
not well-formed (invalid token) at line 1, column 5, byte 5 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 79.
 at line 1 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82.
.
Message was:
ARRAY(0x252ef50)</faultstring><faultactor>urn:localhost</faultactor></Fault>
Received message:
HASH(0x252f028)
Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Server</faultcode><faultstring>Error deserializing message:
not well-formed (invalid token) at line 1, column 4, byte 4 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 79.
 at line 1 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82.
.
Message was:
HASH(0x252f028)</faultstring><faultactor>urn:localhost</faultactor></Fault>
Probing for SOAP 1.2
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Fault xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" 
 xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" 
 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" 
 xmlns:xsd="http://www.w3.o rg/2001/XMLSchema" 
 xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" 
 xmlns:wsdd="http://schemas.xmlsoap.org/ws/2005/04/discovery" 
 xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" 
 xmlns:wsa5= "http://www.w3.org/2005/08/addressing" 
 xmlns:xmime="http://www.w3.org/2005/05/xmlmime" 
 xmlns:xop="http://www.w3.org/2004/08/xop/include" 
 xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2"
 xmlns:tt="http://www.on vif.org/ver10/schema" 
 xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" 
 xmlns:wsrfr="http://docs.oasis-open.org/wsrf/r-2" 
 xmlns:ns1="http://www.onvif.org/ver10/actionengine/wsdl" 
 xmlns:tad="http://www.onvif.org/ve r10/analyticsdevice/wsdl" 
 xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" 
 xmlns:tdn="http://www.onvif.org/ver10/network/wsdl" 
 xmlns:tds="http://www.onvif.org/ver10/device/wsdl" 
 xmlns:tev="http://www.onvif.o rg/ver10/events/wsdl" 
 xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" 
 xmlns:c14n="http://www.w3.org/2001/10/xml-exc-c14n#" 
 xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0 .xsd" 
 xmlns:xenc="http://www.w3.org/2001/04/xmlenc#" 
 xmlns:wsc="http://schemas.xmlsoap.org/ws/2005/02/sc" 
 xmlns:ds="http://www.w3.org/2000/09/xmldsig#" 
 xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200 401-wss-wssecurity-secext-1.0.xsd" 
 xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" 
 xmlns:tls="http://www.onvif.org/ver10/display/wsdl" 
 xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" 
 xmlns:tptz="http:// www.onvif.org/ver20/ptz/wsdl" 
 xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" 
 xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" 
 xmlns:trt="http://www.onvif.org/ver10/media/wsdl" 
 xmlns:trv="http://www.onvif .org/ver10/receiver/wsdl" 
 xmlns:ter="http://www.onvif.org/ver10/error" 
 xmlns:tse="http://www.onvif.org/ver10/search/wsdl" 
 xmlns:tns1="http://www.onvif.org/ver10/topics">
 <faultcode>wsa:ActionNotSupported</faultcode>
 <faultstring>The [action] cannot be processed at the receiver.</faultstring>
 </SOAP-ENV:Fault>
Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/">
<faultcode>SOAP-ENV:Server</faultcode>
<faultstring>Error deserializing message: Bad top node Fault
 at /usr/share/perl5/ONVIF/Deserializer/MessageParser.pm  line 66.
 at line 2 at /usr/share/perl5/SOAP/WSDL/Expat/Base.pm line 82.
 `

Any ideas? Is this an issue with the cameras? Becasue ONVIF Device Manager finds all 50+ of my cameras, and only a few of them don’t load properly.

knnniggett, 2016-03-19:
I have not had a lot of time to really dig deep into the Onvif library or test it thoroughly. I've got several different Onvif compliant cameras, all of which the Windows based Onvif Device Manage can browse and control.

My train of thought with troubleshooting lies with this portion of the message:
`Received message: ARRAY(0x252ef50)`

In addition to that, I also seen:
```
Use of uninitialized value $xaddr in hash element at /bin/zmonvif-probe.pl line 128.
Use of uninitialized value $xaddr in hash element at /bin/zmonvif-probe.pl line 129.
Use of uninitialized value $xaddr in concatenation (.) or string at /bin/zmonvif-probe.pl line 131.
```
These make me wonder if the code should be dereferencing a variable prior to operating on it. For example, I was last looking at the interpret_messages sub:
https://github.com/ZoneMinder/ZoneMinder/blob/master/onvif/scripts/zmonvif-probe.pl#L98

This sub is receiving a reference to the responses array and a reference to the services hash. It is my understanding that the following line does not dereference these values:
```perl
my ($svc_discover, @responses, %services) = @_;
```
Instead, it places the scalar reference value into @references and %services, rather than the actual array and hash. Without having spent any time testing this, that is what I think is happening.

Instead, shouldn't it look something like the following?
```perl
my ($svc_discover, $responses_ref, $services_ref) = @_;
my @responses = @$responses_ref
my %services = %$services_ref
```

I have not looked to see if there are other places which might need a similar change.

I had not known yo have tested this on more than just a single camera. Is there a list somewhere that indicates which cameras have been tested? Perhaps I might own one of those cameras and could use the same camera to help further isolate the cause of this issue.

altaroca, 2016-03-19:
You have a point with the references. 

I do not have list. Maybe I should start one in the zm wiki?

I was looking at this part of the message: 
`The [action] cannot be processed at the receiver`
The fault message uses the WS-Discovery 1.0 namespace "http://schemas.xmlsoap.org/ws/2005/04/discovery". The proxies in zm currently use WS-Discovery 1.1 ("http://docs.oasis-open.org/ws-dd/ns/discovery/2009/01"). This should be changed since the ONVIF specs cite the WS-Discovery 1.0 namespace.
I will try to generate the proxies for WS-Discovery 1.0 instead.

jburgess777, 2016-03-19:
I have fixes for some discovery problem I found with a Tenvis TH661 in #1347. These changes fix one case of a SOAP fault and access to uninitialized xaddr if no address can be parsed from the output.

jburgess777, 2016-03-19:
> This sub is receiving a reference to the responses array and a reference to the services hash

Yes this is causing problems. I added another commit to #1347 which seems to make the discovery and duplicate service detection work as intended but I only have a single camera to test.

Toasty27, 2016-03-19:
I'm the user who originally reported this on @altaroca 's blog.

From what I understand, the model in question is TOPSEE's TH38A2-ONVIF. There are also a few T38J4-ONVIF's in use as well. It's a Chinese manufactured h264 1080p full color camera in both cases. They both show support for ONVIF 2.04 in the ONVIF Device Manager utility.

knnniggett, 2016-03-27:
@altaroca I have several Onvif compliant cameras of various makes and models on my network, and with the latest master branch, the onvif probe is now able to discover one camera, so we seem to be making progress.

```
Probing for SOAP 1.1
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/" xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wsdd="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:chan="http://schemas.microsoft.com/ws/2005/02/duplex" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:xmime="http://tempuri.org/xmime.xsd" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:wsrfbf="http://docs.oasis-open.org/wsrf/bf-2" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:ns3="http://www.onvif.org/ver10/pacs" xmlns:wsrfr="http://docs.oasis-open.org/wsrf/r-2" xmlns:ns1="http://www.onvif.org/ver10/actionengine/wsdl" xmlns:ns2="http://www.onvif.org/ver10/accesscontrol/wsdl" xmlns:ns4="http://www.onvif.org/ver10/doorcontrol/wsdl" xmlns:tad="http://www.onvif.org/ver10/analyticsdevice/wsdl" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:tdn="http://www.onvif.org/ver10/network/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tls="http://www.onvif.org/ver10/display/wsdl" xmlns:tmd="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trc="http://www.onvif.org/ver10/recording/wsdl" xmlns:trp="http://www.onvif.org/ver10/replay/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:trv="http://www.onvif.org/ver10/receiver/wsdl" xmlns:tse="http://www.onvif.org/ver10/search/wsdl" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns:wsse="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-secext-1.0.xsd" xmlns:wsu="http://docs.oasis-open.org/wss/2004/01/oasis-200401-wss-wssecurity-utility-1.0.xsd"><SOAP-ENV:Body><SOAP-ENV:Fault><faultcode>SOAP-ENV:Client</faultcode>
</SOAP-ENV:Fault>
</SOAP-ENV:Body>
</SOAP-ENV:Envelope>

Error deserializing message:
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>SOAP-ENV:Client</faultcode></Fault>
Received message:
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://schemas.xmlsoap.org/soap/envelope/" xmlns:SOAP-ENC="http://schemas.xmlsoap.org/soap/encoding/" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wsa5="http://www.w3.org/2005/08/addressing" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:tt="http://www.onvif.org/ver10/schema" xmlns:ns1="http://www.w3.org/2005/05/xmlmime" xmlns:wstop="http://docs.oasis-open.org/wsn/t-1" xmlns:ns7="http://docs.oasis-open.org/wsrf/r-2" xmlns:ns2="http://docs.oasis-open.org/wsrf/bf-2" xmlns:dndl="http://www.onvif.org/ver10/network/wsdl/DiscoveryLookupBinding" xmlns:dnrd="http://www.onvif.org/ver10/network/wsdl/RemoteDiscoveryBinding" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:dn="http://www.onvif.org/ver10/network/wsdl" xmlns:ns10="http://www.onvif.org/ver10/replay/wsdl" xmlns:ns11="http://www.onvif.org/ver10/search/wsdl" xmlns:ns13="http://www.onvif.org/ver20/analytics/wsdl/RuleEngineBinding" xmlns:ns14="http://www.onvif.org/ver20/analytics/wsdl/AnalyticsEngineBinding" xmlns:tan="http://www.onvif.org/ver20/analytics/wsdl" xmlns:ns15="http://www.onvif.org/ver10/events/wsdl/PullPointSubscriptionBinding" xmlns:ns16="http://www.onvif.org/ver10/events/wsdl/EventBinding" xmlns:tev="http://www.onvif.org/ver10/events/wsdl" xmlns:ns17="http://www.onvif.org/ver10/events/wsdl/SubscriptionManagerBinding" xmlns:ns18="http://www.onvif.org/ver10/events/wsdl/NotificationProducerBinding" xmlns:ns19="http://www.onvif.org/ver10/events/wsdl/NotificationConsumerBinding" xmlns:ns20="http://www.onvif.org/ver10/events/wsdl/PullPointBinding" xmlns:ns21="http://www.onvif.org/ver10/events/wsdl/CreatePullPointBinding" xmlns:ns22="http://www.onvif.org/ver10/events/wsdl/PausableSubscriptionManagerBinding" xmlns:wsnt="http://docs.oasis-open.org/wsn/b-2" xmlns:ns3="http://www.onvif.org/ver10/analyticsdevice/wsdl" xmlns:ns4="http://www.onvif.org/ver10/deviceIO/wsdl" xmlns:ns5="http://www.onvif.org/ver10/display/wsdl" xmlns:ns8="http://www.onvif.org/ver10/receiver/wsdl" xmlns:ns9="http://www.onvif.org/ver10/recording/wsdl" xmlns:tds="http://www.onvif.org/ver10/device/wsdl" xmlns:timg="http://www.onvif.org/ver20/imaging/wsdl" xmlns:tptz="http://www.onvif.org/ver20/ptz/wsdl" xmlns:trt="http://www.onvif.org/ver10/media/wsdl" xmlns:ter="http://www.onvif.org/ver10/error" xmlns:tns1="http://www.onvif.org/ver10/topics" xmlns:tnsn="http://www.eventextension.com/2011/event/topics"><SOAP-ENV:Body><SOAP-ENV:Fault><faultcode>SOAP-ENV:Client</faultcode><faultstring>End of file or no input: Operation interrupted or timed out</faultstring></SOAP-ENV:Fault></SOAP-ENV:Body></SOAP-ENV:Envelope>
Error deserializing message:
[... cut ...]
The camera rejects the initial SAOP 1.1 message with a version mismatch, but then it responds with an "Operation interrupted or timed out" message to the SOAP 1.2 message. If it helps, I can show what the wireshark data looks like when I successfully probe the camera with Onvif Device Manager software.

The other thing to notice is the perl script error hiding in there:
```
Use of uninitialized value $result in concatenation (.) or string at /bin/zmonvif-probe.pl line 109.
Error deserializing message:
```

Looks like the deserialize_message subroutine is returning undef in this case?

knnniggett, 2016-03-28:
related #1368

jburgess777, 2016-03-29:
I spent some time trying to improve the zmonvif-probe.pl script but kept running into problems getting it to generate the discovery messages with the right content. As a speculative alternative, I put together a set of C++ tools using gsoap which perform the same functions as the original script. There is also  a small wrapper to emulate the existing API. With this I can successfully probe all my cameras and read the profile data:
```
$ zmonvif-probe.pl probe
http://192.168.1.153:8080/onvif/devices, 1.1, ()
http://192.168.1.150:8899/onvif/device_service, 1.1, ()
http://192.168.1.152:5000/onvif/device_service, 1.1, ()
http://192.168.1.151:8000/onvif/device_service, 1.2, ()

$ zmonvif-probe.pl profiles http://192.168.1.153:8080/onvif/devices 1.1 user password
MainProfileToken, MainProfile, H264, 1280, 720, 25, rtsp://192.168.1.153:554/11
SubProfileToken, SubProfile, H264, 640, 352, 10, rtsp://192.168.1.153:554/12
```
If anyone has problems discovering cameras then it might be interesting to try these as an alternative. The following steps will build the tools and copy them to ```/usr/local/bin```
* Install gsoap development tools, e.g. ```sudo yum install gsoap-devel```
```
git clone -b zmonvif-probe https://github.com/jburgess777/ws-discovery.git
cd ws-discovery/gsoap
make
sudo make install
```
If you want to overwrite the ```/usr/bin/zmonvif-probe.pl``` provided by ZoneMinder then you can install in /usr:
```
sudo make install PREFIX=/usr
```

SteveGilvarry, 2016-03-29:
@jburgess777 beat me to it, going to check this out for sure.

altaroca, 2016-04-05:
It's great to see so much activity on this topic!

@knnniggett I think your camera actually responds to the SOAP 1.2 probing. I can see a ProbeMatches XML element and this result:
`http://192.168.1.84:8999/onvif/device_service, 1.2, (type= ...`

It seems that errors in the Perl script lead to the response not being recognized. The "End of file or no input: Operation interrupted or timed out" message may well be the XML parser's answer to incomplete input from the script.

I will look into this on the weekend.

altaroca, 2016-04-09:
PR #1408 should make more cameras recognize the probing request. 
@knnniggett Please try this on your cameras. If this is not enough then I have another commit that forces utf-8 encoding. That might help as well.

jburgess777, 2016-04-09:
PR #1410 helps detect one more type of camera for me. It still misses two of the cameras detected by the gsoap code I mentioned earlier. I spent some time looking at the differences in the probe messages between the Perl and gsoap but ran into problems trying to convince the Perl code to generate the same messages.

siigna, 2016-06-16:
Hopefully this is the right spot.  I'm getting some syntax and "Global symbol requires explicit package name" errors.
Debian wheezy with Perl v5.14.2 (Debian v5.14.2-21+deb7u3)

```
# ./zmonvif-probe.pl probe
syntax error at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 95, near "}{"
Global symbol "$attr" requires explicit package name at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 96.
Global symbol "$value" requires explicit package name at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 96.
Global symbol "$attr_str" requires explicit package name at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 99.
syntax error at /usr/share/perl5/WSDiscovery10/Types/ProbeType.pm line 102, near "}"
Compilation failed in require at (eval 90) line 2.
        ...propagated at /usr/share/perl/5.14/base.pm line 93.
BEGIN failed--compilation aborted at /usr/share/perl5/WSDiscovery10/Elements/Probe.pm line 18.
Compilation failed in require at (eval 89) line 2.
```

Any ideas?

connortechnology, 2016-08-10:
@jburgess777 I wanted to post an issue with your gsoap ws-discovery tree, but github isn't giving me the option to do so... 

I tried it out, but it doesn't detect my TrendNet TV-IP862-IC.

rhaamo, 2017-01-04:
I'm using ZM 1.30 and my camera isn't detected by the probe.
Using ```ONVIF Device Manager``` no problems to get infos for it but the ```probe``` doesn't list anything.
However ```zmonvif-probe.pl profiles http://192.168.0.48:8899/onvif/device_service 1.1 user password``` correctly show three profiles.
The probe from @jburgess777 doesn't detect the camera too, but ```profiles``` or ```info``` are ok.

What can I provides to improve this ?

connortechnology, 2017-01-04:
@rhaamo I made some improvements in master that aren't in 1.30.  You might give it a try.

rhaamo, 2017-01-04:
@connortechnology thanks, I've built the package from master and installed but that doesn't seems to be better:
```
root@ananas:~/zm# zmonvif-probe.pl -v probe
Probing for SOAP 1.1
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>soap:Server</faultcode><faultstring>Error sending / receiving message: Timed out waiting for response</faultstring><faultactor>urn:localhost</faultactor></Fault>
Probing for SOAP 1.2
<Fault xmlns="http://schemas.xmlsoap.org/soap/envelope/"><faultcode>soap:Server</faultcode><faultstring>Error sending / receiving message: Timed out waiting for response</faultstring><faultactor>urn:localhost</faultactor></Fault>
root@ananas:~/zm#
```
profiles option returns some errors now:
```
root@ananas:~/zm# zmonvif-probe.pl -v profiles http://192.168.0.48:8899/onvif/device_service
ERror  cannot add_IPFilter
ERror  cannot add_ZeroConfiguration
ERror  cannot add_IPVersion6
ERror  cannot add_DynDNS
ERror  cannot add_Extension
ERror  cannot add_DiscoveryResolve
ERror  cannot add_DiscoveryBye
ERror  cannot add_RemoteDiscovery
ERror  cannot add_SystemBackup
ERror  cannot add_SystemLogging
ERror  cannot add_FirmwareUpgrade
ERror  cannot add_SupportedVersions
ERror  cannot add_Extension
ERror  cannot add_TLS1__1
ERror  cannot add_TLS1__2
ERror  cannot add_OnboardKeyGeneration
ERror  cannot add_AccessPolicyConfig
ERror  cannot add_X__509Token
ERror  cannot add_SAMLToken
ERror  cannot add_KerberosToken
ERror  cannot add_RELToken
ERror  cannot add_Extension
Received message:
<GetProfilesResponse xmlns="http://www.onvif.org/ver10/media/wsdl"><Profiles token="000" fixed="true"><Name xmlns="http://www.onvif.org/ver10/schema">Profile_000</Name><VideoSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoS_000</Name><UseCount>3</UseCount><SourceToken>000</SourceToken><Bounds x="0" y="0" width="1280" height="720"></Bounds></VideoSourceConfiguration><AudioSourceConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>Audio_000</Name><UseCount>2</UseCount><SourceToken>000</SourceToken></AudioSourceConfiguration><VideoEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>VideoE_000</Name><UseCount>1</UseCount><Encoding>H264</Encoding><Resolution><Width>1280</Width><Height>720</Height></Resolution><Quality>5</Quality><RateControl><FrameRateLimit>25</FrameRateLimit><EncodingInterval>1</EncodingInterval><BitrateLimit>3072</BitrateLimit></RateControl><H264><GovLength>2</GovLength><H264Profile>Main</H264Profile></H264><Multicast><Address><Type>IPv4</Type><IPv4Address>224.1.2.3</IPv4Address></Address><Port>0</Port><TTL>0</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT10S</SessionTimeout></VideoEncoderConfiguration><AudioEncoderConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>AudioE_000</Name><UseCount>2</UseCount><Encoding>G711</Encoding><Bitrate>44</Bitrate><SampleRate>8</SampleRate><Multicast><Address><Type>IPv4</Type><IPv4Address>224.1.2.3</IPv4Address></Address><Port>0</Port><TTL>0</TTL><AutoStart>false</AutoStart></Multicast><SessionTimeout>PT10S</SessionTimeout></AudioEncoderConfiguration><VideoAnalyticsConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>Analytics_000</Name><UseCount>2</UseCount><AnalyticsEngineConfiguration><AnalyticsModule Name="tt:CellMotionEngine" Type="tt:CellMotionEngine"><Parameters><SimpleItem Name="Sensitivity" Value="4"></SimpleItem><ElementItem Name="Layout"></ElementItem></Parameters></AnalyticsModule><AnalyticsModule Name="tt:TamperEngine" Type="tt:TamperEngine"><Parameters><SimpleItem Name="Sensitivity" Value="4"></SimpleItem><ElementItem Name="Field"></ElementItem><ElementItem Name="Transform"></ElementItem></Parameters></AnalyticsModule></AnalyticsEngineConfiguration><RuleEngineConfiguration><Rule Name="tt:CellMotionDetector" Type="tt:CellMotionDetector"><Parameters><SimpleItem Name="ActiveCells" Value="zwA"></SimpleItem><SimpleItem Name="AlarmOffDelay" Value="1000"></SimpleItem><SimpleItem Name="AlarmOnDelay" Value="1000"></SimpleItem><SimpleItem Name="MinCount" Value="4"></SimpleItem></Parameters></Rule><Rule Name="tt:TamperDetector" Type="tt:TamperDetector"><Parameters><ElementItem Name="Field"></ElementItem></Parameters></Rule></RuleEngineConfiguration></VideoAnalyticsConfiguration><PTZConfiguration xmlns="http://www.onvif.org/ver10/schema"><Name>PTZ_000</Name><UseCount>2</UseCount><NodeToken>000</NodeToken><DefaultAbsolutePantTiltPositionSpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/PositionGenericSpace</DefaultAbsolutePantTiltPositionSpace><DefaultAbsoluteZoomPositionSpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/PositionGenericSpace</DefaultAbsoluteZoomPositionSpace><DefaultRelativePanTiltTranslationSpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/TranslationGenericSpace</DefaultRelativePanTiltTranslationSpace><DefaultRelativeZoomTranslationSpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/TranslationGenericSpace</DefaultRelativeZoomTranslationSpace><DefaultContinuousPanTiltVelocitySpace>http://www.onvif.org/ver10/tptz/PanTiltSpaces/VelocityGenericSpace</DefaultContinuousPanTiltVelocitySpace><DefaultContinuousZoomVelocitySpace>http://www.onvif.org/ver10/tptz/ZoomSpaces/VelocityGenericSpace</DefaultContinuousZoomVelocitySpace><DefaultPTZSpeed><PanTilt x="0" y="0" space="http://www.onvif.org/ver10/tptz/PanTiltSpaces/GenericSpeedSpace"></PanTilt><Zoom x="0" space="http://www.onvif.org/ver10/tptz/ZoomSpaces/ZoomGenericSpeedSpace"></Zoom></DefaultPTZSpeed><DefaultPTZT [... cut ...] oStart></Multicast><SessionTimeout>PT10S</SessionTimeout></VideoEncoderConfiguration></Profiles></GetProfilesResponse>
000, Profile_000, H264, 1280, 720, 25, rtsp://192.168.0.48:554/user=admin_password=tlJwpbo6_channel=1_stream=0.sdp?real_stream
001, Profile_001, H264, 704, 576, 25, rtsp://192.168.0.48:554/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp?real_stream
002, Profile_002, JPEG, 704, 576, 1, rtsp://192.168.0.48:554/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp?real_stream
```

knnniggett, 2017-05-18:
Closing all the Onvif issues out since the latest release contains improvements to the Onvif probe. Please create a new issue if problems persist.

closed 2017-05-18 by knnniggett

ksya, 2018-01-22:
I cannot find my camera using the probe tool, but I am able to find the profiles and get a similar response as rhaamo.
How to manually fill in the credentials? Or isn't that enough to make it work?
See the issue I wrote here: https://www.reddit.com/r/ZoneMinder/comments/7s0hlw/onvif_control_with_authentication/