ZoneMinder/zoneminder #1351: Pid field is too small

abishai, 2016-03-20:
Pid field in database schema is smallint, however PIDs at least in FreeBSD are 1 to 99999
https://www.freebsd.org/doc/handbook/basics-processes.html
At least MEDIUMINT (better to use INT for compatibility) is needed for that.

knnniggett, 2016-03-20:
Yes, this is what I suspected. I just don't know FreeBSD well enough to find documentation which states what the range was. Glad you referenced it.

I'll let @kylejohnson take the lead on writing a PR for this since he just recently made the jump to FreeBSD on one of his production machines. I think we caught this just in time for the nest release.

abishai, 2016-03-20:
I'm planning to move from CentOS to FreeBSD (I just love FreeBSD). I found that ZoneMinder port http://www.freshports.org/multimedia/zoneminder/ is extremely outdated. I started to work on the new one. At the moment my port compiles and installs ZoneMinder HEAD. Now I'm looking if I can use nginx to power zms instead of apache (I just love nginx). If @kylejohnson already works on the port, I can halt or share it.

josh4trunks, 2016-03-30:
@abishai
I have some experience with porting stuff to freebsd. *can you host your modified port on github so we can work on it together?* then over we have it ready we can file a bug on bugs.freebsd.org. I think this needs a maintainer as well so maybe we can request to transfer it to you when we're done.

I also am decent at writing nginx configs and did have zoneminder working with nginx several years ago when I still used Ubuntu on my servers.

when I have more time i'd love to dive in and get a working port & freebsd based pbi/docker image & nginx config

abishai, 2016-03-31:
@josh4trunks 
https://github.com/abishai/Zoneminder-port

josh4trunks, 2016-03-31:
@abishai thanks when I have some free time I'll take a look

PR #1389 (2016-04-05): Larger pid
Fixes #1351
closed 2016-04-05 by knnniggett