ZoneMinder/zoneminder #1354: zmrepo ffmpeg package for el7 won't accept buffer_size option

jburgess777, 2016-03-20:
I experience similar problems to those described on #811 which notes that the default socket buffer used by ffmpeg rtsp streaming is too small to hold an entire frame sent by the camera. When I try to set the buffer_size option on my CentOS7 setup it was ignored. The output below shows that ffmpeg still sets the UDP sockets to the default 64kB receive size:
```
$ sudo strace -e trace=setsockopt src/zmc -m 3
...
setsockopt(9, SOL_SOCKET, SO_RCVBUF, [65536], 4) = 0
setsockopt(10, SOL_SOCKET, SO_RCVBUF, [65536], 4) = 0
setsockopt(11, SOL_SOCKET, SO_RCVBUF, [65536], 4) = 0
setsockopt(12, SOL_SOCKET, SO_RCVBUF, [65536], 4) = 0
```
After digging around in code I added some extra logging (#1353) which confirms that ffmpeg doesn't recognize the option. 
```
zmc[24658]: WAR [Option buffer_size not recognized by ffmpeg]
```
This is using using the ffmpeg-2.1.5-1.el7.centos.x86_64 from zmrepo.

On a different machine running F23 I compiled same ZoneMinder code against the rpmfusion ffmpeg-2.8.6-1.fc23.x86_64 and found that it correctly accepts and applies the requested buffer_size:
```
setsockopt(9, SOL_SOCKET, SO_RCVBUF, [262144], 4) = 0
```
It seems likely that this is a bug in the old ffmpeg shipped in zmrepo and it would probably be useful to upgrade it.

knnniggett, 2016-03-22:
It would probably be better if you created any future zmrepo issues at this github site:
https://github.com/knnniggett/zmrepo

I'm the only one on the team that uses Redhat so we might as well have this conversation elsewhere.

The short answer to your request is not now. 

Rebuilding ffmpeg requires rebuilding vlc and then of course zoneminder. There are a lot of dependencies which make this a very time consuming task. It really sucks that RPMfusion does not host rpms for EL7. That would make my life a lot easier.

All the ffmpeg and vlc packages come straight from Rpmfusion. For the case of Fedora 23, they are merely rsynced right from the site. They are not rebuilt if the target distro matches.

SteveGilvarry, 2016-07-27:
Distro specific issue as per @knnniggett 's suggestion discuss in his zmrepo issues list.

closed 2016-07-27 by SteveGilvarry

knnniggett, 2016-07-27:
We have to release 1.30.0 first before I can update ffmpeg, vlc, and then rebuild zoneminder against them both. 

Note recently, rpmfusion has shown signs of *finally* adding support for EL7 on their web site:
http://rpmfusion.org/

This will effectively relieve me of having to manage ffmpeg and vlc in zmrepo.