ZoneMinder/zoneminder #1357: (FreeBSD) Got signal 4 (Illegal instruction), crashing

abishai, 2016-03-22:
I'm continue to test ZM on FreeBSD. I hope, this one is new.
ZM version ecb8c48
I setup IP camera (MJPEG), default zone. Motion detection works, however I have a crash on playback attempt (still frames works). 

2016-03-22 23:22:26.646094	zms		100208	ERR	Got signal 4 (Illegal instruction), crashing	zm_signal.cpp	54

Here is monitor config it it matters (funny warning included, but crashing is on another camera)
https://gist.github.com/abishai/5119cc918a4fe64ccf5b

abishai, 2016-04-17:
Crashing occurs when sendfile support is used and zm_sendfile fires from EventStream::sendFrame

int zm_sendfile(int out_fd, int in_fd, off_t *offset, off_t size) {
        int err;
        err = sendfile(in_fd, out_fd, *offset, size, NULL, &size, 0);  <- Signal 4

josh4trunks, 2016-04-27:
I'm assuming there's no way to disable sendfile with with a cmake flag? We just gotta patch the source then in the freebsd port?
I'm getting this same issue today in my testing

abishai, 2016-04-27:
We disabled it already. ZM has option for that.

abishai, 2016-04-27:
Ouch, right. I accidently rolled back it :(
https://github.com/abishai/Zoneminder-port/commit/b6a520d69d72e154b9bd405cb24293647e931db0#diff-b67911656ef5d18c4ae36cb6741b7965

josh4trunks, 2016-04-27:
ohh, i seem to remember that line there. ok, now I know how to fix it thanks =]

josh4trunks, 2016-04-27:
unrelated, but since you're here. I think a good use of `pkg-message` is to tell users to mount a tmpfs somewhere. I don't think you can mount to /dev/shm, but I just did /tmp and changed the setting in zoneminder

abishai, 2016-04-27:
Make pull request, please :) I can't do it myself now and will forget it when come home.

You can, but putting user stuff in /dev is a bad idea. I think to write dedicated readme for FreeBSD and pull request it here. After that, we can just mention it in pkg-message.

BTW, should we patch ZM default path from /dev/shm to /tmp ?

josh4trunks, 2016-04-27:
yeah, I think that would be good so at least it works. it would have bad performance unless they setup a tmpfs but better than it not working and admins being confused.

ok, I'll work on a PR tomorrow for the following
- disable sendfile
- change tmp path
- move sudo to run dependency

abishai, 2016-04-27:
I doubt @connortechnology will change default ZM_PATH_MAP and there is no simple solution to make setting platform-dependent. I think just patch this in port. It's not critical thing after all.

josh4trunks, 2016-04-27:
ohh, I meant a PR on your port. didn't understand what you were saying earlier, sorry getting late for me. ok I'll deal with this tomorrow.

abishai, 2016-04-27:
That was thinking out loud :) I looked into ZM repo to see if we can put platform aware default settings.

knnniggett, 2016-04-27:
Trouble is, every distro, Linux, FreeBSD or otherwise, have their own unique defaults for many of the settings that ZoneMinder relies upon.

Personally, I hate managing a pile of patches (which incidentally was what one had to do to build CentOS & Fedora before I starting building packages).

To overcome this exact issue, I wrote zmeditconfigdata.sh
https://github.com/ZoneMinder/ZoneMinder/blob/master/utils/zmeditconfigdata.sh

What this script does is scan configdata.pm.in for the variable you give it and then change its value to the value you give it. You **HAVE** to run it **BEFORE** you call cmake. Otherwise it will have exactly zero effect.

It is written in bash and I admit I did not have FreeBSD in mind when I wrote it. However, it should work as long as you've got bash installed. It could be rewritten in Perl if there is a good reason to.

Call it like this in your packaging script:
./utils/zmeditconfigdata.sh ZM_PATH_SWAP /tmp

Call it as many times as you need.

abishai, 2016-04-27:
Thank you for the idea!

josh4trunks, 2016-04-27:
looks like you already did the work =]
now I feel like I need to contribute, maybe I'll work on `pkg-message` / freebsd entry

closed 2016-07-27 by SteveGilvarry