ZoneMinder/zoneminder #1359: Zoneminder mixes multicast streams

mattpepin, 2016-03-24:
Zoneminder seems to confuse multicast streams. If I use multiple cameras in RTP/Multicast mode, all cameras will use the same picture and switch randomly between cameras. All devices have different multicast groups. It seems like I'm not the only one who experienced this issue. I found a forum post here:

https://forums.zoneminder.com/viewtopic.php?t=23239

I'm using up-to-date h264-video-storage git branch.

knnniggett, 2016-03-24:
If one reads through the link you referenced, you can see the end user worked around the problem by switching to a source type of libvlc. This implies ZoneMinder is not mixing multicast streams, rather ffmpeg is. Suggest you try switching source types to libvlc or investigate additional ffmpeg options which might help. Perhaps @SteveGilvarry can provide some advice on what ffmpeg options might be applicable.

mattpepin, 2016-03-24:
Yes, I saw it. However, to use the h264 video record features, you need to use a ffmpeg source type. I can't switch to libvlc.

knnniggett, 2016-03-24:
Then looking into the many ffmpeg options sounds like your best bet. If it's possible, try a newer version of ffmpeg.

mattpepin, 2016-03-24:
Ah, it looks like a bug in ffmpeg indeed. There is a recent patch (Jan 2016):

https://ffmpeg.org/pipermail/ffmpeg-devel/2016-January/186756.html

knnniggett, 2016-03-24:
> Ah, it looks like a bug in ffmpeg indeed. There is a recent patch (Jan 2016):
> 
> https://ffmpeg.org/pipermail/ffmpeg-devel/2016-January/186756.html

Here is the same commit in github:
https://github.com/FFmpeg/FFmpeg/commit/6f5048f4a0f702d92794dda7752c20ed6033233b

I've gone through their most recent release branches, and this commit is not in any of them, so there won't be any ffmpeg package currently available with this fix in it. Looks like you will need to roll your own or just wait for that change to trickle down to the masses.

closed 2016-03-24 by knnniggett

mattpepin, 2016-03-24:
Thanks for finding the commit. I appreciate it. I tried the patch and it works fine :+1: