ZoneMinder/zoneminder #1363: Empty System Log screen

czettnersandor, 2016-03-25:
When I open the System Log, it's not refreshing. In Chrome's Developer toolbar, I can see a request to index.php with the following parameters:

view:request
request:log
task:query
limit:100
minTime:0

And this is the content I'm getting back:

https://gist.github.com/czettnersandor/332d962d0e1328421ea9

Zoneminder is installed to an Ubuntu 14.04.1 server (32 bit) using the official documenattion:
http://zoneminder.readthedocs.org/en/stable/installationguide/ubuntu.html#easy-way-install-zoneminder-from-a-package-ubuntu-14-x

There was log, but then it broke after a few days of usage.

schrorg, 2016-04-02:
Do you see any recent log entries in /var/log/zm/* ?
How is the setting of LOG_LEVEL_DATABASE (in Options -> Logging)?

czettnersandor, 2016-04-02:
LOG_LEVEL_DATABASE is set to "Warning"

/var/log/zm/* is empty. After a few days, logs are back again, so I can see them. Maybe it's logging to somewhere else?

codewise-nicolas, 2016-05-19:
I too had the same problem, empty Log window.
As  czettnersandor gist shows, the problem is related to a parsing of the JSON. 
"unexpected error 5" means 
`JSON_ERROR_UTF8 	Malformed UTF-8 characters, possibly incorrectly encoded`

I believe something in the logs is malformed and causing the parsing to fail.
After some time the bad/troublesome log entry gets cleared, and so the Log display was working again.

I wonder which entry in his case (and mine) is causing the parsing to fail.

SteveGilvarry, 2016-07-27:
@czettnersandor and @codewise-nicolas any chance you are both using utf-8 characters? Like a particular language?
It will help to get a dump of the logs table while the logs are not displaying.

SteveGilvarry, 2016-07-29:
I was actually able to replicate this condition. I was debugging some code and passed a std:string without c_str into the log. When I checked the DB it had written garbage and the log didn't work until I deleted those log entries in the DB.

mbm329, 2017-04-05:
Just had the same issue on a new install running Fedora 25 and ZoneMinder v1.30.2 from zmrepo
"Resolved" the issue by:
1. Setting Options > Logging > LOG_DATABASE_LIMIT = 1 hour
2. Wait for a bit to let the logs purge from the DB.
3. Set LOG_DATABASE_LIMIT back to original setting.

czettnersandor, 2017-04-11:
@SteveGilvarry: It didn't happen since I last reported it, but I'm using the default locale and I haven't used anything than ASCII anywhere near this system to name cameras, etc.

czettnersandor, 2017-04-11:
I just discovered this part of the log error I attached earlier:
```
      12 => \'zmpkg\',
      13 => \'zms\',
      14 => \'zmupdate\',
      15 => \'zmwatch\',
16 => \'¸…ëQ¸¾?\',
```
How can this get into the logs?

harshmoorjani, 2017-07-05:
Hi,

I digged into my logs and found these -

`1499206278.872092, 'zmc_m3', 0, 30656, -3, 'FAT', 'swscale does not support the codec format: ÿÿÿÿ', 'zm_ffmpeg_camera.cpp', 366`

`1499206878.967885, 'zmc_m1', 0, 31477, -3, 'FAT', 'swscale does not support the codec format: ÿÿÿÿ', 'zm_ffmpeg_camera.cpp', 366`

removing these opened the logs properly.

closed 2017-07-07 by knnniggett

jeroenst, 2020-05-31:
A button to clear the complete log would be a very nice feature....

PR #1942 (2017-07-07): Strip non 7-bit ascii chars from log message
Fix #1363

Some elements of our code can insert non-7-bit ascii chars into the logs.  When encoding/decoding these to/from json to view in the web ui the json decode dies.  

So we strip anything that is non 7-bit ascii out.