ZoneMinder/zoneminder #1365: Default install directory is on a the small /var partition in Centos

tpcarmen, 2016-03-26:
/var/lib on a default Centos install is a small partition, while /home gets all the available disk space.

A new ZM install will fill up it's tiny partition very quickly, leaving the large /home partition untouched.

While /var is the traditional home for services, the ZM data should be on whatever partition has the most free space.

knnniggett, 2016-03-26:
ZoneMinder follows the Filesystem Hierarchy Standard (https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard) to determine where files should be placed on the filesystem, and in the case of RedHat based distros, the Fedora Packaging Guidlines are adhered to as well (https://fedoraproject.org/wiki/Packaging:Guidelines).

These folders aren't going to change because they are inconvenient for you.

Instead, you should dedicate an entire drive or partition to the events folder, and mount it according to the following guide, in order to prevent critical areas on your filesystem from filling up:
https://wiki.zoneminder.com/Using_a_dedicated_Hard_Drive

closed 2016-03-26 by knnniggett

tpcarmen, 2016-03-26:
> These folders aren't going to change because they are inconvenient 
for you.

The storage location isn't "inconvenient" it's wrong.

ZM makes lots of files. They don't belong in /var and the way it is now, 
is simply an ambush for every single new ZM user.

If the ZM package maintainer is unwilling to put the data somewhere that 
won't fill up and crash the system, then the app should ask where to put 
the data, when it's first run. Or refuse to run until properly configured.

New users aren't going to see the wiki until they've already filled up 
their filesystem.

There's no reason to leave an ambush waiting.

Terry

knnniggett, 2016-03-26:
We follow the documentation and standards that the rest of the Linux community follows.
We certainly don't follow the opinion of strangers who show up out of nowhere and try to claim their opinions are "right" while the rest of the community is "wrong".

Like all video recording software, ZoneMinder will fill the disk up. However, ZoneMinder ships with the purge filter enabled, which will prevent the system from crashing. And yes, that is something else the end user will need to read about after installing ZoneMinder. The fact is, there are many things the end user needs to read and learn about after installing ZoneMinder. A future version of ZoneMinder will have an install wizard to help ease the learning curve, but for now there is no substitute for rtfm.

Feel free to mount your events folder under the home folder. I think that is a terrible idea, but its your system. Do what you want. One thing we aren't going to do is continue with this discussion of where files should be placed when there are clear standards which already tell us what to do, which, by the way, would get our package kicked out of popular repos like Debian if we deviated from the filesystem standards.